espressif/esp-idf #298: always reboot like this

mybays, 2017-01-25:
Rebooting...
32ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0008,len:8
load:0x3fff0010,len:3388
ho 0 tail 12 room 4
load:0x40078000,len:7504
load:0x40080000,len:252
entry 0x40080034
I (47) boot: ESP-IDF v2.0-rc1-2-g8e467801 2nd stage bootloader
I (47) boot: compile time 21:13:04
I (79) boot: Enabling RNG early entropy source...
I (79) boot: SPI Speed      : 40MHz
I (80) boot: SPI Mode       : DIO
I (92) boot: SPI Flash Size : 4MB
I (105) boot: Partition Table:
I (116) boot: ## Label            Usage          Type ST Offset   Length
I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (162) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (185) boot:  2 factory          factory app      00 00 00010000 00100000
I (209) boot: End of partition table
I (221) boot: Disabling RNG early entropy source...
I (239) boot: Loading app partition at offset 00010000
I (628) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512) 
I (628) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x03224 ( 12836) map
I (644) boot: segment 2: paddr=0x00023234 vaddr=0x3ffc0000 size=0x01c08 (  7176) load
I (674) boot: segment 3: paddr=0x00024e44 vaddr=0x40080000 size=0x00400 (  1024) load
I (697) boot: segment 4: paddr=0x0002524c vaddr=0x40080400 size=0x11114 ( 69908) load
I (756) boot: segment 5: paddr=0x00036368 vaddr=0x400c0000 size=0x00000 (     0) load
I (757) boot: segment 6: paddr=0x00036370 vaddr=0x00000000 size=0x09c98 ( 40088) 
I (774) boot: segment 7: paddr=0x00040010 vaddr=0x400d0018 size=0x10b2c ( 68396) map
I (801) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
I (824) heap_alloc_caps: At 3FFC37F8 len 0001C808 (114 KiB): DRAM
I (844) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
I (866) heap_alloc_caps: At 40091514 len 0000EAEC (58 KiB): IRAM
I (886) cpu_start: Pro cpu up.
I (898) cpu_start: Starting app cpu, entry point is 0x400809c0
I (0) cpu_start: App cpu up.
I (929) cpu_start: Pro cpu start user code
V (980) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (981) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (994) intr_alloc: Connected src 56 to int 2 (cpu 0)
V (1010) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1029) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0xE
D (1054) intr_alloc: Connected src 16 to int 3 (cpu 0)
V (1070) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1090) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x40E
D (1115) intr_alloc: Connected src 24 to int 12 (cpu 0)
I (1132) cpu_start: Starting scheduler on PRO CPU.
:cohyckdntaafgom[400EspVintrOal o1_1n1
V892 1o iffaeal0,cl e p0in18_a
osei .bst 0u0 fcpm 1)fbAr00 oo 3.fRfsult lgnf0ags70700EDm 
OM VER(IO1)010tr_BllAcD C nne0t:d4s0c 25 t6 
ED[0;Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4008fc78  PS      : 0x00060d30  A0      : 0x8008282e  A1      : 0x3ffc7750  
A2      : 0x3ffc25e4  A3      : 0xffff00ff  A4      : 0x00000050  A5      : 0x3ffb0090  
A6      : 0x47454341  A7      : 0x4f4d4b49  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000067  A11     : 0x00000001  A12     : 0x0000000f  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000a8  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008fc78:0x3ffc7750 0x4008282e:0x3ffc7770 0x400559c9:0x3ffc77e0 0x40082115:0x3ffc7810 0x40082141:0x3ffc7830 0x400dc5bc:0x3ffc7850

Rebooting...

Spritetm, 2017-01-25:
Yeah, that doesn't really help... What code are you trying to run, what is your hardware, are you sure your power supply can deliver >=500mA?

mybays, 2017-01-25:
Ok,I`ll check the power supply first.

closed 2017-01-25 by mybays