espressif/esp-idf #299: connecting wifi gives unhandled debug exception in phy_chip_v7

lollizze, 2017-01-25:
Simplified my application does as follows
Through the uart i can configure the wifi settings and then enabling it to connect to an AP
When i try to connect to an AP it sometimes gives an exception. It seems random, because i can't reproduce it every time and I can't see the what is on line 1708 in phy_chip_v7 to try and find a solution. 
Could it be some timing issue, whatever ppTask does it gets interfered by wifi API calls? 

Heres the backtrace:
#0  spur_cal (freq=<optimized out>, BW_h=<optimized out>,
    spur_freq_cfg=32767, spur_freq_cfg_div=0 '\000') at phy_chip_v7.c:1708
#1  0x40087c51 in spur_cal (spur_freq_cfg_div=1 '\001',
    spur_freq_cfg=<optimized out>, BW_h=<optimized out>, freq=2417)
    at phy_chip_v7.c:1701
#2  ram_spur_coef_cfg (channel=2 '\002', freq=2417,
    sub_chan_cfg=<optimized out>) at phy_chip_v7.c:1739
#3  0x40087360 in chip_v7_set_chan (channel=<optimized out>,
    sub_chan_cfg=<optimized out>) at phy_chip_v7_ana.c:653
#4  0x40088d38 in set_chanfreq (chanFreq=2417, sub_chan_cfg=0 '\000')
    at phy_chip_v7.c:1850
#5  0x400856e4 in phy_change_channel (chanFreq=2417, bHomeChan=1 '\001',
    pBReset=0 '\000', sub_chan_cfg=0 '\000') at phy.c:13
#6  0x400ddc4e in chm_set_current_channel ()
#7  0x400ddc81 in ?? ()
#8  0x400ddce6 in chm_start_op ()
#9  0x400dc150 in ?? ()
#10 0x400dc16e in scan_enter_oper_channel_process ()
#11 0x400dd9a8 in ?? ()
#12 0x400dda31 in ieee80211_timer_do_process ()
#13 0x40098179 in ppTask ()

negativekelvin, 2017-01-25:
Are you sure is not weak power supply issue?

lollizze, 2017-01-25:
100% sure no, but i figured it would show up more frequently if it where that

projectgus, 2017-01-26:
This feels like a power supply/decoupling issue to me, calibration tends to be a power consumption spike.

Do you have any details of the type of exception? If IllegalInstructionException then almost certainly a power issue. If something else then possibly.

Do you get the problem running any of the examples bundled with esp-idf?

lollizze, 2017-01-26:
Tried to simplify it to pin point the problem, but with no luck. Maybe it is a power problem, but it is connected to an usb hub with external power so i thought that would be sufficient

closed 2017-03-07 by lollizze