espressif/esp-idf #300: I2S-DMA buffer issue

valioIv, 2017-01-25:
I'm trying to use the I2S test application as a foundation for some audio project:

https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2s/main/app_main.c

I did NO modifications on that, I've just connected a WM8731 audio codec chip and it started working properly. I hear the sound, but I noticed it's a bit buzzy. I observe the following waveforms for the left and right channel respectively:

[image]

[image]

It is obvious there are buffers which are not filled with the real data but just zeros.
I really suspect this is some DMA buffer management issue. Most probably the buffer indexes are not updated and/or used properly. I tried to dig in the I2S-DMA affairs but I don't feel myself confident with it. Can somebody share their opinion at least?

valioIv, 2017-01-25:
Forgot to share my code (the only changes I did are in the App):

`#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "driver/i2c.h"
#include "driver/i2s.h"
#include <math.h>

#define I2C_MASTER_SCL_IO    19    /*!< gpio number for I2C master clock */
#define I2C_MASTER_SDA_IO    18    /*!< gpio number for I2C master data  */
#define I2C_MASTER_NUM I2C_NUM_1   /*!< I2C port number for master dev */
#define I2C_MASTER_TX_BUF_DISABLE   0   /*!< I2C master do not need buffer */
#define I2C_MASTER_RX_BUF_DISABLE   0   /*!< I2C master do not need buffer */
#define I2C_MASTER_FREQ_HZ    100000     /*!< I2C master clock frequency */

#define WM8731_I2C_ADDR  0x1A    /*!< slave address for WM8731 sensor */
#define WRITE_BIT  I2C_MASTER_WRITE /*!< I2C master write */
#define READ_BIT   I2C_MASTER_READ  /*!< I2C master read */
#define ACK_CHECK_EN   0x1     /*!< I2C master will check ack from slave*/
#define ACK_CHECK_DIS  0x0     /*!< I2C master will not check ack from slave */
#define ACK_VAL    0x0         /*!< I2C ack value */
#define NACK_VAL   0x1         /*!< I2C nack value */

#define WM8731_LINVOL   0x00
#define WM8731_RINVOL   0x01
#define WM8731_LOUT1V   0x02
#define WM8731_ROUT1V   0x03
#define WM8731_APANA    0x04
#define WM8731_APDIGI   0x05
#define WM8731_PWR      0x06
#define WM8731_IFACE    0x07
#define WM8731_SRATE    0x08
#define WM8731_ACTIVE   0x09
#define WM8731_RESET	0x0f

#define WM8731_SAMPLING_FREQ_8K		8000u
#define WM8731_SAMPLING_FREQ_32K	32000u
#define WM8731_SAMPLING_FREQ_44K1	44100u
#define WM8731_SAMPLING_FREQ_48K	48000u
#define WM8731_SAMPLING_FREQ_88K2	88200u
#define WM8731_SAMPLING_FREQ_96K	96000u

#define WM8731_MCLK 12000000u //6144000u//12000000u

//#define CODEC_IS_MASTER

#define SAMPLE_RATE     (36000)
#define I2S_NUM         (0)
#define WAVE_FREQ_HZ    (100)
#define PI 3.14159265

#define SAMPLE_PER_CYCLE (SAMPLE_RATE/WAVE_FREQ_HZ)

typedef struct wm8731_cmd_entry_s {
	unsigned int reg_addr;
	unsigned int reg_val;
} wm8731_cmd_entry_t;

typedef struct _sampling_ctrl {
         long mclk;
         long rate;
         unsigned int fs;
         unsigned char sr:4;
         unsigned char bosr:1;
         unsigned char usb:1;
[... 152 lines cut ...]
    i2s_set_pin(I2S_NUM, &pin_config);
    i2s_set_pin(I2S_NUM, NULL);

    triangle_float = -32767;

    for(i = 0; i < SAMPLE_PER_CYCLE; i++) {
        sin_float = sin(i * PI / 180.0);
        if(sin_float >= 0)
            triangle_float += triangle_step;
        else
            triangle_float -= triangle_step;
        sin_float *= 32767;

        sample_val = 0;
        sample_val += (short)triangle_float;
        sample_val = sample_val << 16;
        sample_val += (short) sin_float;

        //printf("0x%x\n", sample_val);

        i2s_push_sample(I2S_NUM, (char *)&sample_val, portMAX_DELAY);
    }
}

`

valioIv, 2017-01-25:
And this is my code (the only changes I did are in the App):

`#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include "driver/i2c.h"
#include "driver/i2s.h"
#include <math.h>

#define I2C_MASTER_SCL_IO    19    /*!< gpio number for I2C master clock */
#define I2C_MASTER_SDA_IO    18    /*!< gpio number for I2C master data  */
#define I2C_MASTER_NUM I2C_NUM_1   /*!< I2C port number for master dev */
#define I2C_MASTER_TX_BUF_DISABLE   0   /*!< I2C master do not need buffer */
#define I2C_MASTER_RX_BUF_DISABLE   0   /*!< I2C master do not need buffer */
#define I2C_MASTER_FREQ_HZ    100000     /*!< I2C master clock frequency */

#define WM8731_I2C_ADDR  0x1A    /*!< slave address for WM8731 sensor */
#define WRITE_BIT  I2C_MASTER_WRITE /*!< I2C master write */
#define READ_BIT   I2C_MASTER_READ  /*!< I2C master read */
#define ACK_CHECK_EN   0x1     /*!< I2C master will check ack from slave*/
#define ACK_CHECK_DIS  0x0     /*!< I2C master will not check ack from slave */
#define ACK_VAL    0x0         /*!< I2C ack value */
#define NACK_VAL   0x1         /*!< I2C nack value */

#define WM8731_LINVOL   0x00
#define WM8731_RINVOL   0x01
#define WM8731_LOUT1V   0x02
#define WM8731_ROUT1V   0x03
#define WM8731_APANA    0x04
#define WM8731_APDIGI   0x05
#define WM8731_PWR      0x06
#define WM8731_IFACE    0x07
#define WM8731_SRATE    0x08
#define WM8731_ACTIVE   0x09
#define WM8731_RESET	0x0f

#define WM8731_SAMPLING_FREQ_8K		8000u
#define WM8731_SAMPLING_FREQ_32K	32000u
#define WM8731_SAMPLING_FREQ_44K1	44100u
#define WM8731_SAMPLING_FREQ_48K	48000u
#define WM8731_SAMPLING_FREQ_88K2	88200u
#define WM8731_SAMPLING_FREQ_96K	96000u

#define WM8731_MCLK 12000000u //6144000u//12000000u

//#define CODEC_IS_MASTER

#define SAMPLE_RATE     (36000)
#define I2S_NUM         (0)
#define WAVE_FREQ_HZ    (100)
#define PI 3.14159265

#define SAMPLE_PER_CYCLE (SAMPLE_RATE/WAVE_FREQ_HZ)

typedef struct wm8731_cmd_entry_s {
	unsigned int reg_addr;
	unsigned int reg_val;
} wm8731_cmd_entry_t;

typedef struct _sampling_ctrl {
         long mclk;
         long rate;
         unsigned int fs;
         unsigned char sr:4;
         unsigned char bosr:1;
         unsigned char usb:1;
[... 156 lines cut ...]
    i2s_set_pin(I2S_NUM, &pin_config);
    i2s_set_pin(I2S_NUM, NULL);

    triangle_float = -32767;

    for(i = 0; i < SAMPLE_PER_CYCLE; i++) {
        sin_float = sin(i * PI / 180.0);
        if(sin_float >= 0)
            triangle_float += triangle_step;
        else
            triangle_float -= triangle_step;
        sin_float *= 32767;

        sample_val = 0;
        sample_val += (short)triangle_float;
        sample_val = sample_val << 16;
        sample_val += (short) sin_float;

        //printf("0x%x\n", sample_val);

        i2s_push_sample(I2S_NUM, (char *)&sample_val, portMAX_DELAY);
    }
}

`

valioIv, 2017-01-26:
@tuanpmt , can you comment on this? Do you have any observations like that?

tuanpmt, 2017-01-27:
@valiolv 
I also met a similar problem, but other reasons. I will find out the cause next week. Củently, I'n on Lunar New Year vacation and not have any hardware to test

valioIv, 2017-01-27:
Great! Happy new year and thanks in advance!

tuanpmt, 2017-02-06:
@valioiv Because i2s tx faster than sinwave calculation. Please re-calculate to a buffer first, then using i2s_write_bytes, or loop and push

valioiv, 2017-02-08:
Oh sure, I see... I'll test tomorrow and will close the "issue". Thanks for the support!

valioiv, 2017-02-09:
Hi :)

I've just modified the part of generation and sending via I2S as you suggested like this:

```c
    unsigned int sample_val_arr[SAMPLE_PER_CYCLE] = { 0 };

    for(i = 0; i < SAMPLE_PER_CYCLE; i++) {
        sin_float = sin(i * PI / 180.0);
        if(sin_float >= 0)
            triangle_float += triangle_step;
        else
            triangle_float -= triangle_step;
        sin_float *= 32767;

        sample_val_arr[i] = 0;
        sample_val_arr[i] += (short)triangle_float;
        sample_val_arr[i] = sample_val_arr[i] << 16;
        sample_val_arr[i] += (short) sin_float;

        //printf("0x%x\n", sample_val);

        // i2s_push_sample(I2S_NUM, (char *)&sample_val, portMAX_DELAY);
    }

    i2s_write_bytes(I2S_NUM, sample_val_arr, sizeof(sample_val_arr), portMAX_DELAY);
```

It works fine, man! Thanks a lot!

closed 2017-02-09 by valioiv

AutoEM, 2020-08-13:
Dear valioiv,

I am trying also to connect my esp32 to  wm8731 audio codec board with no success. Could you please clarify how did you connect your esp32 pins to the codec?

Are the three i2s pins enough or do I have to set up the mclk also?
like here: https://github.com/espressif/arduino-esp32/issues/3932

Thanks in advance!

darwin1234, 2021-01-07:
Hi is there someone can share the schematic to this thread?