espressif/esp-idf #306: undefined reference to `app_main'

ESP32DE, 2017-01-27:
hi guys

this 

`undefined reference to app_main `

 phenomenon comes again over me ( win 7, msys32, last esp-idf )

CC trax.o
CC eri.o
AR libxtensa-debug-module.a
LD hello-world.elf
C:/nod/11/build/esp32\libesp32.a(cpu_start.o):(.literal.main_task+0x8): undefined reference to `app_main'
C:/nod/11/build/esp32\libesp32.a(cpu_start.o): In function `main_task':
C:/sdk32/esp-idf/components/esp32/cpu_start.c:267: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/sdk32/esp-idf/make/project.mk:323: /c/nod/11/build/hello-world.elf] Fehler 1

i search just in time but find no mistake - i used the hello world example for minimal code.

best wishes
rudi ;-)

projectgus, 2017-01-27:
Hi rudi,

Can you run the following command and copy/paste the output, please:
```
make V=1
```

Also, if you can, please run `make clean` and then re-run the above command and copy/paste that output separately.

ESP32DE, 2017-01-27:
Hi Angus,
txs,
step 1:
$ make V=1
including /c/sdk32/esp-idf/components/bootloader/Makefile.projbuild...
including /c/sdk32/esp-idf/components/bootloader_support/Makefile.projbuild...
including /c/sdk32/esp-idf/components/coap/Makefile.projbuild...
including /c/sdk32/esp-idf/components/esp32/Makefile.projbuild...
including /c/sdk32/esp-idf/components/esptool_py/Makefile.projbuild...
including /c/sdk32/esp-idf/components/mbedtls/Makefile.projbuild...
including /c/sdk32/esp-idf/components/nghttp/Makefile.projbuild...
including /c/sdk32/esp-idf/components/partition_table/Makefile.projbuild...
make -C /c/sdk32/esp-idf/components/bootloader/src V=1 BUILD_DIR_BASE=/c/nod/11/build/bootloader TEST_COMPONENTS= /c/nod/11/build/bootloader/bootloader.bin
make[1]: Verzeichnis „/c/sdk32/esp-idf/components/bootloader/src“ wird betreten
including /c/sdk32/esp-idf/components/esptool_py/Makefile.projbuild...
including /c/sdk32/esp-idf/components/bootloader/Makefile.projbuild...
including /c/sdk32/esp-idf/components/bootloader_support/Makefile.projbuild...
make -C /c/nod/11/build/bootloader/bootloader_support -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/bootloader_support/component.mk COMPONENT_NAME=bootloader_support build
make[2]: Verzeichnis „/c/nod/11/build/bootloader/bootloader_support“ wird betreten
make[2]: Verzeichnis „/c/nod/11/build/bootloader/bootloader_support“ wird verlassen
Target 'bootloader_support-build' responsible for '/c/nod/11/build/bootloader/bootloader_support/libbootloader_support.a'
make -C /c/nod/11/build/bootloader/log -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/log/component.mk COMPONENT_NAME=log build
make[2]: Verzeichnis „/c/nod/11/build/bootloader/log“ wird betreten
make[2]: Verzeichnis „/c/nod/11/build/bootloader/log“ wird verlassen
Target 'log-build' responsible for '/c/nod/11/build/bootloader/log/liblog.a'
make -C /c/nod/11/build/bootloader/spi_flash -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/spi_flash/component.mk COMPONENT_NAME=spi_flash build
make[2]: Verzeichnis „/c/nod/11/build/bootloader/spi_flash“ wird betreten
make[2]: Verzeichnis „/c/nod/11/build/bootloader/spi_flash“ wird verlassen
Target 'spi_flash-build' responsible for '/c/nod/11/build/bootloader/spi_flash/libspi_flash.a'
make -C /c/nod/11/build/bootloader/micro-ecc -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/micro-ecc/component.mk COMPONENT_NAME=micro-ecc build
make[2]: Verzeichnis „/c/nod/11/build/bootloader/micro-ecc“ wird betreten
make[2]: Verzeichnis „/c/nod/11/build/bootloader/micro-ecc“ wird verlassen
Target 'micro-ecc-build' responsible for '/c/nod/11/build/bootloader/micro-ecc/libmicro-ecc.a'
make -C /c/nod/11/build/bootloader/main -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/bootloader/src/main/component.mk COMPONENT_NAME=main build
make[2]: Verzeichnis „/c/nod/11/build/bootloader/main“ wird betreten
make[2]: Verzeichnis „/c/nod/11/build/bootloader/main“ wird verlassen
Target 'main-build' responsible for '/c/nod/11/build/bootloader/main/libmain.a'
make[1]: Verzeichnis „/c/sdk32/esp-idf/components/bootloader/src“ wird verlassen
make -C /c/nod/11/build/app_update -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/app_update/component.mk COMPONENT_NAME=app_update build
make[1]: Verzeichnis „/c/nod/11/build/app_update“ wird betreten
make[1]: Verzeichnis „/c/nod/11/build/app_update“ wird verlassen
Target 'app_update-build' responsible for '/c/nod/11/build/app_update/libapp_update.a'
make -C /c/nod/11/build/bootloader_support -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/bootloader_support/component.mk COMPONENT_NAME=bootloader_support build
make[1]: Verzeichnis „/c/nod/11/build/bootloader_support“ wird betreten
make[1]: Verzeichnis „/c/nod/11/build/bootloader_support“ wird verlassen
Target 'bootloader_support-build' responsible for '/c/nod/11/build/bootloader_support/libbootloader_support.a'
make -C /c/nod/11/build/bt -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/bt/component.mk COMPONENT_NAME=bt build
make[1]: Verzeichnis „/c/nod/11/build/bt“ wird betreten
make[1]: Verzeichnis „/c/nod/11/build/bt“ wird verlassen
Target 'bt-build' responsible for '/c/nod/11/build/bt/libbt.a'
make -C /c/nod/11/build/coap -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/coap/component.mk COMPONENT_NAME=coap build
make[1]: Verzeichnis „/c/nod/11/build/coap“ wird betreten
make[1]: Verzeichnis „/c/nod/11/build/coap“ wird verlassen
Target 'coap-build' responsible for '/c/nod/11/build/coap/libcoap.a'
make -C /c/nod/11/build/cxx -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/cxx/component.mk COMPONENT_NAME=cxx build
make[1]: Verzeichnis „/c/nod/11/build/cxx“ wird betreten
make[1]: Verzeichnis „/c/nod/11/build/cxx“ wird verlassen
Target 'cxx-build' responsible for '/c/nod/11/build/cxx/libcxx.a'
make -C /c/nod/11/build/driver -f /c/sdk32/esp-idf/make/component_wrapper.mk COMPONENT_MAKEFILE=/c/sdk32/esp-idf/components/driver/component.mk COMPONENT_NAME=driver build
make[1]: Verzeichnis „/c/nod/11/build/driver“ wird betreten
[... cut ...] -Wl,--end-group -Wl,-EL -o /c/nod/11/build/hello-world.elf -Wl,-Map=/c/nod/11/build/hello-world.map
C:/nod/11/build/esp32\libesp32.a(cpu_start.o):(.literal.main_task+0x8): undefined reference to `app_main'
C:/nod/11/build/esp32\libesp32.a(cpu_start.o): In function `main_task':
C:/sdk32/esp-idf/components/esp32/cpu_start.c:267: undefined reference to `app_main'
collect2.exe: error: ld returned 1 exit status
make: *** [C:/sdk32/esp-idf/make/project.mk:323: /c/nod/11/build/hello-world.elf] Fehler 1

ESP32DE, 2017-01-27:
and here after make clean, the output of make V=1
( You can't comment at this time — your comment is too long (maximum is 65536 characters). 

here the second as file
[log2.txt](https://github.com/espressif/esp-idf/files/736436/log2.txt)

thank you!
best wishes
rudi ;-)

ESP32DE, 2017-01-27:
Hi Angus

sry for this action.
usually i create the "empty" component.mk file in the main folder.
but in this example i have forget to cpy the empty file...
and the undefined mistake comes.

Please - can you check this:

if we have no File: component.mk 
in the main, then the undefined mistake comes

if we have a File: component.mk
in the main, then compile is ok 

but i think this is not ok?
the file in hello_world is empty ( only coments ) :

but if this file is missed, then undefined mistake comes for app_main()

#
# "main" pseudo-component makefile.
#
# (Uses default behaviour of compiling all source files in directory, adding 'include' to include path.)

ESP32DE, 2017-01-27:
Hi Angus,
for reproduce this you can example use the hello world example, 
rename the component.mk example to component.mk_
compile, and compiler catch with undefined reference to `app_main'.

do we need this file (component.mk) allways created, named in the project main folder?

sry again for the action! after i created this ( cpy ) compile is ok.
i had the same problem in the past, but have not reminder me, 
that i have forget to create the ( component.mk ) file in the main folder.

[image]

if we allways need this file, would be nice, you can short answere,
yes - you can close this.

if this is a bug, and we do not need to create this file 
how we can stop the compile error message?

thank you again for your effort!

best wishes
rudi ;-)

Spritetm, 2017-01-28:
The existence of a component.mk is actually how esp-idf recognizes a component, and the 'main' component effectively isn't much more than just another component... but if you don't know that logic, I agree this behaviour can be puzzling. We'll discuss internally if for the main directory we want to make the existence of this file optional.

ESP32DE, 2017-01-28:
Txs Jeroen,
yes, the logic it self how esp-idf recognizes a component is wellknowed,
the "entry" to the main folder and "missing" the ( empty ) component.mk file itself has stopped the compile. that have me let puzzled after i saw my mistake to forget to create an empty file named component.mk in the main

but ok, txs

closed 2017-01-28 by ESP32DE

weshouman, 2021-01-09:
3 years after and I can't agree more with this proposal

projectgus, 2021-01-11:
@weshouman The CMake build system we introduced in ESP-IDF V4.0 does this. If there's a directory in the component search patch with no CMakeLists.txt file (the equivalent of component.mk) then it will log a line like

```
-- Component directory /path/to/direcctory does not contain a CMakeLists.txt file. No component will be added
```