espressif/esp-idf #308: Suggest rename for function - ledc_update_duty

nkolban, 2017-01-29:
If we look at the function called `ledc_update_duty` found here:

https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/ledc.h#L134

We find that its purpose is to cause changes to the duty cycle *or* fade settings to take effect.  However, when one looks at the name of the function, it appears to reference only the `duty` parameter.   This is confusing.  I suggest consideration of a new name such as `ledc_update`.

I'd also like to pose the question on the need for this function.  Could a call to `ledc_set_duty` or `ledc_set_fade` not also incorporate an underlying internal call to this logic to cause the change to take effect.  Under what circumstances would we want to change the duty cycle and THEN later cause it to take effect?  I can only imagine a call to change the duty cycle taking effect immediately.

negativekelvin, 2017-01-30:
I believe fade is a software concept and duty is a hardware concept.  So the function does technically only update duty by updating the internal peripheral registers which may be a time critical operation (setup now, apply later).  It seems like after ledc_set_fade functions one should call ledc_fade_start which then calls ledc_update_duty internally.

nkolban, 2017-01-30:
If we look in the ESP32 Technical Ref, we seem to see that the fade technologies are hardware assist.  We specify the start value, the delta value to be added or subtracted each period, the period length and how many periods the delta should be applied.  But other than that, I don't know ....

negativekelvin, 2017-01-30:
Well it uses interrupts so fade code is implemented in the isr.

Either way, I believe the two use cases are:
ledc_set_duty
ledc_update_duty

Or

ledc_set_fade_with_
ledc_fade_start

So names seem appropriate

nkolban, 2017-01-30:
I'm not sure I know enough about fade to comment.  If the function call "ledc_update_duty" has nothing to do with fade, then maybe the docs on this line need updated:

https://github.com/espressif/esp-idf/blob/master/components/driver/include/driver/ledc.h#L137

negativekelvin, 2017-01-30:
Right, it seems it is relevant to idf developers but not users because ledc_set_fade is not a user function and users would call ledc_fade_start with fade functions.

closed 2017-11-21 by FayeY

DavidAntliff, 2018-01-22:
I think it is a bit confusing as it is right now - to me, if `ledc_update` is not acceptable, then a better name for the function would be `ledc_update_duty_and_fade` since it does **both**. Although duty is used to implement fade, you still need to make a call to `ledc_update_duty` after calling `ledc_set_fade` and the naming doesn't make that clear at all.  Whether the fade is implemented in hardware or software makes no difference - the API is an abstraction.

Alternatively (or additionally), consider providing a function that allows both duty and fade to be set in the same call, and then internally updates them both atomically.

negativekelvin, 2018-01-23:
The naming may not make that clear but the function docs do.  Would suggest you make a pull request for function ledc_update_fade as alias for ledc_update_duty if you think it is important.

DavidAntliff, 2018-01-23:
@negativekelvin yes, the docs do explain the situation (thank goodness), however that is insufficient to meet one of the most important aspects of good API design - make your API self-documenting. That means, first and foremost, using sensible and consistent function names. The docs around them are important, no doubt about that, but they aren't sufficient for good API design.

An alias isn't suitable because it needs to be clear that the current `ledc_update_duty` *also* updates the fade parameter. Otherwise it leads to client code calling two separate function invocations that do the same thing, which isn't actually how the abstraction works. In particular, it will be surprising (another rule of API design - don't surprise people) that the `ledc_update_duty` function also happens to update the fade parameter as a side effect. That's even worse than what we have right now.

As I said before, the solution is to either call it `ledc_update_duty_and_fade` or just call it `ledc_update`. There are other options too, but leaving it as `ledc_update_duty` is not a good one.

negativekelvin, 2018-01-23:
`ledc_update` is a bit ambiguous and `ledc_update_duty_and_fade` is potentially confusing if "I don't want to update fade, I'm not even using fade!?".  What are the use cases to avoid `ledc_set_fade_with` 
and `ledc_fade_start` functions?  I guess `ledc_update_duty` could return an error if fade is enabled on that channel and `ledc_update_fade` could return an error if fade is not enabled