espressif/esp-idf #309: Uploading error: Failed to enter RAM download mode. Only got 2 byte status response.

infosmick, 2017-01-31:
I'm using these ESP:

https://www.aliexpress.com/item/Free-shipping-1pcs-ESP-WROOM-02-development-board-wemos-D1-Nodemcu-wifi-Internet-of-Things/32768794010.html?spm=2114.13010608.0.0.ckE7cZ

and when I upload the Blink sketch obtain these error:
esptool.py v2.0-dev
Connecting...
Uploading stub...

A fatal error occurred: Failed to enter RAM download mode. Only got 2 byte status response.
A fatal error occurred: Failed to enter RAM download mode. Only got 2 byte status response.

Please, can everyone help me?

negativekelvin, 2017-02-01:
This esp-idf project is for ESP32 but that board is ESP8266 so please try http://bbs.espressif.com or http://esp8266.com

closed 2017-02-16 by projectgus

denzelwamburu, 2017-04-27:
Am getting the same error, using Sparkfun ESP8266 thing

Spritetm, 2017-04-27:
Then please do not try to use esp-idf. Esp-idf, for now, is for the ESP32 only.