espressif/esp-idf #311: blufi example does not compile

h-vishal, 2017-02-02:
Same as issue #262 tried compiling on Windows 10, Ubuntu 16.10 and OS X 10.11. The error is `fatal error: esp_blufi_api.h: No such file or directory`

Changing the order of the headers gives the same error for a different file in the `api/include` directory. These files exist and are linked correctly when the other bluetooth examples are compiled.

krzychb, 2017-02-02:
Do you have Bluetooth enabled in menuconfig?
Ref. [readme](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth#bluetooth-examples) in Bluetooth Examples directory.

h-vishal, 2017-02-02:
Yes bluetooth is enabled, this is the default configuration setting.

negativekelvin, 2017-02-02:
For some reason the default Bluetooth setting doesn't always get picked up properly.  I had to go in and enable it manually in menuconfig.

h-vishal, 2017-02-03:
I have verified that the `sdkconfig` parameters are setup correctly. I was able to compile the example by deleting the build directory and configuration file and running the `make` command, on Windows running `make menuconfig` before running the make command causes the process to not exit correctly on selecting the exit option, on Linux it does exit but some of the output is repeated. So `make menuconfig` followed by `make` leads to the error that I have mentioned. This error also appears when using the Arduino and PlatformIO IDEs for which I do not know of a workaround.

projectgus, 2017-02-22:
This should be fixed in master now, there were some issues in the build system where component paths were not being correctly shared to the rest of the system.

I'm not 100% sure it's the same issue you were seeing, but it probably is. Please reopen if you see this problem still occurring, and we'll investigate further.

closed 2017-02-22 by projectgus

h-vishal, 2017-02-25:
I can confirm that the issue seems to be fixed.