espressif/esp-idf #312: Using SD instead of SPI Flash

mseerig, 2017-02-02:
Is there any enhancement in progress, wich allows booting from SD Card?
May be, it's possible to use the SD also as program storage as well as data storage...
That would be a nice solution to have more pins available and makes a firmware update in a integrated product a lot easier. 

Espressif-Team, thank you for doing a great Job! :1st_place_medal:  :)

igrr, 2017-02-02:
ESP32 can not run code from SD card like it can run the code from SPI flash. Also the ROM bootloader is not capable of initializing the card and loading code from it into the RAM.

Both these features are tracked as possible improvements for the next chip.

projectgus, 2017-02-16:
Closing this issue as it's fundamentally a hardware rather that software issue. But rest assured the hardware team is considering it. :)

(And thanks for the kind words!)

closed 2017-02-16 by projectgus