espressif/esp-idf #313: Partition Questions

forthlightning, 2017-02-03:
I was looking through the partition api and came across some questions about the partition interface.

1. What is PHY partition and why would I need it? In the source it looks like it's linked to the ethernet library. 

2. What is "otadata" and how is that different from an explicitly ota partition?

3. What does it mean to be "aligned to 0x10000" ? is that multiples of 64k?

4. Why is syntax 1M ok in the example partitions.csv? Shouldnt all addresses be in the form 0x0000000etc?

5. Is necessary size for factory partition solely determined by the size of the user application?

negativekelvin, 2017-02-03:
1. It is for WiFi RF phy calibration data
2. otadata stores info about which ota partitions are active and selected
3. Yes, so 64k block erase can be used
4. Because gen_esp32part.py tool supports those notation formats for convenience

http://esp-idf.readthedocs.io/en/latest/partition-tables.html

You may consider using esp32.com official forum for tech support that is not a source code issue

projectgus, 2017-02-17:
To add to the answer for (1), unless the configuration option Component Config -> PHY -> "Use a partition to store PHY init data" is set (defaults to unset) then this partition is unused.

I also realised a lot of these details are missing from the partition table docs. An update is on the way.

closed 2017-04-05 by igrr