espressif/esp-idf #314: Hit upper limit for BLE attributes

collin80, 2017-02-03:
I'm trying to do an admittedly abnormal thing - I want to create 62 characteristics with descriptions, values, and presentation bytes. All together there's 248 attributes. I had things working before going all of the way to 62 characteristics so it appears I've hit the upper limit of something somewhere. I've tried to increase everything I could think of - the queue size, all the stack sizes, I set ESP_GATT_ATTR_HANDLE_MAX to       400. 

I added some debugging print statements and here is the relevant section of debugging log:

E (983) BT: app_main init bluetooth

E (983) BT: Startup BTU

E (983) BT: GATTS_CreateService:  handle of service handle1
E (1003) BT: bta_dm_co_ble_load_local_keys: func not ported

E (1013) BT: app_main 528

E (1013) BT: EVT 0, gatts if 4

E (1013) BT: event = 0

E (1013) BT: gatts_profile_event_handler 403

E (1013) BT: gatts_profile_event_handler 405

E (1023) BT: gatts_profile_event_handler 408

E (1023) BT: gatts_profile_event_handler 411

E (1033) BT: btc_to_bta_adv_data - In 128-UUID_data
E (1033) BT: Partial data write into ADV
E (1103) BT: EVT 22, gatts if 4

E (1103) BT: event = 16

E (1103) BT: The number handle =248

E (1103) BT: Number of attributes 248
E (1113) BT: Handle 0 is 40
E (1113) BT: Handle 1 is 41
E (1113) BT: Handle 2 is 42
E (1123) BT: Handle 3 is 43
E (1123) BT: Handle 4 is 44
E (1123) BT: Handle 5 is 45
E (1133) BT: Handle 6 is 46
E (1133) BT: Handle 7 is 47
E (1143) BT: Handle 8 is 48
E (1143) BT: Handle 9 is 49
E (1143) BT: Handle 10 is 50
E (1153) BT: Handle 11 is 51
E (1153) BT: Handle 12 is 52
E (1153) BT: Handle 13 is 53
E (1163) BT: Handle 14 is 54
E (1163) BT: Handle 15 is 55
E (1163) BT: Handle 16 is 56
E (1173) BT: Handle 17 is 57
E (1173) BT: Handle 18 is 58
E (1173) BT: Handle 19 is 59
E (1183) BT: Handle 20 is 60
E (1183) BT: Handle 21 is 61
E (1183) BT: Handle 22 is 62
E (1193) BT: Handle 23 is 63
E (1193) BT: Handle 24 is 64
E (1203) BT: Handle 25 is 65
E (1203) BT: Handle 26 is 66
E (1203) BT: Handle 27 is 67
E (1213) BT: Handle 28 is 68
E (1213) BT: Handle 29 is 69
E (1213) BT: Handle 30 is 70
E (1223) BT: Handle 31 is 71
E (1223) BT: Handle 32 is 72
E (1223) BT: Handle 33 is 73
E (1233) BT: Handle 34 is 74
E (1233) BT: Handle 35 is 75
[... 199 lines cut ...]
E (1953) BT: Handle 235 is 0
E (1953) BT: Handle 236 is 0
E (1963) BT: Handle 237 is 0
E (1963) BT: Handle 238 is 0
E (1963) BT: Handle 239 is 0
E (1973) BT: Handle 240 is 0
E (1973) BT: Handle 241 is 0
E (1983) BT: Handle 242 is 0
E (1983) BT: Handle 243 is 0
E (1983) BT: Handle 244 is 0
E (1993) BT: Handle 245 is 0
E (1993) BT: Handle 246 is 0
E (1993) BT: Handle 247 is 0
E (2003) BT: Attempted to start service with table ID 40
E (2003) BT: GAP_EVT, event 0

E (2013) BT: osi_alarm_set chg period error

***ERROR*** A stack overflow in task hciHostT has been detected.
abort() was called at PC 0x40086090
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x400f1dc8:0x3ffdafb0 0x400f1dd7:0x3ffdafd0 0x40084708:0x3ffdaff0 0x40085be7:0x3ffdb010 0x40085c78:0x3ffdb030 0x4011cb24:0x3ffdb070

You can see that it quit assigning handles at 124 which corresponds to about the 31st characteristic. This leads me to believe that it only wants to register 30 characteristics. Additionally, there was a hciHostT stack overflow but I didn't see that stack and besides it obviously quit giving me handles about half way through so that couldn't be the only issue I don't think. Can you point me toward where I might find more information about how to expand support so that it works? Or, would you suggest I create multiple services instead of trying to put 62 characteristics into one service?

collin80, 2017-02-03:
I found the HCI stack sizes in the bluetooth Thread.h file and upping the values for both HCI stacks there makes my program not crash any longer but it still only registers the first 30 or so characteristics. After that all the rest of the attributes that would have been put on subsequent characteristics instead get put on the last properly registered characteristic. Something is limiting the number of characteristics I can register but not necessarily the number of attributes I can register.

negativekelvin, 2017-02-03:
Seems like some of the values involved are 8-bit so 256 max

`The number handle =248`

`E (1563) BT: Handle 124 is 0`

Looks like it stops at exactly half

Other than that, not sure.  Have you tried using the table api? If it was me I may consider interfacing to the Bluetooth stack at a lower level with that many characteristics.

collin80, 2017-02-03:
That was with the attribute table api. As mentioned, it is able to work
fine if i just try to use 30 or less characteristics. Maybe I will have to
interface at a lower level. I'm not aware of any BLE standard related
reason why it should not work.

negativekelvin, 2017-02-03:
Did you check https://github.com/espressif/esp-idf/blob/master/components/bt/bluedroid/include/bt_target.h there are a lot of limits defined there

collin80, 2017-02-05:
As it turns out, the second half of the handles being zero seems to apparently be correct behavior in my case. I'm adding four things per characteristic - a characteristic definition start, the value, the description, then the presentation bytes. So, four things are required to set up the characteristic but it appears only two of them get handles. The other two don't. So, there really are only half as many handles as input attributes. That part seems normal. Testing indicates that the 37th attribute is where it goes south. All attributes are registered but they get weird after the 37th attribute, not the 30th as I previously thought. 

I didn't see anything in bt_target that jumps out but I'll look a few more times to see if I can find anything that seems to relate to this. At this point the program remains stable and operable. It's just that nothing past characteristic number 37 makes any sense.

Bottom line, the issue isn't solved yet but I felt that I should clarify for anyone else that not all attributes get handles so that isn't necessarily an issue.

Another related issue is that it appears that the last attribute gets thrown away even when I only have a few characteristics. What I'll see is that the presentation bytes are not set on the last characteristic. Otherwise everything comes through. I'll check the example I had copied from to see if it has the issue too. If it does I'll raise another issue for that one because it is a separate thing.

FayeY, 2017-05-02:
Hi collin80, is this problem still unsolved?

collin80, 2017-05-03:
I'm trying to figure that out. I had left the project aside while the IDF was worked on. Now I'm trying to come back and see if it works but my board isn't cooperating. I will update once I have more details. I just wanted to write to say that I'm not ignoring this issue and don't close it just yet.

collin80, 2017-05-04:
I've got the board to work and I can upload my program and try it out. Sadly, a default installation of esp-idf still causes the program to die if I try to use all 62 characteristics. I have the 62 spread out on three service IDs. This won't work. If I comment out a bunch of characteristics in my set up for the table call then I can set up the first service with its 14 attached characteristics. This works fine and I can connect to the module and query the service and its characteristics. I'm back to stock here so I haven't tried messing with any buffer sizes in the headers of the IDF yet. But, the answer is - it's still broken if you try to use the newest code in the master branch of esp-idf. Here is the relevant chunk of the output as it dies:

BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:03:84:4A
NVDS MAGIC FAILED
RF Init OK with coex
I (419) phy: phy_version: 350, Mar 22 2017, 15:02:06, 0, 0
Enable Classic BT
Enable Low Energy
I (419) GATT_SERVER: app_main init bluetooth

E (419) BT: Startup BTU

E (419) BT: L2CA_RegisterFixedChannel()  CID: 0x0004, 0x3ffd7320
E (429) BT: GATTS_CreateService:  handle of service handle1
E (439) BT: L2CA_RegisterFixedChannel()  CID: 0x0006, 0x3ffd7300
E (459) BT: #####################################num_lm_ble_bufs = 10
E (459) BT: bta_dm_co_ble_load_local_keys: func not ported

I (469) GATT_SERVER: app_main 579

I (469) GATT_SERVER: app_main 584

E (469) GATT_SERVER: GAP_EVT, event 14

I (469) GATT_SERVER: EVT 0, gatts if 3

E (479) GATT_SERVER: event = 0

I (479) GATT_SERVER: gatts_profile_event_handler 435

I (489) GATT_SERVER: gatts_profile_event_handler 437

I (489) GATT_SERVER: gatts_profile_event_handler 440

I (499) GATT_SERVER: gatts_profile_event_handler 443

E (499) GATT_SERVER: GAP_EVT, event 15

E (529) BT: Btc Post failed

E (529) BT: btc_gatts_inter_cb btc_transfer_context failed

E (539) BT: Btc Post failed

E (539) BT: btc_gatts_inter_cb btc_transfer_context failed

E (549) BT: Btc Post failed

E (549) BT: btc_gatts_inter_cb btc_transfer_context failed

E (559) BT: Btc Post failed

E (559) BT: btc_gatts_inter_cb btc_transfer_context failed

E (569) BT: Btc Post failed

E (569) BT: btc_gatts_inter_cb btc_transfer_context failed

E (579) BT: Btc Post failed

E (579) BT: btc_gatts_inter_cb btc_transfer_context failed

E (589) BT: Btc Post failed

E (589) BT: btc_gatts_inter_cb btc_transfer_context failed

E (599) BT: Btc Post failed

E (599) BT: btc_gatts_inter_cb btc_transfer_context failed

E (609) BT: Btc Post failed

E (609) BT: btc_gatts_inter_cb btc_transfer_context failed

E (619) BT: Btc Post failed

E (619) BT: btc_gatts_inter_cb btc_transfer_context failed

E (629) BT: Btc Post failed

E (629) BT: btc_gatts_inter_cb btc_transfer_context failed

E (639) BT: Btc Post failed

E (639) BT: btc_gatts_inter_cb btc_transfer_context failed

E (649) BT: Btc Post failed

E (649) BT: btc_gatts_inter_cb btc_transfer_context failed

E (659) BT: Btc Post failed

E (659) BT: btc_gatts_inter_cb btc_transfer_context failed

E (669) BT: Btc Post failed

E (669) BT: btc_gatts_inter_cb btc_transfer_context failed

E (679) BT: Btc Post failed

E (679) BT: btc_gatts_inter_cb btc_transfer_context failed

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400863aa  PS      : 0x00060633  A0      : 0x80085943  A1      : 0x3ffd2f50  
0x400863aa: vListInsert at /home/collin/esp32/esp-idf/components/freertos/./list.c:188 (discriminator 1)

A2      : 0x3ffd009c  A3      : 0x3ffd1904  A4      : 0x00060620  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00060023  A8      : 0xa5a5a5a5  A9      : 0x3ffd00a4  
A10     : 0x3ffc13f0  A11     : 0x00000006  A12     : 0x00060623  A13     : 0x00000018  
A14     : 0x3ffd7f34  A15     : 0x3ffd2ff0  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xa5a5a5a5  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x400863aa:0x3ffd2f50 0x40085943:0x3ffd2f70 0x40084033:0x3ffd2f90 0x400e71fa:0x3ffd2fd0 0x400fc999:0x3ffd3000 0x400fb5ea:0x3ffd3020 0x400e2365:0x3ffd3050 0x400e2721:0x3ffd3100 0x400def1c:0x3ffd33a0
0x400863aa: vListInsert at /home/collin/esp32/esp-idf/components/freertos/./list.c:188 (discriminator 1)

0x40085943: xPortGetCoreID at /home/collin/esp32/esp-idf/components/freertos/./tasks.c:4415
 (inlined by) vTaskPlaceOnEventList at /home/collin/esp32/esp-idf/components/freertos/./tasks.c:2852

0x40084033: xQueueGenericSend at /home/collin/esp32/esp-idf/components/freertos/./queue.c:2034

0x400e71fa: btu_task_post at /home/collin/esp32/esp-idf/components/bt/bluedroid/stack/btu/btu_task.c:636

0x400fc999: bta_sys_sendmsg at /home/collin/esp32/esp-idf/components/bt/bluedroid/bta/sys/bta_sys_main.c:644

0x400fb5ea: BTA_GATTS_AddCharacteristic at /home/collin/esp32/esp-idf/components/bt/bluedroid/bta/gatt/bta_gatts_api.c:554

0x400e2365: btc_gatts_act_create_attr_tab at /home/collin/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c:373

0x400e2721: btc_gatts_call_handler at /home/collin/esp32/esp-idf/components/bt/bluedroid/btc/profile/std/gatt/btc_gatts.c:508

0x400def1c: btc_task at /home/collin/esp32/esp-idf/components/bt/bluedroid/btc/core/btc_task.c:83

collin80, 2017-06-25:
I figured out how to make it work. I split my characteristics into three services and I now create a table for each service and send everything in batches, one batch per service. This allows me to register all of the characteristics I need - so long as I do not put more than 24 characteristics into a single service. Seemingly I can have as many services as I want but not more than 24 characteristics per service. This seems to be related to the number of attributes I register for each characteristic. Debugging shows that 24 characteristics leaves me with 97 attributes total. The 25th would make it exceed 100 and this causes it to crash. So, I assume there is somewhere where an upper limit of 100 is set. I'm far beyond too frustrated with the whole thing at this point to track that down. For now I'll just use a lot of services, each of which has no more than 24 characteristics. This seems to be working with the esp-idf current to the most recent commit as of this message. So, it works but I'm not going to close this issue myself because I feel that there is still an issue with hitting the upper limit of attributes. But, perhaps you feel that 24 characteristics per service is plenty enough. In that case feel free to override me and close it anyway.

Yulong-espressif, 2017-09-27:
@collin80 In the new SDK, you can add the characteristics as much as you want. So can you update your SDK and try it again? If any other question, don't hesitate to let me know, thanks.

closed 2017-10-10 by FayeY