espressif/esp-idf #315: Make flash doesn't work -> previous declaration of ‘kconf_id_lookup’ was here

drony, 2017-02-03:
make[1]: Entering directory '/c/esp32/esp-idf/tools/kconfig'
cc  -I/usr/include/ncursesw -DCURSES_LOC="<curses.h>"  -DNCURSES_WIDECHAR=1 -DLOCALE   -c -o zconf.tab.o zconf.tab.c
In file included from zconf.tab.c:212:0:
zconf.hash.c:166:1: error: conflicting types for ‘kconf_id_lookup’
 kconf_id_lookup (register const char *str, register size_t len)
 ^~~~~~~~~~~~~~~
zconf.hash.c:33:31: note: previous declaration of ‘kconf_id_lookup’ was here
 static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
                               ^~~~~~~~~~~~~~~
make[1]: *** [<builtin>: zconf.tab.o] Error 1
make[1]: Leaving directory '/c/esp32/esp-idf/tools/kconfig'
make: *** No rule to make target '/c/esp32/esp-idf/tools/kconfig/conf', needed by '/c/esp32/esp-idf-template/build/include/config/auto.conf'.  Stop.

negativekelvin, 2017-02-04:
https://github.com/espressif/esp-idf/issues/251

projectgus, 2017-02-15:
gperf 3.1 changed the type signature for this function, for 64-bit compatibility reasons.

A permanent fix is pending and will be in master soon.

In the meantime you can delete this entire line from zconf.gperf and it will fix the problem:
https://github.com/espressif/esp-idf/blob/master/tools/kconfig/zconf.gperf#L12

(Editing zconf.hash.c will work until you `make clean`, because this file is generated from zconf.gperf.)

closed 2017-02-15 by igrr