espressif/esp-idf #316: promiscuous callback broken

RouquinBlanc, 2017-02-04:
Sniffing wifi frames (as in this [gist](https://gist.github.com/brainstorm/24e843ae0295ee1e41dff47c5b43a02c)) was broken recently by an update of libpp.a in https://github.com/espressif/esp-idf/commit/03e3b137bfcc88d521f56be4d0a26c10256215af - verified via bisect. The callback is now called only one or twice and then never again... a simple rollback of libpp.a to [prior version](https://github.com/espressif/esp32-wifi-lib/blob/ddc4b0cbc59c2c63ba1f0e02a9cd4972c083782c/libpp.a) allows to work around the issue.

tucanae47, 2017-02-10:
got a question, do you know is this mode is the same as monitor mode?, (first try with this), so far i understand the only difference between this monitor and promiscuous modes is that in promiscuous mode, you need to be connected to an AP, but in esp8266 and esp32 gists, its not needed or at least i haven't seen the need of ssid, pwd config. 

other question is are you able to read packets from devices not connected to any network?

igrr, 2017-04-05:
Sorry that this issue wasn't updated, the fix for the monitor mode was merged into master somewhere in February.

closed 2017-04-05 by igrr