espressif/esp-idf #317: esp-idf needs libncursesw.so.6

feelfreelinux, 2017-02-04:
When esp-idf always needed ncurses in version 5, now when i try to run `make menuconfig` I'm getting: 
`libncursesw.so.6: cannot open shared object file: No such file or directory`

Shouldn't esp-idf use ncurses in version 5?

projectgus, 2017-02-07:
Without more information it's hard to tell where the problem is.

* What OS (Linux distro?) are you using?

* If you run `xtensa-esp32-elf-gcc --version`, does this work?

* If the above line works, the problem is probably the `mconf` executable which is compiled on your system. Try `make clean` then `make menuconfig` and see if it works after being rebuilt (maybe your ncurses version changed?) If that doesn't work, can you please post the full output of `make clean; make menuconfig V=1`? Thanks.

projectgus, 2017-02-16:
@feelfreelinux did you have any luck progressing this?

feelfreelinux, 2017-02-18:
@projectgus Oh sorry, I forgot about this. Problem no longer occur, propably fixed when I was updating my system. Thanks!

closed 2017-02-18 by feelfreelinux