espressif/esp-idf #318: reading from uart causes crash

kglowacki, 2017-02-04:
Hi, I'm trying to modify uart example to read data from my sensor (only RX on pin 36, no control flow). When I run the code, it always crashes (stack below) and reboots chip two or three times and after then it settles down and starts reading data correctly.

It seems to be related somehow to ticks_to_wait in uart_read_bytes? (longer timeouts causes more frequent crashes). Any idea what this is caused by?

```
//value below 128 causes crash!
#define PM_UART_BUF_SIZE (128)

#define PM_UART_NUM UART_NUM_1

#define PM_UART_RXD  (34) //we care only about RX
#define PM_UART_TXD  (4)
#define PM_UART_RTS  (18)
#define PM_UART_CTS  (19)

static void pmsInitUart(void) {
	//configure UART
    uart_config_t uart_config = {
        .baud_rate = 9600,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,//UART_HW_FLOWCTRL_CTS_RTS,
        .rx_flow_ctrl_thresh = 122,
    };
    uart_param_config(PM_UART_NUM, &uart_config);
    uart_set_pin(PM_UART_NUM, PM_UART_TXD, PM_UART_RXD, PM_UART_RTS, PM_UART_CTS);
    //Install UART driver( We don't need an event queue here)
    //In this example we don't even use a buffer for sending data.
    uart_driver_install(PM_UART_NUM, PM_UART_BUF_SIZE * 2, 0, 0, NULL, 0);
}

//task
static void pmsUartRead()
{
	pmsInitUart();
    uint8_t* data = (uint8_t*) malloc(32);
    while(1) {
        int len = uart_read_bytes(PM_UART_NUM, data, 32, 10 / portTICK_RATE_MS);

        //check 'header'
        if (len >= 24 && data[0]==0x42 && data[1]==0x4d) {
        	ESP_LOGI(TAG, "got frame of %d bytes", len);
        } else if (len > 0) {
        	ESP_LOGI(TAG, "invalid frame of %d", len);
        }
    }
}

void pmsInit(void) {
    xTaskCreate(pmsUartRead, "pmsUartRead", 1024, NULL, 10, NULL);
}

```

```
/Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./queue.c:1445 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40083359
Guru Meditation Error: Core  0 panic'ed (abort)

Backtrace: 0x40008155:0x3ffb7cc0 0x40007d16:0x3ffb7ce0 0x400834ac:0x3ffb7d20 0x40081611:0x3ffb7d40 0x400816e8:0x3ffb7d70 0x4000be38:0x3ffb7d90 0x40059374:0x3ffb7db0 0x40058bc2:0x3ffb7dd0 0x400e2750:0x3ffb7e00 0x400e7f19:0x3ffb7e30 0x400e805c:0x3ffb8140 0x40081520:0x3ffb8170 0x400efb78:0x3ffb81c0

(gdb) list *0x40083359
0x40083359 is in xQueueGenericReceive (/Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./queue.c:1445).
1440	TimeOut_t xTimeOut;
1441	int8_t *pcOriginalReadPosition;
1442	Queue_t * const pxQueue = ( Queue_t * ) xQueue;
1443
1444		configASSERT( pxQueue );
1445		configASSERT( !( ( pvBuffer == NULL ) && ( pxQueue->uxItemSize != ( UBaseType_t ) 0U ) ) );
1446		#if ( ( INCLUDE_xTaskGetSchedulerState == 1 ) || ( configUSE_TIMERS == 1 ) )
1447		{
1448			configASSERT( !( ( xTaskGetSchedulerState() == taskSCHEDULER_SUSPENDED ) && ( xTicksToWait != 0 ) ) );
1449		}
```

kglowacki, 2017-02-04:
I think it may be caused by low size of stack allocated for the task (1024 bytes), increasing it twice makes it more stable.

ggsubs, 2017-02-05:
Take a look at https://github.com/espressif/esp-idf/issues/93 
In my case the fix was to disabled the interrupt on framing errors.

projectgus, 2017-02-07:
If you run 'bt' (backtrace) in gdb, what do you get?

projectgus, 2017-02-16:
@kglowacki , have you had time to make any progress with this?

kglowacki, 2017-02-19:
finnally had time to get back to this, but in the meantime I've updated SDK and the problem disappeared... now it works fine even with task stack reduced to 512 bytes. happy days.

closed 2017-02-19 by kglowacki

ndunello, 2018-10-05:
Hi, I have same (?) problem

```
#define ECHO_TEST_TXD  (GPIO_NUM_10)
#define ECHO_TEST_RXD  (GPIO_NUM_9)
#define ECHO_TEST_RTS  (UART_PIN_NO_CHANGE)
#define ECHO_TEST_CTS  (UART_PIN_NO_CHANGE)

#define BUF_SIZE (128)

static void echo_task()
{
uart_config_t uart_config = {
        .baud_rate = 115200,
        .data_bits = UART_DATA_8_BITS,
        .parity    = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE
    };
    uart_param_config(UART_NUM_1, &uart_config);
    uart_set_pin(UART_NUM_1, ECHO_TEST_TXD, ECHO_TEST_RXD, ECHO_TEST_RTS, ECHO_TEST_CTS);
    uart_driver_install(UART_NUM_1, BUF_SIZE * 2, 0, 0, NULL, 0);

    uint8_t *data = (uint8_t *) malloc(BUF_SIZE);

    while (1) {
        // Read data from the UART
        int len = uart_read_bytes(UART_NUM_1, data, BUF_SIZE, 20 / portTICK_RATE_MS);
        if(len > 0)
            ESP_LOGI("UART TASK", "Received %d bytes\n", len);
        // Write data back to the UART
        uart_write_bytes(UART_NUM_1, (const char *) data, len);
    }
}

void app_main()
{
    ESP_LOGI("APP MAIN","TaskCreate: uart_echo_task\n");
    xTaskCreate(echo_task, "uart_echo_task", 1024, NULL, 10, NULL);
}
```

When send data via uart application crash!
The monitor output is:

```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5736
load:0x40078000,len:9020
load:0x40080000,len:6064
0x40080000: _iram_start at /home/anto/projects/esp32/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080330
0x40080330: _KernelExceptionVector at ??:?

I (28) boot: ESP-IDF v3.1 2nd stage bootloader
I (28) boot: compile time 09:43:05
I (28) boot: Enabling RNG early entropy source...
I (33) boot: SPI Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 4MB
I (45) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x092b4 ( 37556) map
I (104) esp_image: segment 1: paddr=0x000192dc vaddr=0x3ffb0000 size=0x022e0 (  8928) load
I (108) esp_image: segment 2: paddr=0x0001b5c4 vaddr=0x3ffb22e0 size=0x00000 (     0) load
I (111) esp_image: segment 3: paddr=0x0001b5cc vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _iram_start at /home/anto/projects/esp32/esp-idf/components/freertos/xtensa_vectors.S:1685

I (121) esp_image: segment 4: paddr=0x0001b9d4 vaddr=0x40080400 size=0x0463c ( 17980) load
I (137) esp_image: segment 5: paddr=0x00020018 vaddr=0x400d0018 size=0x138bc ( 80060) map
0x400d0018: _flash_cache_start at ??:?

I (166) esp_image: segment 6: paddr=0x000338dc vaddr=0x40084a3c size=0x04f9c ( 20380) load
0x40084a3c: prvInitialiseNewTask at /home/anto/projects/esp32/esp-idf/components/freertos/tasks.c:4609

I (175) esp_image: segment 7: paddr=0x00038880 vaddr=0x400c0000 size=0x00000 (     0) load
I (175) esp_image: segment 8: paddr=0x00038888 vaddr=0x50000000 size=0x00000 (     0) load
I (188) boot: Loaded app from partition at offset 0x10000
I (188) boot: Disabling RNG early entropy source...
I (193) cpu_start: Pro cpu up.
I (197) cpu_start: Starting app cpu, entry point is 0x40080ee8
0x40080ee8: call_start_cpu1 at /home/anto/projects/esp32/esp-idf/components/esp32/cpu_start.c:225

I (0) cpu_start: App cpu up.
I (208) heap_init: Initializing. RAM available for dynamic allocation:
I (214) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (220) heap_init: At 3FFB3330 len 0002CCD0 (179 KiB): DRAM
I (227) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (233) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (239) heap_init: At 400899D8 len 00016628 (89 KiB): IRAM
I (246) cpu_start: Pro cpu start user code
I (264) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (265) APP MAIN: TaskCreate: uart_echo_task

I (39925) UART TASK: Received 1 bytes

/home/anto/projects/esp32/esp-idf/components/freertos/ringbuf.c:1038 (xRingbufferGetMaxItemSize)- assert failed!
abort() was called at PC 0x400870e7 on core 0
0x400870e7: xRingbufferGetMaxItemSize at /home/anto/projects/esp32/esp-idf/components/freertos/ringbuf.c:1038 (discriminator 1)

Backtrace: 0x40087d74:0x3ffb6f60 0x40087f4b:0x3ffb6f80 0x400870e7:0x3ffb6fa0 0x400e1bb5:0x3ffb6fc0 0x400e23dd:0x3ffb7000 0x400d2399:0x3ffb7030
0x40087d74: invoke_abort at /home/anto/projects/esp32/esp-idf/components/esp32/panic.c:649

0x40087f4b: abort at /home/anto/projects/esp32/esp-idf/components/esp32/panic.c:649

0x400870e7: xRingbufferGetMaxItemSize at /home/anto/projects/esp32/esp-idf/components/freertos/ringbuf.c:1038 (discriminator 1)

0x400e1bb5: uart_tx_all at /home/anto/projects/esp32/esp-idf/components/driver/uart.c:1241

0x400e23dd: uart_write_bytes at /home/anto/projects/esp32/esp-idf/components/driver/uart.c:1241

0x400d2399: echo_task at /home/anto/projects/esp32/workspace/uart_echo/main/uart_echo_example_main.c:46

Rebooting...
```

Note that if i remove 
`ESP_LOGI("UART TASK", "Received %d bytes\n", len);`
application don't crash