espressif/esp-idf #319: ioctl(socket, FIONREAD, &result) returns -1/Errno 88;

feelfreelinux, 2017-02-05:
Hi! I'm trying to make not blocking socket connection. I'm using ioctl(socket, FIONREAD, &result), to test is data here. But when I try this code, ioctl returns -1/errno 88. Code http://pastebin.com/vzcUpdtP
Using newest esp-idf, this issue looks related: https://github.com/espressif/arduino-esp32/issues/85

yschaeff, 2017-02-05:
> espressif/arduino-esp32#85

Your errno means this:
	#define ENOTSOCK    88  /* Socket operation on non-socket */

is your 'socket' really an (opened) socket and not an other type of file
descriptor?

Also, you can use strerror(errno) to get the above textual description.
Makes debugging easier. :)

//Yuri

feelfreelinux, 2017-02-05:
@yschaeff: Yep, I'm sure that socket is open - I received messages before, etc. I'm using C++, if this matter. @me-no-dev How do you got this working? Will upload result of strerror( tomorrow. Thanks!

igrr, 2017-02-05:
Could you please check the value of `CONFIG_LWIP_SO_RCVBUF` in your sdkconfig file? According to the help for this option,

> Enabling this option allows checking for available data on a netconn.

me-no-dev, 2017-02-05:
arduino enables ```CONFIG_LWIP_SO_RCVBUF``` when included as component

feelfreelinux, 2017-02-05:
@igrr @me-no-dev Thanks! Thanks for help, this fixes this issue. Will create PR adding hint for troubleshooting

closed 2017-02-05 by feelfreelinux