espressif/esp-idf #321: espcoredump.py CORE DUMP TO UART (no GDB debug session!)

pcbreflux, 2017-02-07:
Using this sdkconfig while dumping to UART (no GDB debug session!):

CONFIG_TRACEMEM_RESERVE_DRAM=0x0
CONFIG_ESP32_ENABLE_COREDUMP_TO_UART=y
CONFIG_ESP32_ENABLE_COREDUMP=y
CONFIG_ESP32_CORE_DUMP_UART_DELAY=0
CONFIG_ESP32_CORE_DUMP_LOG_LEVEL=1

Output from espcoredump.py:
python $IDF_PATH/components/espcoredump/espcoredump.py info_corefile -t b64 -c core.txt build/app.elf
espcoredump.py v0.1-dev

================== CURRENT THREAD REGISTERS ===================
/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: value_of_register_lazy: Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/home/ivan/e/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: value_of_register_lazy: Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
GDB/MI command failed (None / None)!
==================== CURRENT THREAD STACK =====================
Traceback (most recent call last):
  File "/home/pcbreflux/esp/esp-idf/components/espcoredump/espcoredump.py", line 1107, in <module>
    main()
  File "/home/pcbreflux/esp/esp-idf/components/espcoredump/espcoredump.py", line 1102, in main
    operation_func(args)
  File "/home/pcbreflux/esp/esp-idf/components/espcoredump/espcoredump.py", line 1010, in info_corefile
    p.stdin.write("-interpreter-exec console \"bt\"\n")
IOError: [Errno 32] Broken pipe

pcbreflux, 2017-02-07:
Additional info: using xtensa 5.2.0 with esp-idf commit c78aa138b061407c555e8913c0e46ec81a658eb3 from 20170206.

pcbreflux, 2017-02-09:
Even if it's nice to know where ivan has install his gdb an what is the version, it don't help to debug your own core dump ;).

igrr, 2017-02-13:
Could you please upload your elf file and the core dump info, and share a link? Thanks.

pcbreflux, 2017-02-14:
Hi, 

you can find all source code here 
https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_mqtt_secure_subscribe
The dump:
https://github.com/pcbreflux/espressif/tree/master/esp32/app/ESP32_mqtt_secure_subscribe/core.txt.issue_321
The ELF
https://github.com/pcbreflux/espressif/blob/master/esp32/app/ESP32_mqtt_secure_subscribe/mqtt-secure-publish.elf

Thanks

kglowacki, 2017-02-19:
same problem here. after updating SDK espcoredump stopped working.
I've tried a few different dumps with the same result.

MacOS, Python 2.7.10 (default, Oct 23 2015, 19:19:21)
[GCC 4.2.1 Compatible Apple LLVM 7.0.0 (clang-700.0.59.5)]

gerekon, 2017-02-20:
@pcbreflux, I had a look at your base64 core dump file. There are 10 tasks running. Espcoredump has problem with the first (crashed) one. The reason of failure is that return address of crashed task is 0x40000000. This is abnormal address for program execution. Your program should never get there.
In any case this situation should be handled by core dump script in more user-friendly way and IT will be fixed.
Thanks.

gerekon, 2017-02-20:
@kglowacki, Could you please upload your elf file and the core dump info, and share a link?
> after updating SDK espcoredump stopped working.

Are you sure that espcoredump worked with the same program ELF binary before?
The problem which is described by @pcbreflux exists in all versions of espcoredump script.

kglowacki, 2017-02-20:
[dump.zip](https://github.com/espressif/esp-idf/files/787101/dump.zip)

gerekon, 2017-02-21:
@kglowacki, Thanks for the files. You have a bit different situation. There 9 tasks and one fails inside ROM's memcpy. To unwind functions from ROM you need ROM elf file which is not provided with IDF as of now. We plan to provide this file in order to handle such situations and for other things  such as source level debugging of ROM functions etc. For now if you encounter such problems with core dumps I recommend you to use 'dbg_corefile' command of espcoredump. It runs GDB debug session with you core file and you can use standard GDB commands. So even ROM elf is not present it can give you more info as compared to failed 'info_corefile'. Below is output of that command for your files. In GDB session I printed out the list of threads using 'info threads'. As you can see your program fails in 'main_task' at address 0x4000c2ff. According to ESP32 LD script this address is inside ROM's memcpy. Please, see 'components/esp32/ld/esp32.rom.ld'. 

`./components/espcoredump/espcoredump.py dbg_corefile -t b64 -c /media/sf_Temp/vb/dump/core.dat /media/sf_Temp/vb/dump/sensor-esp32.elf
espcoredump.py v0.1-dev
GNU gdb (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /media/sf_Temp/vb/dump/sensor-esp32.elf...done.
[New <main task>]
[New process 1]
[New process 2]
[New process 3]
[New process 4]
[New process 5]
[New process 6]
[New process 7]
[New process 8]
[New process 9]
[New process 10]
#-1 0x4000c2ff in ?? ()
[Current thread is 1 (<main task>)]
(gdb) info threads
  Id   Target Id         Frame 
  11   process 10        0x40083b03 in xQueueGenericReceive (xQueue=0x0, pvBuffer=0x0, 
    xTicksToWait=<unavailable>, xJustPeeking=0)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./queue.c:1594
  10   process 9         0x4008522a in prvProcessTimerOrBlockTask (xNextExpireTime=<optimized out>, 
    xListWasEmpty=<optimized out>)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./timers.c:487
  9    process 8         0x40083b03 in xQueueGenericReceive (xQueue=0x3ffbd120, pvBuffer=0x3ffb9040, 
    xTicksToWait=<unavailable>, xJustPeeking=0)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./queue.c:1594
  8    process 7         0x400847c4 in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./tasks.c:1407
  7    process 6         0x400847c4 in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./tasks.c:1407
  6    process 5         0x400847c4 in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./tasks.c:1407
  5    process 4         0x400847c4 in vTaskDelay (xTicksToDelay=<optimized out>)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./tasks.c:1407
  4    process 3         0x40083b03 in xQueueGenericReceive (xQueue=0x1, 
    pvBuffer=0x3ffb43ec <cmd_queue>, xTicksToWait=<unavailable>, xJustPeeking=0)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./queue.c:1594
  3    process 2         0x400d1c54 in esp_vApplicationIdleHook ()
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/esp32/./freertos_hooks.c:52
  2    process 1         0x40083b03 in xQueueGenericReceive (xQueue=0x3ffb4304 <samples_queue>, 
    pvBuffer=0x0, xTicksToWait=<unavailable>, xJustPeeking=0)
    at /Users/kris/Dev/iot/ESP32/esp-idf/components/freertos/./queue.c:1594
* 1    <main task>       0x4000c2ff in ?? ()
(gdb) 
`

pcbreflux, 2017-02-21:
@gerekon thanks for you efford and explaination now I see my major error: starting new tasks without delete the old one. thanks

gerekon, 2017-02-21:
@pcbreflux, you are welcome ). Also you can use 'dbg_corefile' command as described in my answer for @kglowacki. Also while script does not handle ROM unwinding properly that command can give you more info about the problem.

closed 2017-03-01 by pcbreflux

vonnieda, 2018-06-26:
Hi @gerekon, in Feb 2017 you said "To unwind functions from ROM you need ROM elf file which is not provided with IDF as of now. We plan to provide this file in order to handle such situations and for other things such as source level debugging of ROM functions etc.". Has there been any update on this? I have a core dump that is dying in ROM and I'd like to be able to backtrace it.

igrr, 2018-06-27:
@vonnieda No update on ROM elf file, unfortunately. We are still looking for solution.

vonnieda, 2018-06-27:
Thanks @igrr - is there any way you can think of to be able to get a backtrace under these conditions? I have a coredump from a user and I'm trying to figure out what is happening.