espressif/esp-idf #322: WiFi throughput

gctronic, 2017-02-07:
Hi,
in the "v2.0 Release Candidate 1" there is a note in the known issues indicating "WiFi throughput need more optimization", what is the expected throughput now?
I can get only a few Mbit/s from the module to a PC (and viceversa) through an AP.

Best regards.

FayeY, 2017-05-02:
Hi gctronic, is this problem still unsolved? Does the WiFi throughput become better?

gctronic, 2017-05-09:
Hi, yes recently I did some tests and the throughput is actually better: TCP around 10 Mbit/s, UDP around 18 Mbit/s.

bvernoux, 2017-05-11:
It will be very interesting to have official values with different use cases (esp<->esp or esp<->ap), latency Average/Min/Max in ms / Average/Min/Max MBytes/s (how many meter from in different conf esp<->esp or esp<->ap)

liuzfesp, 2017-10-10:
HI @bvernoux @gctronic  we already support iperf example in latest IDF master branch, I think you can run the demo to get the result. BTW, my latest result with the iperf example is TCP TX 53M, TCP RX 36M, UDP TX 60M, UDP RX 70M

bvernoux, 2017-10-10:
Thanks for the update and ipref example the improvements are very impressive I will test it on official esp-idf v3.0
I'm also very interested to see the low level API to send raw frame over WiFi (with or without any AP connection), a bit like what seems used in espnow sample for ESP32 but it is an other subject (the aim is to have maximum TX speed on raw frame mainly for debug/traces/telemetry... purposes)

liuzfesp, 2017-10-11:
HI @bvernoux currently we don't have demo to test the throughput with RAW API. But we have throughput test on 8689 which has same WiFi chip, the difference the TCPIP stack and WiFi stack is on different CPU. The maximum UDP throughput is about 130M, almost reaches the max theoretical HT40 WiFi throughput.

closed 2017-11-21 by FayeY

abcdcadc, 2017-12-05:
Hi,
I use iperl on the newest version and the speed is around 1 Mbit/sec. it really is so slow. is there any way to speed up, unless, what is the highest speed of esp32

regard and thanks

liuzfesp, 2017-12-05:
Hi @gctronic recently we have changed the menuconfig, but the iperf example is not changed accordingly, we update the iperf example after we finish the throughput optimization this month.