espressif/esp-idf #323: Clock stretching feature for I2C slave

devsaurus, 2017-02-07:
Related to #189.
While working on slave support for the NodeMCU firmware (WIP at https://github.com/devsaurus/nodemcu-firmware/blob/dev-esp32-i2c_slave/components/modules/i2c_hw_slave.c), I observed that the ESP32 slave responds with invalid data to the master in case the application didn't provide data to the send buffer in time (via `i2c_slave_write_buffer()`).

Does the I2C slave support clock stretching and how can it be enabled to delay a transfer? Specifically for the two cases:
- master reads from slave but slave's TX FIFO is empty
- master writes to slave but slave's RX FIFO is full

Spritetm, 2017-02-21:
Huh, according to what I can find, the I2C hardware should automatically take care of this. I'll ask the hw dev.

FayeY, 2017-05-27:
Hi, @devsaurus , I2C does not support HS mode (cannot clock stretching). And if the master allowed frequency and slave allowed frequency are different, the smaller one should be used.

devsaurus, 2017-05-27:
Thanks for clarifying this point, @FayeY.

closed 2017-05-27 by devsaurus