espressif/esp-idf #325: ble crash

donny681, 2017-02-08:
I find that ble crash when I scan the peripheral.

> BTDM CONTROLLER VERSION: 010101
btip start
copy .data from 4000d890 to 3ffae6e0, len 00001830
set .bss 0x0 from 3ffb8000 to 3ffbff70, len 00007f70
BTDM ROM VERSION 0101
BD_ADDR: 24:0A:C4:01:66:A2
NVDS MAGIC FAILED
RF Init OK with coex
E (2700) BT: Startup BTU
E (2703) BT: GATTS_CreateService:  handle of service handle1
E (2719) BT: bta_dm_co_ble_load_local_keys: func not ported
E (3733) BT: btc_search_callback  BLE observe complete. Num Resp 0
E (9793) BT: btc_search_callback  BLE observe complete. Num Resp 0
E (15855) BT: btc_search_callback  BLE observe complete. Num Resp 0

you can download the demo https://github.com/petrosyanvan/esp32_sniffer ,waiting for relaying!!!!

pcbreflux, 2017-02-19:
github message for your code: petrosyanvan doesn’t have any repositories you can view. 
Have you check the core dump? Howto from Neil Kolban https://m.youtube.com/watch?v=MpD_3oVJAEs and here http://esp-idf.readthedocs.io/en/latest/core_dump.html

FayeY, 2017-05-02:
Hi donny681, is this problem still unsolved?

closed 2017-07-07 by FayeY