espressif/esp-idf #326: esp_wifi_start_scan() not performing an ACTIVE scan

ry111, 2017-02-09:
The esp32 data sheet specifies:

_"Passive or active scanning, as well as the P2P discovery procedure are performed autonomously when initiated by appropriate commands"_

However, when testing the esp_wifi_start_scan() API, we observe that a scan of a single channel takes a constant amount of time (200 msec), and a scan of all 11 channels takes 2.2 seconds - which hints the API is performing passive scanning.
An active scan should take 2 orders of magnitude less time (if implemented according to common practices, with appropriate _minCT_ and _maxCT_ durations).

Please update the API to perform active-only Wifi scan. I'd be happy to provide further information or contribute here, please let me know.

Thanks

negativekelvin, 2017-02-09:
> Active scanning is the most thorough mechanism used to find APs because it actively sends out 802.11 probes across all channels to find an AP. It requires the client to dwell on a particular channel for a set length of time, roughly 10 to 20 milliseconds (ms) depending on the vendor, waiting for the probe response.

Seems like one order of magnitude.  Do all APs respond to null probes?

ry111, 2017-02-09:
The active scan algorithm actually uses two timeouts (minCT, maxCT) to reduce the average dwell time, so on average I'd expect it to take less than 10-20 msec per channel.  It would be good to allow these timeouts to be configured by an API to the low level WIFI driver. Regardless... 1 order of magnitude is also critical for some applications.

AFAIK, APs do respond to null probes, it is mandated by the 802.11 specification. Actually, unique-SSID (not null) probes are discouraged by many for security reasons.

negativekelvin, 2017-02-10:
Yes I agree that there should be an option for either passive or active (probe based) scan and configurable channel timing windows.  I was reading a paper that said on an active network a probe request latency can be >10ms ~30% of the time.  So user will have to make some choices.

TimXia, 2017-03-15:
@ry111 The job is on going now.

negativekelvin, 2017-03-28:
@TimXia Do we have updated wifi_scan_config_t definition to use new features from https://github.com/espressif/esp32-wifi-lib/commit/b440aba0a53c60446f24ccb58d7dab5023ff9494 ?

TimXia, 2017-03-28:
@negativekelvin Not yet. The job has been done, but we haven't done enough test on it. We will push it to github immediately after test it over.

nevercast, 2017-03-29:
Thanks for the work on this, I look forward to this being rolled out.
In the mean time, is an API available to get an event for AP broadcasts so I can do my own passive scan without blocking?

Dake01, 2017-04-12:
@TimXia Hello, does the recent functionality of choice of passive/active scan could be technically included in the esp8266 API ?

TimXia, 2017-04-12:
Maybe in the next version.

Dake01, 2017-04-12:
@TimXia Thank for your quick response, do you know when it could be implemented in the API esp8266 ?

TimXia, 2017-04-12:
Sorry I haven't gotten the schedule.

FayeY, 2017-05-02:
Hi Dake01, Here is an example of [passive scan on ESP8266](http://bbs.espressif.com/viewtopic.php?f=31&p=12595)

FayeY, 2017-05-02:
Hi ry111, can we close this issue?

closed 2017-07-07 by FayeY

wasted150, 2018-06-20:
hi can anyone answer this?
can esp 8266 hear my smartphone without being connected to the wifi but with the wifi enabled, and forget  to put it in ap mode  (hotspot)