espressif/esp-idf #327: make project have problem

long585, 2017-02-09:
ESP32 IDF 2.0    xtensa-esp32-elf-gcc  5.2.0

`make[1]: Leaving directory '/home/long/esp/esp-idf-v2.0-rc1/tools/kconfig'
GENCONFIG
CC src/efuse.o
CC src/bootloader_flash.o
CC src/secure_boot_signatures.o
CC src/secure_boot.o
CC src/flash_partitions.o
CC src/esp_image_format.o
CC src/flash_encrypt.o
CC src/bootloader_random.o
AR libbootloader_support.a
CC log.o
AR liblog.a
CC spi_flash_rom_patch.o
AR libspi_flash.a
CC micro-ecc/uECC.o
AR libmicro-ecc.a
CC bootloader_start.o
AR libmain.a
LD bootloader.elf
/home/long/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc
collect2: error: ld returned 1 exit status
/home/long/esp/esp-idf-v2.0-rc1/make/project.mk:313: recipe for target '/home/long/esp/esp-idf-v2.0-rc1/examples/get-started/hello_world/build/bootloader/bootloader.elf' failed
make[1]: *** [/home/long/esp/esp-idf-v2.0-rc1/examples/get-started/hello_world/build/bootloader/bootloader.elf] Error 1
/home/long/esp/esp-idf-v2.0-rc1/components/bootloader/Makefile.projbuild:31: recipe for target '/home/long/esp/esp-idf-v2.0-rc1/examples/get-started/hello_world/build/bootloader/bootloader.bin' failed
make: *** [/home/long/esp/esp-idf-v2.0-rc1/examples/get-started/hello_world/build/bootloader/bootloader.bin] Error 2`

closed 2017-02-09 by long585

projectgus, 2017-02-09:
For anyone who finds this in a search, the solution is `git submodule update --init --recursive` which will fetch the submodule which contains librtc.

A better error message for handling this is going to be merged soon.

saurubh, 2017-05-02:
Building LoPy from Pycom. Getting this error on linkux 64 bit OS. Please resolve this. Even i have tried the method above but no luck.

Error shown below-----

skmis@skmis-VirtualBox:~/esp/esp32$ make BOARD=LOPY -j5 LORA_BAND=USE_BAND_868 TARGET=app
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build/LOPY_868/release/application.elf
/home/skmis/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc_clk
collect2: error: ld returned 1 exit status
application.mk:360: recipe for target 'build/LOPY_868/release/application.elf' failed
make: *** [build/LOPY_868/release/application.elf] Error 1

Spritetm, 2017-05-02:
Rtc_clk is not an existing library anymore: the code has been rewritten and included as open-source. I suggest you contact the LoPy people for this: they need to update their list of dependencies because their build process does not seem to use our esp-idf make process.