espressif/esp-idf #329: BLE: HCI not accepting EIR.

benjaminaigner, 2017-02-09:
Dear Espressif team,

I would like to implement the RFC7668 (6LoWPAN via BLE; also implemented in Contiki & Linux Kernel) within the IDF.

Currently I'm struggling with the Extended Inquiry Response (EIR).

Sending following data to the HCI is no problem (similar to the ble_adv example):

-) hci_reset
-) hci_set_adv_params
-) hci_set_adv_data
-) hci_adv_start

Every command is responded by the HCI 0x04 0x0E (HCI_EVENT, HCI_COMMAND_COMPLETE_EVT), as it is defined in hcidefs.h.

But if I'm adding hci_set_scan_response_data before starting advertising, the HCI responses with 0x04 0x10 -> HCI_HARDWARE_ERROR_EVT

Could you lead me anyway near the error?

Code of set scan response data:

```
//H4 header
UINT8_TO_STREAM (buf_p, HCIT_TYPE_COMMAND);
UINT16_TO_STREAM (buf_p, HCI_BLE_WRITE_SCAN_RSP_DATA);
//attention: the following value is very sensitiv! Wrong value -> exception...
UINT8_TO_STREAM  (buf_p, strlen(ble_name) + 6);

//device name
UINT8_TO_STREAM(buf_p, 1 + strlen(ble_name));
UINT8_TO_STREAM(buf_p, 0x09);
while(*(name_p)) {
  UINT8_TO_STREAM(buf_p, *(name_p)++);
}

//slave connection interval range
UINT8_TO_STREAM (buf_p, 5);
UINT8_TO_STREAM (buf_p, 0x12);
UINT16_TO_STREAM(buf_p, BLE_SLAVE_CONN_INTERVAL_MIN);
UINT16_TO_STREAM(buf_p, BLE_SLAVE_CONN_INTERVAL_MAX);
```

TianHao-Espressif, 2017-02-10:
Hi, this problem happens because the HCI command have something invalid. If the invalid HCI command send to controller, it maybe cause HCI_HARDWARE_ERROR_EVT.
So please check your HCI command form following the HCI protocol or not .

benjaminaigner, 2017-02-10:
@TianHao-Espressif 
Thanks for the hint, yes it was related to the missing length field at the beginning of the HCI command

closed 2017-02-10 by benjaminaigner