espressif/esp-idf #330: 500

krzychb, 2017-02-09:
ESP-IDF repository has been avarded 500 :star: 's
Congratulations to Espressif's Team!

[image]

Spritetm, 2017-02-23:
It's already 536 now :) Thanks for the congrats! I'm going to close this issue to keep the issue list clean though.

closed 2017-02-23 by Spritetm

krzychb, 2017-07-05:
888 :star: 's yay!
[image]

krzychb, 2017-08-12:
Couldn’t resist
[image]

krzychb, 2017-10-27:
1234 :star: 's !
[image]
Congratulations! :clinking_glasses:

krzychb, 2018-06-16:
Another 1000 :star:'s in 10 months!
[image]

krzychb, 2019-03-09:
[image]
Another 1000 :star: 's in 9 months :+1: