espressif/esp-idf #342: gpio_set_pull_mode(<GPIO>, GPIO_PULLUP_ONLY) not work?

ESP32DE, 2017-02-14:
hi guys,

not sure but seems like we have a problem with Silicon Rev1 with set PULLUP intern ?
this simple code set the GPIO 34 and 35 as input, and i try to set the pullup,
the push button connector one is connected to GPIO,
the push button second connector is connected to GND,
but there is no high level if i try to get the level. what is wrong?

if i set up in hardware the pushbutton connector one with pullup resistor then i get the high level
if i push the button, then i get the low level. this work in HW , but not in SW

`#define BUTTON1  GPIO_NUM_34`
`#define BUTTON2  GPIO_NUM_35`
 
```
/* Set the GPIO as a input */
     gpio_set_direction(BUTTON1, GPIO_MODE_INPUT);
     gpio_set_direction(BUTTON2, GPIO_MODE_INPUT);
    
     /* Set the GPIO pull */
     gpio_set_pull_mode(BUTTON1, GPIO_PULLUP_ONLY);
     gpio_set_pull_mode(BUTTON2, GPIO_PULLUP_ONLY);
```

```
sprintf(sendbuff, "BTN VAL: %d\n\r", gpio_get_level(BUTTON1) );
esp_at_port_write_data((uint8_t *)sendbuff,strlen(sendbuff));
```

best wishes
rudi ;-)

negativekelvin, 2017-02-14:
Gpio 34-39 are input only no pullups/pulldowns

ESP32DE, 2017-02-14:
Thanks!

You never learn all - there is always something new to discover ;-)
Just with INPUT functions we need such pullup, pulldown possibility in software. 
Why this is not available, the vulture knows ;-)

Can someone update the pin list and add this note?

best wishes
Rudi

ESP32DE, 2017-02-14:
> GPIO pins 34-39 are input-only. These pins do not feature an output driver or internal pull-up/pull-down circuitry.

ok note 1 and note 2 was updated, have not read deaper, sry for excitement.
solved.

best wishes
rudi ;-)

closed 2017-02-14 by ESP32DE