espressif/esp-idf #344: I2C problems reading data ... not sending an ACK after reading data.

nkolban, 2017-02-15:
I am studying the I2C support in the ESP-IDF in the code base as of this date.  I decided to test out one of the simplest I2C devices available ... the HMC5883L ... a simple compass module.  The way this works is we send it the current register that we want to work with and then read data back from it.  As I worked with it, I could not get it working.

The device contains within it a "test" mode where if we set the register to be 10 (decimal) and then read 3 bytes, we get back an identifier that says that it is indeed an HMC5883L.  The data we should get back is 0x48, 0x34 and 0x33 which in ASCII is the code "H43".

I created a program which sends this data and what I find is that we fail to read the data.  When I dug further, I attached a logic analyzer.  The image for that is shown next.

[image]

As we see, we are sending a command to to 0x3D which is address 0x1E + Read ... and we get an ACK back.  Next we receive a byte which is of hex value 0x48 ... which is exactly what we expect.

Now ... pause ... notice that the trace shows NO acknowledgement ... i.e. NAK.  The logic should be that when a byte is received, the receiver sends an acknowledgement.  Since in our case, it is the ESP32 that is receiving the byte, it should be the ESP32 that sends the acknowledgement.  Here is the code fragment that I am using to perform the receive:

```
cmd = i2c_cmd_link_create();
ESP_ERROR_CHECK(i2c_master_start(cmd));
ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (I2C_ADDRESS << 1) | I2C_MASTER_READ, 1));
ESP_ERROR_CHECK(i2c_master_read(cmd, data, 3, 1));
ESP_ERROR_CHECK(i2c_master_stop(cmd));
ESP_ERROR_CHECK(i2c_master_cmd_begin(I2C_NUM_0, cmd, 100/portTICK_PERIOD_MS));
i2c_cmd_link_delete(cmd);
```

The core statement is `i2c_master_read` which, via the last parameter, is indeed claiming that it will send an ACK.

This is the end of the report ... however anticipating that you would want to see the whole program, here it is:

```
#include <driver/i2c.h>
#include <esp_log.h>
#include <freertos/FreeRTOS.h>
#include <freertos/task.h>

#include "sdkconfig.h"

#define PIN_SDA 18
#define PIN_CLK 19
#define I2C_ADDRESS 0x1e

static char tag[] = "hmc5883l";

#undef ESP_ERROR_CHECK
#define ESP_ERROR_CHECK(x)   do { esp_err_t rc = (x); if (rc != ESP_OK) { ESP_LOGE("err", "esp_err_t = %d", rc); assert(0 && #x);} } while(0);

void task_hmc5883l(void *ignore) {
	ESP_LOGD(tag, ">> hmc5883l");
	i2c_config_t conf;
	conf.mode = I2C_MODE_MASTER;
	conf.sda_io_num = PIN_SDA;
	conf.scl_io_num = PIN_CLK;
	conf.sda_pullup_en = GPIO_PULLUP_ENABLE;
	conf.scl_pullup_en = GPIO_PULLUP_ENABLE;
	conf.master.clk_speed = 100000;
	ESP_ERROR_CHECK(i2c_param_config(I2C_NUM_0, &conf));
	ESP_ERROR_CHECK(i2c_driver_install(I2C_NUM_0, I2C_MODE_MASTER, 0, 0, 0));

	i2c_cmd_handle_t cmd;

	//vTaskDelay(1000/portTICK_PERIOD_MS);

	uint8_t data[6];

	cmd = i2c_cmd_link_create();
	ESP_ERROR_CHECK(i2c_master_start(cmd));
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (I2C_ADDRESS << 1) | I2C_MASTER_WRITE, 1));
	//ESP_ERROR_CHECK(i2c_master_write_byte(cmd, 10, 1)); // Data registers
	ESP_ERROR_CHECK(i2c_master_stop(cmd));
	i2c_master_cmd_begin(I2C_NUM_0, cmd, 10/portTICK_PERIOD_MS);
	i2c_cmd_link_delete(cmd);

	cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (I2C_ADDRESS << 1) | I2C_MASTER_WRITE, 1);
	i2c_master_write_byte(cmd, 0x02, 1); // Mode register
	i2c_master_write_byte(cmd, 0x00, 1); // value 0
	i2c_master_stop(cmd);
	i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000/portTICK_PERIOD_MS);
	i2c_cmd_link_delete(cmd);

	cmd = i2c_cmd_link_create();
	i2c_master_start(cmd);
	i2c_master_write_byte(cmd, (I2C_ADDRESS << 1) | I2C_MASTER_WRITE, 1);
	i2c_master_write_byte(cmd, 0x01, 1); // Mode register
	i2c_master_write_byte(cmd, 0x20, 1); // value 0
	i2c_master_stop(cmd);
	i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000/portTICK_PERIOD_MS);
	i2c_cmd_link_delete(cmd);

	cmd = i2c_cmd_link_create();
	ESP_ERROR_CHECK(i2c_master_start(cmd));
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (I2C_ADDRESS << 1) | I2C_MASTER_WRITE, 1));
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, 10, 1)); // Data registers
	ESP_ERROR_CHECK(i2c_master_stop(cmd));
	ESP_ERROR_CHECK(i2c_master_cmd_begin(I2C_NUM_0, cmd, 100/portTICK_PERIOD_MS));
	i2c_cmd_link_delete(cmd);
	int i;

	cmd = i2c_cmd_link_create();
	ESP_ERROR_CHECK(i2c_master_start(cmd));
	ESP_ERROR_CHECK(i2c_master_write_byte(cmd, (I2C_ADDRESS << 1) | I2C_MASTER_READ, 1));
	ESP_ERROR_CHECK(i2c_master_read(cmd, data, 3, 1));
	//i2c_master_read_byte(cmd, data, 1);
	//i2c_master_read_byte(cmd, data+1, 1);
	//i2c_master_read_byte(cmd, data+2, 1);
	ESP_ERROR_CHECK(i2c_master_stop(cmd));
	ESP_ERROR_CHECK(i2c_master_cmd_begin(I2C_NUM_0, cmd, 100/portTICK_PERIOD_MS));
	i2c_cmd_link_delete(cmd);

	ESP_LOGD(tag, "ID Registers:")
	for (i=0; i<3; i++) {
		ESP_LOGD(tag, "%d: 0x%.2x", i, data[i]);
	}
	vTaskDelete(NULL);
} // task_hmc5883l

```

negativekelvin, 2017-02-15:
https://github.com/espressif/esp-idf/blob/master/examples/peripherals/i2c/main/i2c_test.c#L70

nkolban, 2017-02-15:
Wow!!  Thank you Mr @negativekelvin ... if I am reading the link you gave me correctly, then "ack" parameter of the I2C read APIs is "0" for YES ... send an ACK and "1" for NO ... don't send an ACK.

If we look at the docs here:

http://esp-idf.readthedocs.io/en/latest/api/peripherals/i2c.html#_CPPv220i2c_master_read_byte16i2c_cmd_handle_tP7uint8_ti

The `ack` parameter docs simply state that the value is "ack value for read command".  I would have assumed that "1" means send an ack and 0 means don't send an ack as opposed to the apparent reverse.

I'll leave this issue open a couple more days to see if Espressif are happy with the semantics as they stand and, if they are, I'll close out the issue.

MANY thanks for your post and helping me to understand better.

negativekelvin, 2017-02-15:
Perhaps better to think of as raw ack bit value, 0 = pull the line low.

nkolban, 2017-02-15:
If you yourself were designing the I2C read functions and wanted to control whether or not an ACK was transmitted by the ESP32 when a read completes ... what would YOU have designed?  Choices are:

* 1=Send Ack, 0=Don't send Ack

or

* 1=Don't send Ack, 0=Send Ack

negativekelvin, 2017-02-15:
Hard to say.  If I was thinking in waveforms, I may have had the parameter as the raw value of the ack bit as it is now.  I prefer using a descriptive enum/alias like the example because it is potentially confusing regardless of the implementation (ack=true, bit=0).  It is probably better to do that than introduce a breaking change.

nkolban, 2017-02-16:
I retested my circuit this evening with the new knowledge about the ack values and all now works.  Thank you for the help mr @negativekelvin

closed 2017-02-16 by nkolban