espressif/esp-idf #345: PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_EMAC_CLK_OUT) problem

ESP32DE, 2017-02-16:
hi guys

want to use RMII Interface, read Docu that GPIO16 pin provides 50MHz of clock output as a reference clock for PHY, but why this not work for use RMII CLK Out ?

`PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_EMAC_CLK_OUT);`

what is missing, what is need, what we have to read?
can't see CLK to set. is this the standard 50MHz CLK or how we can set/use FUNC_GPIO16_EMAC_CLK_OUT

can we use this in silicon V1 ?

best wishes
rudi ;-)

negativekelvin, 2017-02-16:
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/soc/emac_ex_reg.h#L24
https://github.com/espressif/esp-idf/blob/aece3d6fc62fd91d3c5a87e0ab2ddb5ac51d8ff0/components/ethernet/emac_dev.c#L104

Some configuration is required, not supported by driver yet

ESP32DE, 2017-02-16:
puzzled just in time:

Can someone give a construct what we must do for configuration in steps?
how we start the configuration for enabling example.
how we setup the clock it self for 50 MHz
..

best wishes
rudi ;-)

closed 2017-02-16 by ESP32DE

projectgus, 2017-02-16:
I'm going to reopen this so we can address it once we have an answer.

@ESP32DE, I noticed you asked the same question concurrently here, in the Arduino IDE project and on the esp32.com forums. Please don't do this. The same few people read all of these places already, and it makes our job harder when we have to consider duplicate questions in multiple places.

Also, please remember it may take more than 24 hours to get a reply from an Espressif employee.

Please be patient. As mentioned in other places, we don't have a good answer for this yet but we want to give you (and other Ethernet PHY users) one. @negativekelvin's suggestions are a good place to start if you want to progress this on your own.

ESP32DE, 2017-02-16:
hi angus

for what stand EX in EMAC_EX_
if extern, there is no INT ( intern ) register, bit to set just in time in 
https://github.com/espressif/esp-idf/blob/master/components/esp32/include/soc/emac_ex_reg.h#L24
for GPIO16 or have i misunderstood something.

EXT_OSC_CLK stands for extern 

https://www.esp32.com/viewtopic.php?f=12&t=1119#p4932

logial for me for setting intern GPIO16 as CLK output are two things:
#define EMAC_CLK_EN_REG 0x3ff000cc
#define EMAC_CLK_EN (BIT(14))

logial then simple for me:

#include "soc/emac_ex_reg.h"

PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_GPIO16);
gpio_set_direction(16, GPIO_MODE_OUTPUT);
SET_PERI_REG_MASK(EMAC_CLK_EN_REG, EMAC_CLK_EN ); 

but where to set the 50 MHz
it looks, emac_ex_reg.h is not completed ?

best wishes
rudi ;-)

negativekelvin, 2017-02-17:
Tried this

```
    gpio_set_direction(16, GPIO_MODE_OUTPUT);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO16_U, FUNC_GPIO16_EMAC_CLK_OUT);
    gpio_set_direction(0, GPIO_MODE_OUTPUT);
    PIN_FUNC_SELECT(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_EMAC_TX_CLK);
```
```
void emac_set_clk_rmii(void)
{
    //select ex clock source
    //REG_SET_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_EXT_OSC_EN);
    REG_CLR_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_EXT_OSC_EN);
    //ex clk enable
    //REG_SET_BIT(EMAC_EX_OSCCLK_CONF_REG, EMAC_EX_OSC_CLK_SEL);
    REG_CLR_BIT(EMAC_EX_OSCCLK_CONF_REG, EMAC_EX_OSC_CLK_SEL); 
    REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_DLY_NUM, 0);
    REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_H_DIV_NUM, 0xf); //try for slow clk
    REG_SET_FIELD(EMAC_EX_CLKOUT_CONF_REG, EMAC_EX_CLK_OUT_DIV_NUM, 0xf); //try for slow clk
    REG_SET_BIT(EMAC_EX_CLK_CTRL_REG, EMAC_EX_INT_OSC_EN);
    
}
```

Didn't get much other than a single pulse, but don't have a phy hooked up.  I can see all the MDIO data

ESP32DE, 2017-02-17:
have you seen last post:
https://esp32.com/viewtopic.php?f=17&t=1149&p=5315#p5314

> we may have to use the APLL to generate the 50MHz clock for the Ethernet

txs, i will try yours.

best wishes
rudi ;-)

ESP32DE, 2017-02-22:
push
+1

ESP32DE, 2017-02-28:
push
Hi Angus, 

After 12 days I ask times,
Any progress in this direction?
How do I generate the 50 MHz signal
-> GPIO 16 EMAC_CLK_OUT
-> APLL
-> other possibility

thank you for understanding

projectgus, 2017-02-28:
Hi rudi,

I saw via twitter that you have this working based on @igrr's suggestion, which is fantastic news:
https://www.esp32.com/viewtopic.php?f=2&t=813&p=5887#p5869

I'd like to keep this Issue open until we have proper support in IDF for this.

Angus