espressif/esp-idf #352: undefined reference to spi_flash_disable_cache  undefined reference to spi_flash_restore_cache

ESP32DE, 2017-02-18:
hi 

try to use
`#include "../cache_utils.h`

and try to use 
`spi_flash_disable_cache`
`spi_flash_restore_cache`

that are in cache_utils.c
https://github.com/espressif/esp-idf/blob/master/components/spi_flash/cache_utils.c#L35

> CC sd_card.o
> AR libmain.a
> LD sd_card.elf
> C:/nod/27_sd_Card_iram/build/main\libmain.a(sd_card.o):(.iram1.literal+0x14): undefined reference to `spi_flash_disable_cache'
> C:/nod/27_sd_Card_iram/build/main\libmain.a(sd_card.o):(.iram1.literal+0x18): undefined reference to `spi_flash_restore_cache'
> C:/nod/27_sd_Card_iram/build/main\libmain.a(sd_card.o): In function `app_main':
> C:/nod/27_sd_Card_iram/main/sd_card.c:98: undefined reference to `spi_flash_disable_cache'
> C:/nod/27_sd_Card_iram/main/sd_card.c:100: undefined reference to `spi_flash_restore_cache'
> 

best wishes
rudi ;-)

ESP32DE, 2017-02-18:
```
extern void spi_flash_disable_interrupts_caches_and_other_cpu();
extern void spi_flash_enable_interrupts_caches_and_other_cpu();

extern void spi_flash_disable_interrupts_caches_and_other_cpu_no_os();
extern void spi_flash_enable_interrupts_caches_no_os();

/**
 * The following two functions are replacements for Cache_Read_Disable and Cache_Read_Enable
 * function in ROM. They are used to work around a bug where Cache_Read_Disable requires a call to
 * Cache_Flush before Cache_Read_Enable, even if cached data was not modified.
 */

static const uint32_t cache_mask  = DPORT_APP_CACHE_MASK_OPSDRAM | DPORT_APP_CACHE_MASK_DROM0 |
        DPORT_APP_CACHE_MASK_DRAM1 | DPORT_APP_CACHE_MASK_IROM0 |
        DPORT_APP_CACHE_MASK_IRAM1 | DPORT_APP_CACHE_MASK_IRAM0;

static void IRAM_ATTR spi_flash_disable_cache(uint32_t cpuid, uint32_t* saved_state)
{
    uint32_t ret = 0;
    if (cpuid == 0) {
        ret |= GET_PERI_REG_BITS2(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_PRO_DCACHE_DBUG0_REG, DPORT_PRO_CACHE_STATE, DPORT_PRO_CACHE_STATE_S) != 1) {
            ;
        }
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL_REG, 1, 0, DPORT_PRO_CACHE_ENABLE_S);
    } else {
        ret |= GET_PERI_REG_BITS2(DPORT_APP_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_APP_DCACHE_DBUG0_REG, DPORT_APP_CACHE_STATE, DPORT_APP_CACHE_STATE_S) != 1) {
            ;
        }
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL_REG, 1, 0, DPORT_APP_CACHE_ENABLE_S);
    }
    *saved_state = ret;
}

static void IRAM_ATTR spi_flash_restore_cache(uint32_t cpuid, uint32_t saved_state)
{
    if (cpuid == 0) {
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL_REG, 1, 1, DPORT_PRO_CACHE_ENABLE_S);
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, saved_state, 0);
    } else {
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL_REG, 1, 1, DPORT_APP_CACHE_ENABLE_S);
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL1_REG, cache_mask, saved_state, 0);
    }
}

```

```
uint32_t cpuid = 1;
    uint32_t* saved_state=0;
	
		
	printf("try disable:..\n\r");
	spi_flash_disable_cache(cpuid, (uint32_t *) saved_state);
	
	printf("try restore:..\n\r");
	spi_flash_restore_cache( cpuid, (uint32_t) saved_state);
	
	printf("go in while..\n\r");
	while (-1) ;

```

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> ets Jun  8 2016 00:22:57
> 
[... 164 lines cut ...]
> I (657) boot: segment 7: paddr=0x00030010 vaddr=0x400d0018 size=0x0c454 ( 50260) map
> I (684) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (707) heap_alloc_caps: At 3FFB20F8 len 0002DF08 (183 KiB): DRAM
> I (728) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
> I (749) heap_alloc_caps: At 40088108 len 00017EF8 (95 KiB): IRAM
> I (769) cpu_start: Pro cpu up.
> I (781) cpu_start: Starting app cpu, entry point is 0x40080994
> I (0) cpu_start: App cpu up.
> I (812) cpu_start: Pro cpu start user code
> I (871) cpu_start: Starting scheduler on PRO CPU.
> I (877) example: Initializing SD card
> try disable:..
> GGuru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x400d09f6  PS      : 0x00060d33  A0      : 0x800d0cdb  A1      : 0x3ffe7c90
> A2      : 0x0000040e  A3      : 0x00000001  A4      : 0xffffffff  A5      : 0x00000003
> A6      : 0x0000001f  A7      : 0x00000001  A8      : 0x800d0ae3  A9      : 0x00000000
> A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffe7e30
> A14     : 0x0000abab  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400d09f6:0x3ffe7c90 0x400d0cdb:0x3ffe7cd0 0x400d0ebc:0x3ffe7d10 0x400d0934:0x3ffe7d40 0x4008097e:0x3ffe7d60 0x40080a0c:0x3ffe7d80 0x40007c34:0x3ffe7dc0 0x40000740:0x3ffe7e30
> 
> Rebooting...
>

negativekelvin, 2017-02-18:
> extern void spi_flash_disable_interrupts_caches_and_other_cpu();
> extern void spi_flash_enable_interrupts_caches_and_other_cpu();

Already calls 

```
extern void IRAM_ATTR spi_flash_disable_cache(uint32_t cpuid, uint32_t* saved_state);
extern void IRAM_ATTR spi_flash_restore_cache(uint32_t cpuid, uint32_t saved_state);
```

ESP32DE, 2017-02-18:
thank you @negativekelvin 

https://github.com/espressif/esp-idf/issues/353

have changed the code but this reboots.
just in time search with gdbstub

best wishes
rudi ;-)

Spritetm, 2017-02-21:
Okay, closing this issue then.

closed 2017-02-21 by Spritetm