espressif/esp-idf #353: reboot on spi_flash_disable_cache / spi_flash_restore_cache

ESP32DE, 2017-02-18:
```
extern void spi_flash_disable_interrupts_caches_and_other_cpu();
extern void spi_flash_enable_interrupts_caches_and_other_cpu();

extern void spi_flash_disable_interrupts_caches_and_other_cpu_no_os();
extern void spi_flash_enable_interrupts_caches_no_os();

/**
 * The following two functions are replacements for Cache_Read_Disable and Cache_Read_Enable
 * function in ROM. They are used to work around a bug where Cache_Read_Disable requires a call to
 * Cache_Flush before Cache_Read_Enable, even if cached data was not modified.
 */

static const uint32_t cache_mask  = DPORT_APP_CACHE_MASK_OPSDRAM | DPORT_APP_CACHE_MASK_DROM0 |
        DPORT_APP_CACHE_MASK_DRAM1 | DPORT_APP_CACHE_MASK_IROM0 |
        DPORT_APP_CACHE_MASK_IRAM1 | DPORT_APP_CACHE_MASK_IRAM0;

static void IRAM_ATTR spi_flash_disable_cache(uint32_t cpuid, uint32_t* saved_state)
{
    uint32_t ret = 0;
    if (cpuid == 0) {
        ret |= GET_PERI_REG_BITS2(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_PRO_DCACHE_DBUG0_REG, DPORT_PRO_CACHE_STATE, DPORT_PRO_CACHE_STATE_S) != 1) {
            ;
        }
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL_REG, 1, 0, DPORT_PRO_CACHE_ENABLE_S);
    } else {
        ret |= GET_PERI_REG_BITS2(DPORT_APP_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_APP_DCACHE_DBUG0_REG, DPORT_APP_CACHE_STATE, DPORT_APP_CACHE_STATE_S) != 1) {
            ;
        }
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL_REG, 1, 0, DPORT_APP_CACHE_ENABLE_S);
    }
    *saved_state = ret;
}

static void IRAM_ATTR spi_flash_restore_cache(uint32_t cpuid, uint32_t saved_state)
{
    if (cpuid == 0) {
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL_REG, 1, 1, DPORT_PRO_CACHE_ENABLE_S);
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, saved_state, 0);
    } else {
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL_REG, 1, 1, DPORT_APP_CACHE_ENABLE_S);
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL1_REG, cache_mask, saved_state, 0);
    }
}

```

```
uint32_t cpuid = 1;
    uint32_t* saved_state=0;
	
		
	printf("try disable:..\n\r");
	spi_flash_disable_cache(cpuid, (uint32_t *) saved_state);
	
	printf("try restore:..\n\r");
	spi_flash_restore_cache( cpuid, (uint32_t) saved_state);
	
	printf("go in while..\n\r");
	while (-1) ;

```

> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> ets Jun  8 2016 00:22:57
> 
[... 164 lines cut ...]
> I (657) boot: segment 7: paddr=0x00030010 vaddr=0x400d0018 size=0x0c454 ( 50260) map
> I (684) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (707) heap_alloc_caps: At 3FFB20F8 len 0002DF08 (183 KiB): DRAM
> I (728) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
> I (749) heap_alloc_caps: At 40088108 len 00017EF8 (95 KiB): IRAM
> I (769) cpu_start: Pro cpu up.
> I (781) cpu_start: Starting app cpu, entry point is 0x40080994
> I (0) cpu_start: App cpu up.
> I (812) cpu_start: Pro cpu start user code
> I (871) cpu_start: Starting scheduler on PRO CPU.
> I (877) example: Initializing SD card
> try disable:..
> GGuru Meditation Error of type IllegalInstruction occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x400d09f6  PS      : 0x00060d33  A0      : 0x800d0cdb  A1      : 0x3ffe7c90
> A2      : 0x0000040e  A3      : 0x00000001  A4      : 0xffffffff  A5      : 0x00000003
> A6      : 0x0000001f  A7      : 0x00000001  A8      : 0x800d0ae3  A9      : 0x00000000
> A10     : 0x00000001  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x3ffe7e30
> A14     : 0x0000abab  A15     : 0x00000001  SAR     : 0x00000000  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400d09f6:0x3ffe7c90 0x400d0cdb:0x3ffe7cd0 0x400d0ebc:0x3ffe7d10 0x400d0934:0x3ffe7d40 0x4008097e:0x3ffe7d60 0x40080a0c:0x3ffe7d80 0x40007c34:0x3ffe7dc0 0x40000740:0x3ffe7e30
> 
> Rebooting...
>

igrr, 2017-02-18:
In your last snippet you are disabling the cache and then calling printf. Printf can not work with cache disabled because it uses functions which are not in IRAM.

ESP32DE, 2017-02-18:
```
uint32_t cpuid = 1;
    uint32_t* saved_state=0;
	
		
	// printf("try disable:..\n\r");
	spi_flash_disable_cache(cpuid, (uint32_t *) saved_state);
	
	//printf("try restore:..\n\r");
	spi_flash_restore_cache( cpuid, (uint32_t) saved_state);
	
	printf("go in while..\n\r");
	while (-1) ;
```

> $ make flash monitor
> Flashing binaries to serial port COM8 (app at offset 0x10000)...
> esptool.py v2.0-beta1
> Connecting...
> Uploading stub...
> Running stub...
> Stub running...
> Changing baud rate to 921600
> Changed.
> Attaching SPI flash...
> Configuring flash size...
> Auto-detected Flash size: 4MB
> Flash params set to 0x0220
> Wrote 16384 bytes at 0x00001000 in 0.2 seconds (600.1 kbit/s)...
> Hash of data verified.
> Wrote 196608 bytes at 0x00010000 in 2.7 seconds (576.1 kbit/s)...
> Hash of data verified.
> Wrote 16384 bytes at 0x00008000 in 0.2 seconds (646.3 kbit/s)...
> Hash of data verified.
> 
> Leaving...
> Hard resetting...
> --- Miniterm on COM8  115200,8,N,1 ---
> --- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0008,len:8
> load:0x3fff0010,len:3388
> ho 0 tail 12 room 4
> load:0x40078000,len:7504
> load:0x40080000,len:252
> entry 0x40080034
> I (47) boot: ESP-IDF v2.0-rc1-21-g30a557a 2nd stage bootloader
> I (47) boot: compile time 13:34:58
> I (47) boot: Enabling RNG early entropy source...
> I (67) boot: SPI Speed      : 40MHz
> I (80) boot: SPI Mode       : DIO
> I (92) boot: SPI Flash Size : 4MB
> I (105) boot: Partition Table:
> I (116) boot: ## Label            Usage          Type ST Offset   Length
> I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (162) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (185) boot:  2 factory          factory app      00 00 00010000 00100000
> I (209) boot: End of partition table
> I (222) boot: Disabling RNG early entropy source...
> I (239) boot: Loading app partition at offset 00010000
> I (511) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
> I (511) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x02270 (  8816) map
[... 60 lines cut ...]
> I (632) boot: segment 6: paddr=0x0002be2c vaddr=0x00000000 size=0x041dc ( 16860)
> I (657) boot: segment 7: paddr=0x00030010 vaddr=0x400d0018 size=0x0c454 ( 50260) map
> I (684) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (707) heap_alloc_caps: At 3FFB20F8 len 0002DF08 (183 KiB): DRAM
> I (727) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
> I (749) heap_alloc_caps: At 400880F4 len 00017F0C (95 KiB): IRAM
> I (769) cpu_start: Pro cpu up.
> I (781) cpu_start: Starting app cpu, entry point is 0x4008098c
> I (0) cpu_start: App cpu up.
> I (812) cpu_start: Pro cpu start user code
> I (871) cpu_start: Starting scheduler on PRO CPU.
> I (877) example: Initializing SD card
> GGuru Meditation Error of type LoadProhibited occurred on core  1. Exception was unhandled.
> Register dump:
> PC      : 0x400d09fa  PS      : 0x00060d33  A0      : 0x800d0ae3  A1      : 0x3ffe7c70
> A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffe7e30
> A6      : 0x0000abab  A7      : 0x00000001  A8      : 0xbad00bad  A9      : 0x3ffe7c80
> A10     : 0x00000003  A11     : 0x00060d23  A12     : 0x00060d23  A13     : 0x00000000
> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
> EXCVADDR: 0xbad00bad  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400d09fa:0x3ffe7c70 0x400d0ae3:0x3ffe7c90 0x400d0cdb:0x3ffe7cd0 0x400d0ebc:0x3ffe7d10 0x400d0934:0x3ffe7d40 0x40080976:0x3ffe7d60 0x40080a04:0x3ffe7d80 0x40007c34:0x3ffe7dc0 0x40000740:0x3ffe7e30
> 
> Rebooting...
>

igrr, 2017-02-18:
Where is `I (877) example: Initializing SD card` coming from?

Could you please use GDBStub of Crash dump function to produce readable backtrace at the point of an exception.

ESP32DE, 2017-02-18:
comes form app_main
test itself is in app_main too:

```
void IRAM_ATTR app_main(void)
{
    ESP_LOGI(TAG, "Initializing SD card");
	
	 // spi_flash_disable_interrupts_caches_and_other_cpu();
	 // spi_flash_enable_interrupts_caches_and_other_cpu();
	
	// spi_flash_disable_interrupts_caches_and_other_cpu_no_os();
	
	// spi_flash_disable_cache(1, 1 );
	
	// spi_flash_disable_interrupts_caches_and_other_cpu();
	// spi_flash_enable_interrupts_caches_and_other_cpu();
	// printf("disable done..\n\r");
	// while (-1) ;
	
	// new test
	
	uint32_t cpuid = 1;
    uint32_t* saved_state=0;
	
		
	// printf("try disable:..\n\r");
	spi_flash_disable_cache(cpuid, (uint32_t *) saved_state);
	
	//printf("try restore:..\n\r");
	spi_flash_restore_cache( cpuid, (uint32_t) saved_state);
	
	printf("go in while..\n\r");
	while (-1) ;

    sdmmc_host_t host = SDMMC_HOST_DEFAULT();

    // To use 1-line SD mode, uncomment the following line:
    // host.flags = SDMMC_HOST_FLAG_1BIT;

    // This initializes the slot without card detect (CD) and write protect (WP) signals.
    // Modify slot_config.gpio_cd and slot_config.gpio_wp if your board has these signals.
    sdmmc_slot_config_t slot_config  = SDMMC_SLOT_CONFIG_DEFAULT();

//	spi_flash_disable_interrupts_caches_and_other_cpu();
//	spi_flash_enable_interrupts_caches_and_other_cpu();
//	printf("disable done..\n\r");
//	while (-1) ;
// ok
	
    // Options for mounting the filesystem.
    // If format_if_mount_failed is set to true, SD card will be partitioned and formatted
    // in case when mounting fails.
    esp_vfs_fat_sdmmc_mount_config_t mount_config = {
        .format_if_mount_failed = false, /*false was orig*/
        .max_files = 5
    };

//    printf("ok..\n");
//	spi_flash_disable_interrupts_caches_and_other_cpu();
//	spi_flash_enable_interrupts_caches_and_other_cpu();
//	printf("disable done..\n\r");
//	while (-1) ;

    // Use settings defined above to initialize SD card and mount FAT filesystem.
    // Note: esp_vfs_fat_sdmmc_mount is an all-in-one convenience function.
    // Please check its source code and implement error recovery when developing
    // production applications.
    sdmmc_card_t* card;

    // printf("ok..\n");
	// spi_flash_disable_interrupts_caches_and_other_cpu();
	// printf("disable done..\n\r");
    // spi_flash_enable_interrupts_caches_and_other_cpu();
	
	
	// spi_flash_disable_interrupts_caches_and_other_cpu_no_os();
	// spi_flash_enable_interrupts_caches_no_os();
	// printf("ok v2\n");
	// while(-1);;
    
	esp_err_t ret = esp_vfs_fat_sdmmc_mount("/sdcard", &host, &slot_config, &mount_config, &card);
 
	// 
    // spi_flash_enable_interrupts_caches_and_other_cpu();
    
	// spi_flash_enable_interrupts_caches_and_other_cpu();
//	printf("enable done..\n\r");
	// while (-1) ;

 if (ret != ESP_OK) {
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAG, "Failed to mount filesystem. If you want the card to be formatted, set format_if_mount_failed = true.");
        } else {
            ESP_LOGE(TAG, "Failed to initialize the card (%d). Make sure SD card lines have pull-up resistors in place.", ret);
        }
        return;
    }

	
    // Card has been initialized, print its properties
    sdmmc_card_print_info(stdout, card);

    // Use POSIX and C standard library functions to work with files.
    // First create a file.
    ESP_LOGI(TAG, "Opening file");
    FILE* f = fopen("/sdcard/hello.txt", "w");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for writing");
        return;
    }
    fprintf(f, "Hello %s!\r\n", card->cid.name);
    	
	fclose(f);
    ESP_LOGI(TAG, "File written");

    // Check if destination file exists before renaming
    struct stat st;
    if (stat("/sdcard/foo.txt", &st) == 0) {
        // Delete it if it exists
        unlink("/sdcard/foo.txt");
    }

    // Rename original file
    ESP_LOGI(TAG, "Renaming file");
    if (rename("/sdcard/hello.txt", "/sdcard/foo.txt") != 0) {
        ESP_LOGE(TAG, "Rename failed");
        return;
    }

    // Open renamed file for reading
    ESP_LOGI(TAG, "Reading file");
    f = fopen("/sdcard/foo.txt", "r");
    if (f == NULL) {
        ESP_LOGE(TAG, "Failed to open file for reading");
        return;
    }
    char line[64];
    fgets(line, sizeof(line), f);
    fclose(f);
    // strip newline
    char* pos = strchr(line, '\n');
    if (pos) {
        *pos = '\0';
    }
    ESP_LOGI(TAG, "Read from file: '%s'", line);

    // All done, unmount partition and disable SDMMC host peripheral
    esp_vfs_fat_sdmmc_unmount();
    ESP_LOGI(TAG, "Card unmounted");
	// spi_flash_enable_interrupts_caches_and_other_cpu();
    // void spi_flash_enable_interrupts_caches_no_os();

}
```

ivan, ;-) - have not tried GDBStub
must read first ;-)

best wishes
rudi ;-)

ESP32DE, 2017-02-18:
now all without printf

```
// new test
	
	uint32_t cpuid = 1;
    uint32_t* saved_state=0;
	
		
	// printf("try disable:..\n\r");
	spi_flash_disable_cache(cpuid, (uint32_t *) saved_state);
	
	//printf("try restore:..\n\r");
	spi_flash_restore_cache( cpuid, (uint32_t) saved_state);
	
	// printf("go in while..\n\r");
	while (-1) ;
```

no success

negativekelvin, 2017-02-18:
I think you have a bit of work to do: put the sdmmc functions into iram, add the conditional peripheral arbitration for HS1,SPI0/1, make sure the state of the flash and card are preserved/restored between operations, slice SD operations to minimize cache disable effects

ESP32DE, 2017-02-18:
only for the protocoll ;-)

ok - sorted it out in one part,
txs ivan for the tip with gdbstub

> $ xtensa-esp32-elf-gdb ./build/sd_card.elf -b 115200 -ex 'target remote COM8'
> GNU gdb (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 7.10
> Copyright (C) 2015 Free Software Foundation, Inc.
> License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
> This is free software: you are free to change and redistribute it.
> There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
> and "show warranty" for details.
> This GDB was configured as "--host=i686-host_pc-mingw32 --target=xtensa-esp32-elf".
> Type "show configuration" for configuration details.
> For bug reporting instructions, please see:
> <http://www.gnu.org/software/gdb/bugs/>.
> Find the GDB manual and other documentation resources online at:
> <http://www.gnu.org/software/gdb/documentation/>.
> For help, type "help".
> Type "apropos word" to search for commands related to "word"...
> Reading symbols from ./build/sd_card.elf...done.
> Remote debugging using COM8
> spi_flash_disable_cache (cpuid=<optimized out>, saved_state=0x0)
>     at C:/nod/27_sd_card_iram/main/sd_card.c:120
> 120         *saved_state = ret;
> (gdb)
> 

there is a "pointer" mistake with

`*saved_state=ret;`

but simple test get 38 ( 0x26 ) for 
```

void IRAM_ATTR app_main(void)
{
    ESP_LOGI(TAG, "..app_main starts..");
	
	uint32_t testme = 0;
	
	testme |= GET_PERI_REG_BITS2(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_PRO_DCACHE_DBUG0_REG, DPORT_PRO_CACHE_STATE, DPORT_PRO_CACHE_STATE_S) != 1) {
            ;
        }
		
	printf("ret cpu 0 : %p\n\r", (uint32_t *) testme);
	
	testme = 0;
	
	testme |= GET_PERI_REG_BITS2(DPORT_APP_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_APP_DCACHE_DBUG0_REG, DPORT_APP_CACHE_STATE, DPORT_APP_CACHE_STATE_S) != 1) {
            ;
        }
		
	printf("ret cpu 1 : %p\n\r", (uint32_t *) testme);
		
	
	
	 // spi_flash_disable_interrupts_caches_and_other_cpu();
	 // spi_flash_enable_interrupts_caches_and_other_cpu();
	
	// spi_flash_disable_interrupts_caches_and_other_cpu_no_os();
	
	// spi_flash_disable_cache(1, 1 );
	
	// spi_flash_disable_interrupts_caches_and_other_cpu();
	// spi_flash_enable_interrupts_caches_and_other_cpu();
	// printf("disable done..\n\r");
	// while (-1) ;
	
	// new test
	
	// uint32_t cpuid = 0;
    // uint32_t* saved_state= NULL;
	
		
	printf("try disable:..\n\r");
	spi_flash_disable_cache(1, saved_state);
	
	// while (-1) ;
	
	printf("try restore:..\n\r");
	spi_flash_restore_cache( 1, (uint32_t) saved_state);
	
	 printf("go in while..\n\r");
	 while (-1) ;
```

i have disabled cache now? and print after is runing ?

> entry 0x40080034
> I (47) boot: ESP-IDF v2.0-rc1-21-g30a557a 2nd stage bootloader
> I (47) boot: compile time 16:37:51
> I (47) boot: Enabling RNG early entropy source...
> I (67) boot: SPI Speed      : 40MHz
> I (80) boot: SPI Mode       : DIO
> I (92) boot: SPI Flash Size : 4MB
> I (104) boot: Partition Table:
> I (116) boot: ## Label            Usage          Type ST Offset   Length
> I (139) boot:  0 nvs              WiFi data        01 02 00009000 00006000
> I (162) boot:  1 phy_init         RF data          01 01 0000f000 00001000
> I (185) boot:  2 factory          factory app      00 00 00010000 00100000
> I (208) boot: End of partition table
> I (221) boot: Disabling RNG early entropy source...
> I (238) boot: Loading app partition at offset 00010000
> I (513) boot: segment 0: paddr=0x00010018 vaddr=0x00000000 size=0x0ffe8 ( 65512)
> I (513) boot: segment 1: paddr=0x00020008 vaddr=0x3f400010 size=0x02678 (  9848) map
> I (529) boot: segment 2: paddr=0x00022688 vaddr=0x3ffb0000 size=0x01aa8 (  6824) load
> I (558) boot: segment 3: paddr=0x00024138 vaddr=0x40080000 size=0x00400 (  1024) load
> I (582) boot: segment 4: paddr=0x00024540 vaddr=0x40080400 size=0x07c84 ( 31876) load
> I (623) boot: segment 5: paddr=0x0002c1cc vaddr=0x400c0000 size=0x00000 (     0) load
> I (634) boot: segment 6: paddr=0x0002c1d4 vaddr=0x00000000 size=0x03e34 ( 15924)
> I (659) boot: segment 7: paddr=0x00030010 vaddr=0x400d0018 size=0x0ca04 ( 51716) map
> I (686) heap_alloc_caps: Initializing. RAM available for dynamic allocation:
> I (709) heap_alloc_caps: At 3FFB24B0 len 0002DB50 (182 KiB): DRAM
> I (730) heap_alloc_caps: At 3FFE8000 len 00018000 (96 KiB): D/IRAM
> I (751) heap_alloc_caps: At 40088084 len 00017F7C (95 KiB): IRAM
> I (771) cpu_start: Pro cpu up.
> I (783) cpu_start: Starting app cpu, entry point is 0x40080980
> I (0) cpu_start: App cpu up.
> I (814) cpu_start: Pro cpu start user code
> I (873) cpu_start: Starting scheduler on PRO CPU.
> I (879) example: ..app_main starts..
> ret cpu 0 : 0x26
> ret I (879) cpu_start: Starting scheduler on APP CPU.
> cpu 1 : 0x26
> try disable:..
> ret=38
> saved_state=38
> try restore:..
> go in while..
> 

ok .. will try more in this what happend here..
( must check cache_utils.c function ) 

best wishes
rudi ;-)

ESP32DE, 2017-02-18:
@negativekelvin

> I think you have a bit of work to do: put the sdmmc functions into iram, add the conditional peripheral arbitration for HS1,SPI0/1, make sure the state of the flash and card are preserved/restored between operations, slice SD operations to minimize cache disable effects

thank you verry much

https://esp32.com/viewtopic.php?f=17&p=5315#p5314

a- sdmmc functions into iram, 
b- add the conditional peripheral arbitration for HS1,SPI0/1
c- state of the flash and card are preserved/restored between operations

a)
sdmmc_host.c
all function are attributed with IRAM_ATTR

b) need a small help

c) ok, think i have successfull done now ( pointer mistake ) 

best wishes
rudi ;-)

ESP32DE, 2017-02-18:
think, cause write and read to and from SD Card
some function must in IRAM here too:
https://github.com/espressif/esp-idf/tree/master/components/fatfs/src

proposal:
i think, i am not alone to try this,
what you guys think about a template for this if user want use same pins for sd card and flash?
better, in sd card driver make option for makeconfig that we can choice the 
Standard or Extended version ( flash and sd card same pins ) .

best wishes
rudi ;-)
 

best wishes
rudi ;-)

ESP32DE, 2017-02-18:
**now have killed gdbstub** 
:1st_place_medal: 

> I (1139) example: ..app_main starts..
> ret cpu 0 : 0x26
> reI (1139) cpu_start: Starting scheduler on APP CPU.
> t cpu 1 : 0x26
> point1:
> point 2:
> point 3:
> point 4:
> point 5:
> try disable for init card:..
> ret=38
> saved_state=38
> *Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x400dfdc8  PS      : 0x00060230  A0      : 0x80081e4a  A1      : 0x3ffb45b0
> A2      : 0x3f401d3c  A3      : 0x3ffb45f0  A4      : 0x3ffb236c  A5      : 0x3ffb461c
> A6      : 0x3ffb4624  A7      : 0x40082740  A8      : 0x80082862  A9      : 0x3ffb4590
> A10     : 0x3ffb45f0  A11     : 0x3ffb60a8  A12     : 0xffffffff  A13     : 0xffffffff
> A14     : 0xffffffff  A15     : 0x00000004  SAR     : 0x0000001f  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff
> 
> Backtrace: 0x400dfdc8:0x3ffb45b0 0x40081e4a:0x3ffb45f0 0x400d06e1:0x3ffb4650
> 
> Entering gdb stub now.
> Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x400d1014  PS      : 0x00060033  A0      : 0x8008554c  A1      : 0x3ffb4450
> A2      : 0x3ffb44f0  A3      : 0x00000017  A4      : 0x3ffb4550  A5      : 0x00000018
> A6      : 0x3ffb60a8  A7      : 0x0000006c  A8      : 0x800853c8  A9      : 0x3ffb4430
> A10     : 0x3ffb44f0  A11     : 0x00000018  A12     : 0x3ffb4650  A13     : 0x3ffb44b0
> A14     : 0x3ffb1530  A15     : 0x3ffb153c  SAR     : 0x00000020  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400d1014:0x3ffb4450 0x4008554c:0x3ffb44d0 0x40081021:0x3ffb44f0 0x400dfdc8:0x3ffb45b0 0x400dfdc8:0x3ffb45f0 0x400d06e1:0x3ffb4650
> 
> Entering gdb stub now.
> Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x400d1014  PS      : 0x00060033  A0      : 0x8008554c  A1      : 0x3ffb42f0
> A2      : 0x3ffb4390  A3      : 0x00000017  A4      : 0x3ffb43f0  A5      : 0x00000018
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800853c8  A9      : 0x3ffb42d0
> A10     : 0x3ffb4390  A11     : 0x00000018  A12     : 0x3ffb4650  A13     : 0x3ffb4350
> A14     : 0x3ffb1530  A15     : 0x3ffb153c  SAR     : 0x0000001d  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400d1014:0x3ffb42f0 0x4008554c:0x3ffb4370 0x40081021:0x3ffb4390 0x400d1014:0x3ffb4450 0x400d1014:0x3ffb44d0 0x40081021:0x3ffb44f0 0x400dfdc8:0x3ffb45b0 0x400dfdc8:0x3ffb45f0 0x400d06e1:0x3ffb4650
> 
> Entering gdb stub now.
> Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x400d1014  PS      : 0x00060033  A0      : 0x8008554c  A1      : 0x3ffb4190
> A2      : 0x3ffb4230  A3      : 0x00000017  A4      : 0x3ffb4290  A5      : 0x00000018
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800853c8  A9      : 0x3ffb4170
> A10     : 0x3ffb4230  A11     : 0x00000018  A12     : 0x3ffb4650  A13     : 0x3ffb41f0
> A14     : 0x3ffb1530  A15     : 0x3ffb153c  SAR     : 0x0000001a  EXCCAUSE: 0x00000000
> EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000
> 
> Backtrace: 0x400d1014:0x3ffb4190 0x4008554c:0x3ffb4210 0x40081021:0x3ffb4230 0x400d1014:0x3ffb42f0 0x400d1014:0x3ffb4370 0x40081021:0x3ffb4390 0x400d1014:0x3ffb4450 0x400d1014:0x3ffb44d0 0x40081021:0x3ffb44f0 0x400dfdc8:0x3ffb45b0 0x400dfdc8:0x3ffb45f0 0x400d06e1:0x3ffb4650
> 
> Entering gdb stub now.
> Guru Meditation Error of type IllegalInstruction occurred on core  0. Exception was unhandled.
> Register dump:
> PC      : 0x400d1014  PS      : 0x00060033  A0      : 0x8008554c  A1      : 0x3ffb4030
> A2      : 0x3ffb40d0  A3      : 0x00000017  A4      : 0x3ffb4130  A5      : 0x00000018
> A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800853c8  A9      : 0x3ffb4010
> A10     : 0x3ffb40d0  A11     : 0x00000018  A12     : 0x3ffb4650  A13     : 0x3ffb4090
> A14     : 0x3ffb1530  A15     : 0x3ffb153c  SAR     : 0x00000017  EXCCAUSE: 0x00000000
[... 333 lines cut ...]
> PC      : 0x4008537d  PS      : 0x00060033  A0      : 0x8008554c  A1      : 0x3ffb1850
> (▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?Unknown▒▒ occurre}@▒▒? 0}@3Register dump:
> PC      : 0x4008537d  PS      : 0x00060033  A0      : 0x8008554c  A1      : 0x3ffb16f0
> (▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?Unknown▒▒ occurre}@▒▒? 0}@3Register dump:
> PC      (▒?4▒?@▒?L▒?@0▒?P▒?: 0x4008537d  4▒?@▒?L▒?@0▒?P▒?: 0x00060033  @0▒?P▒?: 0x8008554c
> (▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?Unknown▒▒ occurre}@▒▒? 0}@3▒▒?4008537d4▒?@▒?L▒?@0▒?P▒?00060033@0▒?P▒?8008554c
> (▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?Unknown▒▒ occurre}@▒▒? 0}@3▒▒?4008537d4▒?@▒?L▒?@0▒?P▒?00060033@0▒?P▒?8008554c
> (▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?Unknown▒▒ occurre}@▒▒? 0}@3▒▒?4008537d4▒?@▒?L▒?@0▒?P▒?00060033@0▒?P▒?8008554c
> (▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?Unknown▒▒ occurre}@▒▒? 0}@3▒▒?4008537d4▒?@▒?L▒?@0▒?P▒?00060033@0▒?P▒?8008554c
> *IDL▒?X▒?d▒?p▒?|▒?(▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?g exception reas}@0▒? 0L▒?X▒?d▒?p▒?|▒?g exception reas}@0▒?@0▒?P▒?p▒?|▒?g exception reas}@0▒?▒▒?3ffb14404▒?@▒?L▒?@0▒?P▒?00060036@0▒?P▒?7ffb1440
> LE4L▒?X▒?d▒?p▒?|▒?(▒?4▒?@▒?L▒?X▒?d▒?p▒?|▒?g exception reas}@0▒? 0L▒?X▒?d▒?p▒?|▒?g exception reas}@0▒?ets Jun  8 2016 00:22:57
> 
> rst:0x7 (TG0WDT_SYS_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0x00
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0008,len:8
> load:0x3fff0010,len:3388
> ho 0 tail 12 room 4
> load:0x40078000,len:7504
> load:0x40080000,len:252
> entry 0x40080034
> I (280) boot: ESP-IDF v2.0-rc1-21-g30a557a 2nd stage bootloader
> I (281) boot: compile time 16:37:51
> I (281) boot: Enabling RNG early entropy source...

ESP32DE, 2017-02-18:
example mount -> IRAM
https://github.com/espressif/esp-idf/blob/master/components/fatfs/src/vfs_fat_sdmmc.c#L26
...

igrr, 2017-02-18:
Moving functions from sdmmc_host to IRAM will likely be insufficient. If you look at `sdmmc_host_do_transaction`, it will call `sdmmc_host_start_command` and then wait for the interrupt handler and `handle_event` functions to perform the transfer. These in turn may call `fill_dma_descriptors`, `process_events`, `sdmmc_host_wait_for_event`, and some other functions I may have missed. So the transfer is not complete when `sdmmc_host_do_transaction` has returned; it has just started.
This means that IRAM has to be disabled for the duration of the transaction; this may be quite a long duration, especially if multiple sectors are transferred. Transferring sectors one by one, on the other hand, will greatly reduce the throughput.

The other issue is that SD card may need to be de-addressed using CMD7 while SPI flash is used. This is not easy to do in the current design because the higher-level SDMMC driver will always keep the card addressed. This means that disabling of SPI flash cache has to be moved up to the SD protocol layer, which means pulling even more functions into IRAM, and also breaking clean separation between SD host driver and SDMMC protocol driver.

negativekelvin, 2017-02-18:
Hopefully everything above sdmmc_host_do_transaction can stay in flash

ESP32DE, 2017-02-18:
I am glad that it is not so easy.
:) ok i have understand.

Can we perhabs work with the / cs line on SPI flash like
SPI / HSPI overlap mode in ESP8266?

For SD the CMD7 is our temporally /cs,
We need to break SD from powerline and reinit SD after deadressing.

negativekelvin, 2017-02-18:
It seems like settling for low throughput is the only practical way.  But still OK for things like logging, data backup, firmware update from card.  In most cases 1-bit sdio on HS2 may be better.  Some cases could be solved with larger flash chip (wroom-32-128mb plz).

ESP32DE, 2017-02-18:
@igrr 

> In your last snippet you are disabling the cache and then calling printf. Printf can not work with cache disabled because it uses functions which are not in IRAM.

uncleaned test code:
=============

on bottom is the monitor output
( scroll down)

```
#include <stdlib.h>
#include <assert.h>
#include <string.h>
#include <stdio.h>

#include <freertos/FreeRTOS.h>
#include <freertos/task.h>
#include <freertos/semphr.h>
#include <rom/spi_flash.h>
#include <rom/cache.h>
#include <soc/soc.h>
#include <soc/dport_reg.h>
#include "sdkconfig.h"
#include "esp_ipc.h"
#include "esp_attr.h"
#include "esp_intr_alloc.h"
#include "esp_spi_flash.h"
#include "esp_log.h"

#include <stdio.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_vfs_fat.h"
#include "driver/sdmmc_host.h"
#include "driver/sdmmc_defs.h"
#include "sdmmc_cmd.h"

// #include <spi_flash/cache_utils.h>
// #include "spi_flash/cache_utils.h"
// #include "cache_utils.h"
#include "../cache_utils.h"

// #include "driver/uart.h"
// #include "freertos/queue.h"
// #include "esp_log.h"
// #include "soc/uart_struct.h"

static const char* TAG = "example";

uint32_t cpuid = 1;
uint32_t saved_state=0;

// extern void IRAM_ATTR spi_flash_disable_cache(uint32_t cpuid, uint32_t* saved_state);
// extern void IRAM_ATTR spi_flash_restore_cache(uint32_t cpuid, uint32_t saved_state);

// extern void spi_flash_disable_interrupts_caches_and_other_cpu();
// extern void spi_flash_enable_interrupts_caches_and_other_cpu();

// extern void spi_flash_disable_interrupts_caches_and_other_cpu_no_os();
// extern void spi_flash_enable_interrupts_caches_no_os();

/**
 * The following two functions are replacements for Cache_Read_Disable and Cache_Read_Enable
 * function in ROM. They are used to work around a bug where Cache_Read_Disable requires a call to
 * Cache_Flush before Cache_Read_Enable, even if cached data was not modified.
 */

static const uint32_t cache_mask  = DPORT_APP_CACHE_MASK_OPSDRAM | DPORT_APP_CACHE_MASK_DROM0 |
        DPORT_APP_CACHE_MASK_DRAM1 | DPORT_APP_CACHE_MASK_IROM0 |
        DPORT_APP_CACHE_MASK_IRAM1 | DPORT_APP_CACHE_MASK_IRAM0;

static void IRAM_ATTR spi_flash_disable_cache( uint32_t cpuid) // , uint32_t saved_state)
{
    uint32_t ret = 0;
    if (cpuid == 0) {
        ret |= GET_PERI_REG_BITS2(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_PRO_DCACHE_DBUG0_REG, DPORT_PRO_CACHE_STATE, DPORT_PRO_CACHE_STATE_S) != 1) {
            ;
        }
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL_REG, 1, 0, DPORT_PRO_CACHE_ENABLE_S);
    } else {
        ret |= GET_PERI_REG_BITS2(DPORT_APP_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_APP_DCACHE_DBUG0_REG, DPORT_APP_CACHE_STATE, DPORT_APP_CACHE_STATE_S) != 1) {
            ;
        }
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL_REG, 1, 0, DPORT_APP_CACHE_ENABLE_S);
    }
	
	saved_state = ret;
	
    
}

static void IRAM_ATTR spi_flash_restore_cache( uint32_t cpuid) // , uint32_t saved_state)
{
    if (cpuid == 0) {
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL_REG, 1, 1, DPORT_PRO_CACHE_ENABLE_S);
        SET_PERI_REG_BITS(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, saved_state, 0);
    } else {
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL_REG, 1, 1, DPORT_APP_CACHE_ENABLE_S);
        SET_PERI_REG_BITS(DPORT_APP_CACHE_CTRL1_REG, cache_mask, saved_state, 0);
    }
}

void testregister(void) {
	
    // simple register test
	
	uint32_t testme = 0;
	
	testme |= GET_PERI_REG_BITS2(DPORT_PRO_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_PRO_DCACHE_DBUG0_REG, DPORT_PRO_CACHE_STATE, DPORT_PRO_CACHE_STATE_S) != 1) {
            ;
        }
		
	printf("ret cpu 0 : %p\n\r", (uint32_t *) testme);
	
	testme = 0;
	
	testme |= GET_PERI_REG_BITS2(DPORT_APP_CACHE_CTRL1_REG, cache_mask, 0);
        while (GET_PERI_REG_BITS2(DPORT_APP_DCACHE_DBUG0_REG, DPORT_APP_CACHE_STATE, DPORT_APP_CACHE_STATE_S) != 1) {
            ;
        }
		
	printf("ret cpu 1 : %p\n\r", (uint32_t *) testme);
		
}

void IRAM_ATTR app_main(void)
{	
    ESP_LOGI(TAG, "..test app_main starts..");
	vTaskDelay(1000 / portTICK_RATE_MS);
	
	// testregister();
	
    printf("1) try disable cache\n\r");
	fflush(stdout);
	
	// try disable cache cpuid 1
	spi_flash_disable_cache(1); // , (uint32_t) saved_state);

	// cache disabled - try printf here
	printf("2) disabled - really cache_disabled?\n\r");
	fflush(stdout);
	
	printf("3) cache must be disabled..uah.. what happend ?\n\r");
	fflush(stdout);
    
	// cache disabled - try printf here
	// printf("testpoint 3\n\r");
	printf("4) try restore cache\n\r");
	fflush(stdout);

	// try restore cache cpuid 1 
	spi_flash_restore_cache(1); // , (uint32_t) saved_state);

	printf("5) cache restored\n\r");
	fflush(stdout);
	
	printf("u");
	fflush(stdout);
	vTaskDelay(1000 / portTICK_RATE_MS);
	
	printf("p");
	fflush(stdout);
	vTaskDelay(1000 / portTICK_RATE_MS);
	
	printf("s");
	fflush(stdout);
	vTaskDelay(1000 / portTICK_RATE_MS);
	
	printf("\n");
	// not need
	// fflush(stdout);
	
	printf("closed");
	vTaskDelay(3000 / portTICK_RATE_MS);

	// need because no newline \n in printf's "closed" string
	fflush(stdout);
}

```

[image]

best wishes
rudi ;-)

negativekelvin, 2017-02-19:
Tried a few things.  Not able to disable cache from within sdmmc_host_do_transaction without crashing for some reason.  Also have to figure out what do with queue functions.

projectgus, 2017-03-07:
I'm a bit late to this issue, but is the goal here to support sharing the SPI flash pins with another peripheral (SD/MMC?)

For these kind of disable operations, it is probably best to use these SPI flash operation guard functions from `esp_spi_flash.h`:
```
g_flash_guard_default_ops.start();
```
and
```
g_flash_guard_default_ops.end();
```

These aren't really a public API, but they are mapped to the private functions `spi_flash_disable_interrupts_caches_and_other_cpu()` & `spi_flash_enable_interrupts_caches_and_other_cpu()`.

The reason to use these is that they also disable interrupts and preempt the other CPU. Without this, you can still get crashes if interrupts and/or the other CPU try to read from flash.

@ESP32DE I think the reason you were able to [printf() after disabling cache in this post](https://github.com/espressif/esp-idf/issues/353#issuecomment-280878456) was that app_main() runs on CPU 0, where cache was still enabled.

If there isn't an IDF issue here, it seems like the kind of discussion that may be more appropriate for the http://esp32.com discussion forum rather than here.

closed 2017-04-05 by igrr