espressif/esp-idf #354: execute tests from project folder?

mh-dev, 2017-02-18:
It did some effort to get unit tests running that are located in my project directory. I based my test setup on the one from the unit test app, but the tests were not picked up. I did some research and found a workaround to get my tests running.

I modified 'project.mk' and changed the part
```makefile 
# If TEST_COMPONENTS is set, create variables for building unit tests
ifdef TEST_COMPONENTS
override TEST_COMPONENTS := $(foreach comp,$(TEST_COMPONENTS),$(wildcard $(IDF_PATH)/components/$(comp)/test))
TEST_COMPONENT_PATHS := $(TEST_COMPONENTS)
TEST_COMPONENT_NAMES :=  $(foreach comp,$(TEST_COMPONENTS),$(lastword $(subst /, ,$(dir $(comp))))_test)
endif
```
to
```makefile 
# If TEST_COMPONENTS is set, create variables for building unit tests
ifdef TEST_COMPONENTS
override TEST_COMPONENTS := $(foreach comp,$(TEST_COMPONENTS),$(wildcard $(IDF_PATH)/components/$(comp)/test))
override TEST_COMPONENTS += $(foreach comp,$(COMPONENTS),$(wildcard $(PROJECT_PATH)/components/$(comp)/test))
# maybe add $(PROJECT_PATH)/main/test too?
TEST_COMPONENT_PATHS := $(TEST_COMPONENTS)
TEST_COMPONENT_NAMES :=  $(foreach comp,$(TEST_COMPONENTS),$(lastword $(subst /, ,$(dir $(comp))))_test)
endif
```

There might be a smarter way to do this, but I am not the makefile expert. Might it be possible to add something similar to the SDK?

My project structure
```
- project-dir/
        - sdkconfig
        - components
                - module/
                        - src
                                - module.h
                                - module.c
                        - test
                                - test_module.c
                        - component.mk // had to add 'src' folder otherwise the tests did not find the module.h file
                - unity // similar to the one from the unit test app
        - main/
                - main.c // similar to the one from the unit test app
                - component.mk
```

igrr, 2017-02-18:
Thanks for the suggestion. We should also move unity to the components folder of the ESP-IDF so that one doesn't need to copy it into the project folder.

closed 2017-03-08 by igrr

nicola-lunghi, 2018-09-03:
@igrr 

> Thanks for the suggestion. We should also move unity to the components folder of the ESP-IDF so that one doesn't need to copy it into the project folder.

Moving unity into the components folder is still not done??
There are other problems with it?