espressif/esp-idf #356: installing esp-idf: How do I install esp-idf using eclipse (Neon) in ubuntu 16.10 correctlz?

aquilanorica, 2017-02-19:
Hi there, 
if this is not the right place to ask, pls hint where to ask.
I followed the Linux Setup Guide on this site http://esp-idf.readthedocs.io/en/latest/linux-setup.html
(steps 1-4: ubuntu 16.10 64 bit).
Then I went over to this site http://esp-idf.readthedocs.io/en/latest/eclipse-setup.html
following the installation there.
Running a project in eclipse Neon, i got the following errors:
...
make: *** no rule found, ...l „/esp/myapp/build/include/config/auto.conf“, myapp C/C++ Problem
Program "xtensa-esp32-elf-gcc" not found in PATH myapp Project Properties, C++ Preprocessor Include.../Providers, [CDT Cross GCC Built-in Compiler Settings] options	C/C++ Scanner Discovery Problem
Symbol 'ESP_OK' could not be resolved	main.c	/myapp/main	line 11	Semantic Error
...
Any help is very welcome

pcbreflux, 2017-02-19:
Have you set $IDF_PATH and $PATH in your .profile and logoff/login or restart? or you can set this inside eclipse but for me its easier because this works also for command line make.

aquilanorica, 2017-02-20:
IDF_PATH is set to ~/esp/esp-idf
PATH is set to ~/esp/xtensa-esp32-elf and ~/esp/xtensa-esp32-elf/bin
PWD is set to ~/esp/myapp

pcbreflux, 2017-02-20:
Hi,

maybe this helps to set the needed Paths in eclipse:
https://esp32.com/viewtopic.php?f=2&t=1153
https://www.youtube.com/watch?v=bYh2w0HzS7s

but first check if its a eclipse or make error:
run "make" from commandline.
Dont know a way to run "make menuconfig", "make monitor" from eclipse so the commandline is your 2nd friend.

Tip: for make target "all" you can use i.e. "-j8 all" to run 8 parallel compiling processes if your cpu can do this.

aquilanorica, 2017-02-21:
Hi pcbreflux,

indeed, that helped a lot.
Thanks for that. 
Now there are only two unresolved items left: 
symbol 'cfg' cannot be resolved. 
symbol 'CONFIG_FREERTOS_HZ' cannot be resolved.
Cheers

pcbreflux, 2017-02-21:
Hi,

while it is not need for compiling with command line make, for eclipse it's useful to put the following inside (maybe below #include <xxx>) all your project files using the config (i.e. freertos).

#include "sdkconfig.h"

this is also the reason why I include "/<project>/build/include" into C/C++ General -> Path and Symbols -> Gnu C as the first line.

aquilanorica, 2017-02-21:
That did the trick!
Everything is working fine now.
Thank you very much for you patience and your dedicated help.
Cheers!

closed 2017-02-21 by aquilanorica

ahmedawais, 2018-06-01:
i am using eclipse oxygen on windows and running an example code ble_spp_server  but i am facing an error at portTICK_PERIOD_MS   and it shows that symbol CONFIG_FREERTOS_HZ not resolved
while i included all necessary library in program like #include "sdkconfig.h"  #include "freertos/FreeRTOS.h"
#include "freertos/task.h"

please help me