espressif/esp-idf #357: Access BT UART directly (bypass VHCI)

mattkelly, 2017-02-19:
The ESP32 datasheet specifies that there's an internal UART BT HCI interface (up to 4 Mbps). I suspect this is what the VHCI layer is hooked up to from inspecting libbtdm_app.a.

Is it possible to get direct access to this bus, bypassing the VHCI layer? Are there plans to expose this API?

Thanks!

Spritetm, 2017-02-21:
There is no such bus. The lower-level BT stuff is handled by the BT libraries / ROM, and there is a fair amount of programming going on there before the hardware PHY is reached. The interface between these libraries and the outside world is either an UART or SPI driver, or the VHCI.

mattkelly, 2017-02-22:
I see, looks like I misread the datasheet. That makes sense. Is there any documentation on how to configure the ESP32 to use UART/SPI from an external host rather than using VHCI?

Thanks, and sorry for the confusion!

projectgus, 2017-02-27:
> Is there any documentation on how to configure the ESP32 to accept HCI commands over UART/SPI from an external host rather than using VHCI?

We don't have this at the moment. The interface in [bt.h](https://github.com/espressif/esp-idf/blob/master/components/bt/include/bt.h) is essentially all you need from the BT side, but a UART HCI example would be a good thing for us to include in IDF.

Leaving this open as an enhancement request for now.

onosAdmin, 2017-03-07:
This feature would be a killer application!!! because the esp32 is so low cost that if it have a uart hci it will be used everywhere you need to use bluez (linux) with the Ble and wifi..
My company would have used it in all ours products..

negativekelvin, 2017-03-07:
@onosAdmin i think esp8689 is the version that targets this and should be even cheaper. Maybe there will be compatibility like esp8089/esp8266.

onosAdmin, 2017-03-08:
And when and where could we buy it??

mringwal, 2017-03-08:
Hi. I've ported BTstack to run on top of the VHCI:  https://github.com/bluekitchen/btstack/tree/esp32-freertos/port/esp32

To avoid copying packets into an incoming packet queue, I have the code in the host_recv_pkt_cb block on a semaphore until it is processed by the main stack thread.

However, it crashes when sending the first ACL packet from my thread - while the btController thread is blocked  semaphore (see main.c). Sending CMD packets during stack startup is fine.

Is blocking the btController bad? What is the correct way to 'pause' receiving of packets? With a regular HCI H4 and hardware flow control. RTS will go up eventually and the Bluetooth controller needs to deal with it somehow. (not my problem anymore... :)

(I've looked at components/bt/bluedroid/hci/hci_hal_h4.c as a reference, it might drop a packet if malloc fails, or blocks on posting an event on the incoming packets queue, but both are unlikely to happen)

Thanks!

TianHao-Espressif, 2017-03-09:
We will provide uart HCI in ESP-IDF soon(This work is complete, now it is in testing). Then developer could use UART of HCI directly(need not to make a UART-VHCI router). 

@mringwal , I think that when you recv the packet form VHCI, please post to anther task for handling, rather than handle it in host_recv_pkt_cb context.

TianHao-Espressif, 2017-03-09:
Sorry, I must do more explain. Why VHCI is called "VHCI", because the host and controller connected by VHCI are both run on the same IC. There's no external bluetooth host. Besides, VHCI use H4 protocol, same as "normal UART HCI". If someone want't to use external bluetooth host on another IC and use UART/SPI to connect to ESP32, please  forward the packet to VHCI from the UART/SPI. 
Besides, if use UART as IO, develop also can wait some days to use UART directly when we merge the "UART HCI" work into ESP32-IDF.

TianHao-Espressif, 2017-03-09:
There's a function "bool esp_vhci_host_check_send_available(void);", it can check whether VHCI is available. Developer can check this function's return value then do something blocking in the task which even handle HCI packets, rather than in VHCI recv callback.

projectgus, 2017-03-09:
@mringwal Very excited to have BTStack support ESP32! That's awesome.

To elaborate on @TianHao-Espressif 's answer a little bit, the VHCI interface also supports HCI flow control commands to communicate the host's buffer sizes to the controller.

To handle all flow control possibilities:

* Create a queue on the "host" side (BTStack) to copy the responses from the callback (as @TianHao-Espressif says, this callback can't block.)
* Use esp_vhci_host_check_send_available() before sending requests to the VHCI (and ensure the response queue is sufficiently empty before sending requests).
* Use the "Set Host Controller To Host Flow Control" and related flow control commands to communicate the size of the queue to the VHCI (see section "4.2 CONTROLLER TO HOST DATA FLOW CONTROL" in the Bluetooth specification.) This ensures that even if the host becomes busy doing something else, the response queue shouldn't fill to the point where the callback has to block or discard data.

Having to copy data in the callback has a performance impact, which is unfortunate, but this data would need to be buffered somewhere - on the VHCI side or on the host side. By buffering on the host side you get more flexibility about how to handle it, buffer sizes, etc.

mringwal, 2017-03-09:
@projectgus, @TianHao-Espressif Thanks for your answers and glad you like to see BTstack on the ESP32.

Excellent point: the Controller to Host Flow Control is sufficient to slow down a faster sender and there's enough RAM to handle a fair number of HCI Events as well.

How many events/acl/sco packets are buffered by VHCI? Are they buffered at all?

I didn't bother with this feature when working on small Arduino level MCU with little resources since the Bluetooth chipsets (e.g. an TI CC256x, or a CSR 8811) which had 1-2 magnitudes more RAM then the MCU. There, just pausing the Bluetooth via RTS was the better option not to loose any packets.

madscientist42, 2017-03-28:
@TianHao-Espressif Leaving understandings of the fact that it's "not been tested fully yet", is there any way we could get an early cut of this?  It's Open Sourced after all.  :D

A branch of things where you've got implementation done, doing testing, but make it clear it is very, very likely to be broken would be a win.  In the FOSS world we don't expect perfection, but we like/prefer getting at things sooner rather than later when we're working on the edge of things like some of us are.

negativekelvin, 2017-03-31:
https://github.com/espressif/esp32-bt-lib/commit/0986936c6d21a009d7d4249cbae8a23b0f3bd20b

madscientist42, 2017-04-03:
Epic.  Now...any documentation to drive this new functionality?

mringwal, 2017-04-06:
@projectgus, @TianHao-Espressif I've implemented the Controller to Host Flow Control in BTstack and tested with TI's CC2564 and a Broadcom BCM43438, but I'm running into an issue on ESP32.
Could you have a quick look please: https://github.com/espressif/esp-idf/issues/480

Thanks!

Btw. I'm happy to look at the bt_controller code myself (and would sign an NDA if needed).

madscientist42, 2017-04-11:
@mringwal while I'm not involved with Espressif any more than you are, there's a bunch of code in the VHCI code that is under license from CEVA (Riviera Waves), you'd have to be under NDA with them in addition to with Espressif on that score to be able to get a peek.  I've got the first part, but haven't YET needed to get the other part, because all of the work (Personal and Professional...) hasn't needed it to now..

mringwal, 2017-04-19:
@madscientist42 Thanks for info. An NDA with a CEVA/Riviera Wave could get tricky indeed!

I think I've identified the problem with the Controller to Host Flow Control (see issue). It shouldn't be hard to fix it now... I would guess it could be a single line fix.

Also, there are KConfig flags now to enable HCI Controller access via UART and an example at examples/bluetooth/controller_hci_uart to try (if you were looking forward to that).

madscientist42, 2017-04-19:
Heh...  Being under all sorts of NDA's is a common occurrence for myself- and you don't want to be there if you can help it.  It's very much a jumping through hoops thing.  >:-D

And, yes, I was looking forward to it.  It gives me some reference for the Classic functionality as well as raw LE stuff as well.  Until they get BlueDroid going fully, I've got some things with the day job and validation for my own inscrutable projects that needed to be done with raw HCI access short-term.

madscientist42, 2017-04-19:
(And yes, I know about BlueKitchen's stuff.  Right now, I'm trying to keep it FOSS.  If I need to, though, I **WILL** reach for your stuff- and that may still happen if they're delayed...)

closed 2017-05-02 by FayeY