espressif/esp-idf #362: Arduino import of IDF BLE

WALLTECH, 2017-02-20:
Hello,
I made the changes to the gatt client example in the IDF per this [example ](https://gist.github.com/me-no-dev/00bc1a4765350e46992bfd9d5d6c1473) with advertising and set it up in the IDE and it compiles, but when it runs there is this error "E (4930) BT: bta_dm_co_ble_load_local_keys: func not ported". How would I scan for BLE via an Arduino IDF use sketch so I can use more arduino libraries? The ESP32 Arduino page said the IDF page would be more helpful.

projectgus, 2017-02-27:
I don't think it's possible to combine these examples directly, the linked advertising example uses the VHCI directly (ie it crafts and sends raw HCI packets). Whereas the gatt client example uses Bluedroid to manage the HCI, and this is an extra layer of software this sits on top of the HCI. With both of these being used at once, things will get confused.

The GATT server example calls esp_ble_gap_start_advertising to start advertising. This call goes via Bluedroid. This is probably the approach to adapt into the GATT client example.

Yulong-espressif, 2017-06-11:
@WALLTECH for the client, also can advertising and connect to the remote device. And then, the client will act as a slave, and the remote device will act as the master. When the remote device do a conneted with your gatt client, you will receive a GATTC_CONNECT_EVT. So please update your SDK in the github. Any questions, please let me know.

closed 2017-07-05 by FayeY

gitjohnmoon1234, 2018-10-30:
Can i make ESP32 Master-slave integration in the gatt client mode

gitjohnmoon1234, 2018-10-30:
@Yulong-espressif    i want use the ble_spp_client  into "Master-slave integration " mode .In this mode ,the esp32 can be both connect with server and client ,can be both scan and advertise ,but i don't know how to achieve it .Could you tell me how to do it ?