espressif/esp-idf #364: Still problems with gperf 3.1 under macOS

eliabieri, 2017-02-20:
even after the fix #251  , i get the following error:
`$ make menuconfig 
cc  -DCURSES_LOC="<ncurses.h>" -DKBUILD_NO_NLS -Wno-format-security  -DLOCALE   -c -o zconf.tab.o zconf.tab.c
In file included from zconf.tab.c:237:
./zconf.hash.c:166:1: error: conflicting types for 'kconf_id_lookup'
kconf_id_lookup (register const char *str, register size_t len)
^
./zconf.hash.c:33:31: note: previous declaration is here
static const struct kconf_id *kconf_id_lookup(register const char *str, register unsigned int len);
                              ^
1 error generated.
make[1]: *** [zconf.tab.o] Error 1`

I use gperf version 3.1 under macOS 10.12.3
I completely recloned esp-idf and my version includes your fix.

projectgus, 2017-02-20:
Incremental "make" behaviour should have regenerated zconf.hash.c, but there's a chance it didn't. Although you specify it's a clean clone, can you verify  "make clean" doesn't change anything? Any chance `IDF_PATH` is pointing at an old/different clone? (That'd still be a bug, but it narrows things down.)

If neither of these illuminates the problem, could you please zip up your IDF_PATH/tools/kconfig directory and attach it here? Thanks.

eliabieri, 2017-02-21:
I had conflicting 'IDF-PATH' variables set. I'm sorry that I bothered your time.

closed 2017-02-21 by eliabieri