espressif/esp-idf #365: BLE TX output power configured to minimum level (bug/request)

aircable, 2017-02-21:
It seems like the BLE radio is configured to the minimum output power, index 0.
It's really vital that this can be increased from the -21dBm. Otherwise range is really only for testing.

There is no API to adjust the output power during the BLE initialization.
API to read output power is available.

/* adv tx power level */
#define BTM_BLE_ADV_TX_POWER_MIN        0           /* minimum tx power */
#define BTM_BLE_ADV_TX_POWER_LOW        1           /* low tx power     */
#define BTM_BLE_ADV_TX_POWER_MID        2           /* middle tx power  */
#define BTM_BLE_ADV_TX_POWER_UPPER      3           /* upper tx power   */
#define BTM_BLE_ADV_TX_POWER_MAX        4           /* maximum tx power */
typedef UINT8 tBTM_BLE_ADV_TX_POWER;

With this as the output power in dBm configured as: (from bt_target.h)
#define BTM_BLE_ADV_TX_POWER {-21, -15, -7, 1, 9}

TianHao-Espressif, 2017-02-24:
Hi, this codes is not used. This is the bluedroid original codes.
We have no Advertising TX power control in the current ESP-IDF.
Actually, we complete the codes to set adv_tx_power. But will merge into ESP-IDF later.

FayeY, 2017-10-10:
This issue should be solved in the latest IDF, please have a try, and feel free to reopen it if your problem is still unsolved.

closed 2017-10-10 by FayeY

ehsanmqn, 2018-09-10:
@FayeY Hi, Does this feature added to IDF? How I can adjust BLE tx/rx power?

chegewara, 2018-09-10:
@ehsanmqn this should answer your question:
https://github.com/espressif/esp-idf/issues/2390

https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/controller_vhci.html?highlight=esp_ble_tx_power_set#_CPPv220esp_ble_tx_power_set20esp_ble_power_type_t17esp_power_level_t