espressif/esp-idf #369: BLE GATTC: Issue reading Advertising/scan response data

lucashutchinson, 2017-02-22:
Hi all,

Hopefully this has not been previously reported.
I am using the esp-idf with the GATT_Client example.
I have got the latest version of the esp-idf (i am currently at commit: 61c7bd3)

I am currently looking into the reading and resolving the advertising and scan response data when performing a BLE scan.

This scan returns various BLE devices I have around me, including another esp32 running an un-modified version of the GATT_Server example.

However for some reason when I look at the raw advertising data, after the 31st byte the data seems to either be 0x00 or garbage. This means that on some devices only half of the device name is decoded, on others the 128bit uuid is incomplete. 

The GATT_Client seems to be receiving at least part of the scan response data as on one of the devices, part of the data that is in the response packet (at least the first few bytes) are contained in this first 31 bytes.

I initially discovered this when trying to resolve the name of a device which has the full device name in the scan response data rather than the advertising data.

I am reading the raw data using:
```
 case ESP_GAP_BLE_SCAN_RESULT_EVT: {
        esp_ble_gap_cb_param_t *scan_result = (esp_ble_gap_cb_param_t *)param;
        switch (scan_result->scan_rst.search_evt) {
        case ESP_GAP_SEARCH_INQ_RES_EVT:
			printf("\nRAW: ");
			for(int j=0; j<62; j++)
			{
				printf("%c", scan_result->scan_rst.ble_adv[j]);
			}
```

I seem to be able to decode/read other data within the scan result.
Eg, i can read the RSSI and it doesn't seem to be garbage (eg it returns plausible results such as -55dBm)

Please let me know if I am doing something wrong, if you need more information, or if you have a solution.

lucashutchinson, 2017-02-28:
Hi All,

Just checking if anyone can give me some help on this one.
There are a few things about this BLE stack that either dont seem to be working, or seem to be very complex to configure.

heyinling, 2017-03-01:
Hi, 

For parsing raw adv data, you can refer to [Core Specification Supplement (CSS) v7](https://www.bluetooth.com/specifications/adopted-specifications). Part A section 2 Examples will provides simple examples how raw data are parsed. And section 1 gives the specifics about each data type.

You can use API `esp_ble_resolve_adv_data` to parse adv data. For example:
```
value = esp_ble_resolve_adv_data(scan_result->scan_rst.ble_adv,
            ESP_BLE_AD_TYPE_NAME_CMPL, &value_len);
```
The above code will get the complete local device name. If value == NULL, it means device name is not in adv data.

lucashutchinson, 2017-03-01:
Hi there,

Yes i have tried using this.

The main issue that i have currently is that any data in the combined "advertising data+scan response data" `scan_result->scan_rst.ble_adv` past the 31st value in the array is either 0x00 or garbage.

As a result a name is often either not present (as the name starts at a higher array index) or gets cut off half way though.

An example:
Here is the raw data from the combined advertising and scan response data packets for the bose QC35 headphones that I get when using the nRF Connect app on my phone: (in hex)
`02 01 1A 03 03 BE FE 0D FF 10
01 40 0C 01 41 F4 5C 89 BC 0B
92 02 0D 0A 04 15 09 42 6F 73
65 **20** 51 75 69 65 74 43 6F 6D
66 6F 72 74 20 33 35`

and this is the data that i get when i check the `scan_result->scan_rst.ble_adv`
`02 01 1A 03 03 BE FE 0D FF 10
01 40 0C 01 41 F4 5C 89 BC 0B
92 02 0D 0A 04 15 09 42 6F 73
65 **20** 00 24 4F 08 40 00 00 00
00 00 00 00 00 00 00 00 00 00
00 00 00 00 00 3F B3 00 00 00
00 00 00`

as you can see after the highlighted 0x20 in both messages the data is no longer the same.
this leads the device name to appear when decoded using `esp_ble_resolve_adv_data` to read "Bose " instead of "Bose QuietConfort 35"

heyinling, 2017-03-01:
HI,

could you do passive scan and active scan, dump the data from both scan results?

scan type can be set with API `esp_ble_gap_set_scan_params`, `scan_type` in `esp_ble_scan_params_t`

lucashutchinson, 2017-03-01:
Hi,

I have just set the scan type to passive (it was previously active).
This is the raw data that I get from both the nrf app on my phone and from the esp:
```
Bose headphones (esp) (passive scan)
02 01 1A
03 03 BE FE
0D FF 10 01 40 0C 01 41 F4 5C 89 BC 0B 92
00 00 00
00 00 00 00 00 00 00 00 24 4F 08 40 00 00 00 00 00 00 00 00 00 00

Bose headphones (nrf app)
02 01 1A
03 03 BE FE
0D FF 10 01 40 0C 01 41 F4 5C 89 BC 0B 92
02 0A 04
15 09 42 6F 73 65 20 51 75 69 65 74 43 6F 6D 66 6F 72 74 20 33 35
```

I have formatted the data showing the different parts of the advertising and scan response data. 
As you can see the esp doesnt show the last two fields (Tx Power, and Name). I believe these two feilds are located in the scan response data.

Below I have the formatted data from an active scan:
``` 
Bose Headphones (esp) (active scan)
02 01 1A
03 03 BE FE
0D FF 10 01 40 0C 01 41 F4 5C 89 BC 0B 92
02 0D 0A 04
15 09 42 6F 73 65 20 00 24 4F 08 40 00 00 00 00 00 00 00 00 00 00

bose headphones (nrf app)
02 01 1A
03 03 BE FE
0D FF 10 01 40 0C 01 41 F4 5C 89 BC 0B 92
02 0A 04
15 09 42 6F 73 65 20 51 75 69 65 74 43 6F 6D 66 6F 72 74 20 33 35
```

As you can see the active scan has more valid data. So i believe the ESP32 is getting the scan response data, however it seems ot be truncated, always after the 0x20 (ascii space character) in the name field.

On different devices it will also truncate at this same data index in the `scan_result->scan_rst.ble_adv` array, no matter what data was in this array before or after the truncation. 

Thanks for your help so far!

heyinling, 2017-03-02:
Hi,

In core spec 4.2, the max length of adv or scan response is 31 octets (Vol 3 part C section 11). 

according to the API and my understanding, I think currently it only provide only one of adv or scan response in `scan_rst.ble_adv`. only the first 31 octets are meaningful. A bit confused of the design.

Any thoughts on why `nrf app` provides that much scan data? It supports other protocols or I'm getting wrong on core spec 4.2 ?

If you have a linux machine which support BLE (native or by USB dongle), you may use wireshark to capture HCI events (using `sudo hcitool lescan` cmd to  trigger scan) and check does other BLE device support that much scan data.

lucashutchinson, 2017-03-02:
Hi,

Correct, Advertising data has a maximum size of 31 octets.
Scan response data also has a maximum size of 31 octets. 

When advertising a device sends out advertising packets. Then if the GATT_Client is performing an active scan, the client will request a scan response to and advertising devices that it finds. The advertising device will then respond with the scan response packet. Which can contain additional and different data from that found int he regular advertising packet.

The data from the nRF app is a combination of the Advertising data + the Scan response data.

In the esp_gap_ble_api.h file the sble_scan_result_evt_param struct is defined as : 
[code]
struct ble_scan_result_evt_param {
        esp_gap_search_evt_t search_evt;            /*!< Search event type */
        esp_bd_addr_t bda;                          /*!< Bluetooth device address which has been searched */
        esp_bt_dev_type_t dev_type;                 /*!< Device type */
        esp_ble_addr_type_t ble_addr_type;          /*!< Ble device address type */
        esp_ble_evt_type_t ble_evt_type;            /*!< Ble scan result event type */
        int rssi;                                   /*!< Searched device's RSSI */
        uint8_t  ble_adv[ESP_BLE_ADV_DATA_LEN_MAX + ESP_BLE_SCAN_RSP_DATA_LEN_MAX];     /*!< Received EIR */
        int flag;                                   /*!< Advertising data flag bit */
        int num_resps;                              /*!< Scan result number */
    } scan_rst;  
[/code]

The definition of ble_adv implies a size of 62 due to the defines:
[code]
/// Advertising data maximum length
#define ESP_BLE_ADV_DATA_LEN_MAX               31
/// Scan response data maximum length
#define ESP_BLE_SCAN_RSP_DATA_LEN_MAX          31
[/code]

This would lead one to believe that the data contained within ble_adv  contains both the advertising data, as well as the scan response data.

And indeed this seems to be the case.

I have found the issue in the ble stack.
In file "btc_gap_ble.c"
in the memcpy operation on line 479 it memcpy's the wrong data length:
[code]
memcpy(param.scan_rst.ble_adv, p_data->inq_res.p_eir,
               ESP_BLE_ADV_DATA_LEN_MAX);
[/code]

this should be:
[code]
memcpy(param.scan_rst.ble_adv, p_data->inq_res.p_eir,
               ESP_BLE_ADV_DATA_LEN_MAX + ESP_BLE_SCAN_RSP_DATA_LEN_MAX);
[/code]

I will submit a pull request soon.

closed 2017-03-02 by lucashutchinson

Aalhabashna, 2019-01-11:
@lucashutchinson Can you please advise how do you get the rssi value from the scan result?