espressif/esp-idf #371: mbedtls_ssl_free hangs

kglowacki, 2017-02-22:
I'm trying to rework 'https' example to use a client certificate and make a POST to AWS-Iot. 
Request itself actually succeeds, but I'm unable to release ssl resources afterwards. mbedtls_ssl_free() hangs (or, rarely, crashes). I see similar effect with mbedtls_ssl_session_reset.

code available here (last line fails) [https://github.com/openairproject/sensor-esp32/blob/aws-iot/components/awsiot/awsiot_rest.c]

kglowacki, 2017-02-23:
if I skip mbedtls_ssl_close_notify(&ssl_ctx) it crashes and occasionally give me a coredump:

`================== CURRENT THREAD REGISTERS ===================
pc             0x40140e77	0x40140e77 <mbedtls_mpi_zeroize+11>
lbeg           0x400014fd	1073747197
lend           0x4000150d	1073747213
lcount         0xfffffffe	4294967294
sar            0x4	4
ps             0x60a20	395808
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x40125e02	1074945538
a1             0x3ffd3e60	1073561184
a2             0x3f3ffffe	1061158910
a3             0x0	0
a4             0x3ffb13e4	1073419236
a5             0x3ffb7728	1073444648
a6             0x0	0
a7             0x0	0
a8             0xa8bf0300	-1463876864
a9             0x3ffd3e40	1073561152
a10            0x3ffb13e4	1073419236
a11            0x60a20	395808
a12            0x60a20	395808
a13            0x80	128
a14            0x17	23
a15            0x0	0

==================== CURRENT THREAD STACK =====================
#0  0x40140e77 in mbedtls_mpi_zeroize (v=<optimized out>, n=2831090432) at /Users/kris/Dev/iot/ESP32/esp-idf/components/mbedtls/library/bignum.c:65
#1  0x40125e02 in mbedtls_mpi_free (X=0x3ffdf6a4) at /Users/kris/Dev/iot/ESP32/esp-

idf/components/mbedtls/library/bignum.c:104
#2  0x40128f21 in mbedtls_dhm_free (ctx=0x3ffdf634) at /Users/kris/Dev/iot/ESP32/esp-idf/components/mbedtls/library/dhm.c:403
#3  0x4011e3f7 in mbedtls_ssl_handshake_free (handshake=0x3ffdf62c) at /Users/kris/Dev/iot/ESP32/esp-idf/components/mbedtls/library/ssl_tls.c:6942
#4  0x4011e8c2 in mbedtls_ssl_free (ssl=0x3ffd4328) at /Users/kris/Dev/iot/ESP32/esp-idf/components/mbedtls/library/ssl_tls.c:7056
#5  0x40108705 in awsiot_update_shadow (awsiot_config=..., body=<optimized out>) at /Users/kris/Dev/iot/ESP32/workspace/pmsensor/components/awsiot/./awsiot_rest.c:400
#6  0x40107d26 in awsiot_task () at /Users/kris/Dev/iot/ESP32/workspace/pmsensor/components/awsiot/./awsiot.c:97`

negativekelvin, 2017-02-23:
check return code of mbedtls_ssl_close_notify?

kglowacki, 2017-02-23:
mbedtls_ssl_close_notify ends with ESP_OK.

mbedtls_ssl_free crashes while trying to release memory allocated during handshake at ssl_context.handshake->dhm_ctx.pX.p, I commented out everything between handshake and ssl_free, and it still crashes. Also tried with disabled hardware acceleration for mbedtls, no luck.

kglowacki, 2017-02-23:
rookie mistake - I've passed mbedtls_ssl_context instead of a pointer so handshake() modified a copy.

closed 2017-02-23 by kglowacki

projectgus, 2017-02-24:
Thanks for letting us know, glad you got this working correctly.

> handshake() modified a copy

BTW, if passed by value instead of pointer in C then mbedtls_ssl_context will actually treat the first 4 bytes of the structure contents as the pointer. The first 4 bytes of mbedtls_ssl_context struct is a pointer to mbedtls_ssl_config, so the handshake would have treated the config structure as if it was the context and probably totally corrupted it. Yay memory unsafe languages? At least the compiler can warn for this particular one.