espressif/esp-idf #377: Ability to specify ack value for last I2C read in a transaction

nkolban, 2017-02-24:
Consider the notion that we wish to read 6 bytes from an I2C slave.  A first pass at this would be the of the ESP-IDF API called `i2c_master_read` (see http://esp-idf.readthedocs.io/en/latest/api/peripherals/i2c.html#_CPPv215i2c_master_read16i2c_cmd_handle_tP7uint8_t6size_ti)

This takes as input a buffer pointer where the data will be stored and how many bytes to read.  Sounds perfect.   Now let us further look at the last parameter called "ack".   The value of this is 0 if we wish the ESP32 to send an ACK back to the slave when ever the slave transmitted byte is received ... or the value can be 1 if we do not wish to send an ACK back to the slave the the slave transmitted byte is received.  (Note: That logic value feels backwards to me but that is another story).

Since there is only one ack flag for the whole of the `i2c_master_read` API, then either ALL bytes received from the slave are ACKed or NONE of the bytes received from the slave are ACKed.  On the surface, this sounds fine too.  However, as I study I2C slave devices, I am finding a repeating pattern that when reading data from the slaves, the slave I2C devices are expecting all the bytes that they send to the master (ESP32) to be ACKed with the EXCEPTION of the last byte that they transmit.  This byte should NOT be ACKed.

For a written description/example see the MPU6050 data sheet here ... https://www.invensense.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf

Specifically, the sequence tables on page 36 (just above table 9.4) show that the master must send a NACK (negative acknowledgement) following the last byte.   At first I thought this was just a curiosity of this I2C device, but the more devices I am looking at, the more I am seeing this pattern.

We have a workaround.  Instead of reading 6 bytes using `i2c_master_read`, I can read 5 bytes and then read 1 byte with the first 5 bytes "ACKed" and the last byte "NACKed" ... however this feels less than optimal.    If I were to be bold and suggest that the "ack" parameter not be a boolean but instead be one of a set of possible options including:

* ACK all received bytes
* NACK all received bytes
* ACK all received bytes except the last one expected and NACK it

-

santhoshEsp32, 2017-03-15:
kolban sir my this (i2c matster EEPROM Selective (Random) Read problem #422) problem sound like same issue . before posting my issue i have read this and tired to read n-1 byte from EEPROM with ack and last byte with NACK there i didnt got proper output but . it was reading 1st byte from memory.  

above you wrote that : - Since there is only one ack flag for the whole of the i2c_master_read API, then either ALL bytes received from the slave are ACKed or NONE of the bytes received from the slave are ACKed.
 
is this true have you tested this ? i thought like only last byte of total reception will get effect by that i think your right thats why iam getting this error ..

once again i'll try with this approach

pasko-zh, 2017-04-13:
@nkolban, your oberservation is absolutely correct, i.e. according to the i2c standard: When the master has read the last byte from the slave, it has to send a not acknowledge (NACK) instead of an ACK, see the links to the i2c standards on my [wiki](https://github.com/pasko-zh/brzo_i2c/wiki). I remember this very well, when I impleted my i2c library for the esp8266, [brzo_i2c](https://github.com/pasko-zh/brzo_i2c) :-)

MartyMacGyver, 2017-10-04:
Just a note to say that "workaround" is present in the example code...
`/esp-idf/examples/peripherals/i2c/main/i2c_example_main.c`

I stumbled upon this today, though it was originally because of a flipped bit (ACK == 0 versus NACK == 1 and all that). I noticed that specifying ACK correctly worked fine for the device under test, though I'm sure there are some devices that won't appreciate ending on an ACK rather than a NACK. The standards-compliant method based on that example worked fine as well (and Arduino-ESP32 does something similar in its I2C HAL code).

cubisttriangle, 2018-01-01:
Is this what i2c_ack_type_t::I2C_MASTER_LAST_NACK refers to?

nkolban, 2018-01-01:
I believe so.  This is a feature which was just recently added ... perhaps as a mechanism of satisfying this or a similar request?

cubisttriangle, 2018-01-01:
I see I2C_MASTER_LAST_NACK in the current [documentation](http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/i2c.html#_CPPv214i2c_ack_type_t), but when I look at the code for i2c_master_read, I see that it still only acccepts 0 and 1, since the input is filtered through (input_ack_type & 0x1). I'm not sure how to check my IDF version, but it should be very recent. I'm running into this same issue.

cubisttriangle, 2018-01-01:
Looks like the current github version accounts for it though. Time to update!

```
esp_err_t i2c_master_read(i2c_cmd_handle_t cmd_handle, uint8_t* data, size_t data_len, i2c_ack_type_t ack)
{
    I2C_CHECK((data != NULL), I2C_ADDR_ERROR_STR, ESP_ERR_INVALID_ARG);
    I2C_CHECK(cmd_handle != NULL, I2C_CMD_LINK_INIT_ERR_STR, ESP_ERR_INVALID_ARG);
    I2C_CHECK(ack < I2C_MASTER_ACK_MAX, I2C_ACK_TYPE_ERR_STR, ESP_ERR_INVALID_ARG);
    I2C_CHECK(data_len < 1, I2C_DATA_LEN_ERR_STR, ESP_ERR_INVALID_ARG);

    if(ack != I2C_MASTER_LAST_NACK) {
        return i2c_master_read_static(cmd_handle, data, data_len, ack);
    } else {
        if(data_len == 1) {
            return i2c_master_read_byte(cmd_handle, data, I2C_MASTER_NACK);    
        } else {
            esp_err_t ret;
            if((ret =  i2c_master_read_static(cmd_handle, data, data_len - 1, I2C_MASTER_ACK)) != ESP_OK) {
                return ret;
            }
            return i2c_master_read_byte(cmd_handle, data + data_len - 1, I2C_MASTER_NACK);
        }
    }   
}
```

koobest, 2018-01-03:
Hi, @nkolban @cubisttriangle ,
As you said, we've added this feature to i2c's driver. but from the code you paste, your code is not yet up-to-date.
see  https://github.com/espressif/esp-idf/commit/5fb16d9b2ba756f330249e75968a3110b4a45075

closed 2018-01-05 by FayeY