espressif/esp-idf #378: SPI_FLASH_MMAP_INST Exception

elix22, 2017-02-24:
Attached a small example that triggers the exception.
Both the address and the size are 4 bytes aligned .
100% reproducible
Looks like a genuine issue.

`#include "freertos/FreeRTOS.h"
#include "esp_wifi.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_event_loop.h"
#include "nvs_flash.h"
#include "driver/gpio.h"
#include <esp_log.h>
#include <esp_err.h>
#include <esp_spi_flash.h>

#define ADDR 0x300000
#define MEMMAP_SIZE (1024*1024)

static char tag[] = "memmap_test";

static spi_flash_mmap_handle_t handle;

static uint32_t pInstrPtr = 0;

void app_main(void)
{
    nvs_flash_init();
    tcpip_adapter_init();
   spi_flash_init();

   esp_err_t rc = spi_flash_mmap((uint32_t) ADDR, MEMMAP_SIZE, SPI_FLASH_MMAP_INST, (const void **)&pInstrPtr, &handle);
   
   if (rc != ESP_OK) {
      ESP_LOGE(tag, "rc from spi_flash_mmap: %d", rc);
   }
   else
   {
      ESP_LOGD(tag, "spi_flash_mmap success");
   }
}
`

negativekelvin, 2017-02-24:
pInstrPtr should be pointer?

igrr, 2017-02-24:
Thanks for reporting this.
The fix is basically:

```patch
diff --git a/components/spi_flash/flash_mmap.c b/components/spi_flash/flash_mmap.c
index f8d2e32..8d284c6 100644
--- a/components/spi_flash/flash_mmap.c
+++ b/components/spi_flash/flash_mmap.c
@@ -122,7 +122,7 @@ esp_err_t IRAM_ATTR spi_flash_mmap(size_t src_addr, size_t size, spi_flash_mmap_
         region_addr = VADDR0_START_ADDR;
     } else {
         // only part of VAddr1 is usable, so adjust for that
-        region_begin = VADDR1_FIRST_USABLE_ADDR;
+        region_begin = PRO_IRAM0_FIRST_USABLE_PAGE;
         region_size = 3 * 64 - region_begin;
         region_addr = VADDR1_FIRST_USABLE_ADDR;
     }
@@ -177,7 +177,7 @@ esp_err_t IRAM_ATTR spi_flash_mmap(size_t src_addr, size_t size, spi_flash_mmap_
         new_entry->count = page_count;
         new_entry->handle = ++s_mmap_last_handle;
         *out_handle = new_entry->handle;
-        *out_ptr = (void*) (region_addr + start * SPI_FLASH_MMU_PAGE_SIZE);
+        *out_ptr = (void*) (region_addr + (start - region_begin) * SPI_FLASH_MMU_PAGE_SIZE);
         ret = ESP_OK;
     }
```

It'll be in master after review.

elix22, 2017-02-24:
It works
Thanks

closed 2017-03-02 by igrr