espressif/esp-idf #380: BLE GATTS notification

Emaxinvader, 2017-02-25:
calling

esp_ble_gatts_send_indicate(gatts_if, conn_id, handle, (uint16_t)length, (uint8_t*)buffer, false);

the system works for a while, then it disconnects with the following error:

[1B][0;31mE (919966) BT: bta_gattc_conn_cback() - cif=3 connected=0 conn_id=3 reason=0x0013[1B][0m
[1B][0;31mE (919966) BT: btm_sec_disconnected clearing pending flag handle:0 reason:19

lucashutchinson, 2017-03-05:
Can you provide a bit more information about how your code is calling this function?
How often is this function being called?
What is the device that this is connected to?
After what period of time does the disconnection occur?

I have been using this function to send data to a connected central device with great success, with no connection drops.

Emaxinvader, 2017-03-05:
I'm implementing a serial port over BLE, the function is called every time the TxBuffer change value, every second approximately.
The device is connected to android application that is a modified version of [Google demo](https://github.com/googlesamples/android-BluetoothLeGatt)
It disconnects after ten notification approximately.
I tried to add a new characteristic, TxReady, that is checked at polling, but after a couple of minutes there is an exception (store prohibited) and ESP32 restart, it occurs only when communicates, so I think is a bluetooth problem.
I attach the code.
[SPP.zip](https://github.com/espressif/esp-idf/files/819724/SPP.zip)

lucashutchinson, 2017-03-05:
Hi Emaxinvader,

Hmm strange, I have done a similar ble<->uart implementation. I have had no issues so far.

I have been testing with the [nordic nRF connect app](https://play.google.com/store/apps/details?id=no.nordicsemi.android.mcp&hl=en). This is the app that i have used in the past to develop BLE based products and works very well, especially to test connection stability etc.

Not sure why the ESP32 is re-starting however. I did originally have an issue with a stack overflow causing resets, but it is working fine now for me.

heyinling, 2017-03-09:
HI @Emaxinvader @lucashutchinson ,

  I'm running a similar test as you suggested. If I call `esp_ble_gatts_send_indicate` in a loop without any delay, then I will get an exception soon. Is this your use scenario?

  It will be appreciated if you upload the log and elf file. Then I can check if I've got the same exception.

Thanks!

Emaxinvader, 2017-03-22:
I made a mistake, I placed ESP_GATT_RSP_BY_APP instead of ESP_GATT_AUTO_RSP in the notification attribute.
Now it works. Sorry.

closed 2017-05-02 by FayeY