espressif/esp-idf #384: rtc_get_reset_reason question

jolivepetrus, 2017-02-27:
When power on the board we expect to get POWERON_RESET (=1), but RTCWDT_RTC_RESET is reported (= 16), when calling to esp-idf rtc_get_reset_reason API.

Is this the expected behaviour?

The same happens after reset button is pressed, but the bootloader reports:
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

Spritetm, 2017-02-27:
That's a known issue with the v0 silicon, see https://espressif.com/en/content/eco-and-workarounds-bugs-esp32 for more info. This bug has been fixed in the V1 silicon; the first of those should make it to the market any time soon now.

jolivepetrus, 2017-02-28:
Thanks for the information.
Is there any way for know the silicon revision from esp-idf API?

negativekelvin, 2017-02-28:
You can check Efuse chip_version.  I wonder how many code paths conditional on chip revision and whether we will be able to disable support for rev0 in menuconfig.

jolivepetrus, 2017-03-02:
Thanks for the information.
Done!.

:)

closed 2017-03-02 by jolivepetrus