espressif/esp-idf #385: bootloader ad timer

chenkaiyao, 2017-02-27:
hi:
   in bootloader, i must use a timer. I analyze the related code of esp-idf and arduino-esp32, the timer is be related to freertos.  i find funtions in ets_sys.h.
void ets_timer_init(void);
void ets_timer_arm(ETSTimer *timer, uint32_t tmout, bool repeat);
void ets_timer_arm_us(ETSTimer *ptimer, uint32_t us, bool repeat);
void ets_timer_disarm(ETSTimer *timer);
void ets_timer_setfn(ETSTimer *ptimer, ETSTimerFunc *pfunction, void *parg);
void ets_timer_done(ETSTimer *ptimer);
     these functions work?

Spritetm, 2017-02-27:
As far as I recall, these are ROM functions that have no connection to FreeRTOS so these should work just fine in the bootloader.

chenkaiyao, 2017-02-27:
I write a prgram to test. program is work.  but timer not work.
ETSTimer timer;

void fn_timer(void *timer_arg)
{
   BOOT_DEBUG("b ");
}

int main()
{
    BOOT_DEBUG("boot start...\r\n");
    Set_System();
    HAL_UI_RGB_Color(RGB_COLOR_CYAN);

    ets_timer_init();
    ets_timer_disarm(&timer);
    ets_timer_setfn(&timer, fn_timer, NULL);
    ets_timer_arm(&timer, 1000, true);
    //ets_timer_done(&timer);
    while(1)
    {
        delay(1000);
    }
}

Spritetm, 2017-02-27:
Hmm, looking at the ROM, it may be that the interrupt used isn't mapped correctly. Can you try to see if adding ``intr_matrix_set(0, ETS_TIMER2_INTR_SOURCE, 10)`` before ets_timer_init makes a difference?

chenkaiyao, 2017-02-27:
thanks.  but still not work. is  wrong of the usage of timer functons?

```
ETSTimer timer;

void fn_timer(void *timer_arg)
{
BOOT_DEBUG("b ");
}

int main()
{
BOOT_DEBUG("boot start...\r\n");
Set_System();
HAL_UI_RGB_Color(RGB_COLOR_CYAN);
intr_matrix_set(0, ETS_TIMER2_INTR_SOURCE, 10);
ets_timer_init();
ets_timer_disarm(&timer);
ets_timer_setfn(&timer, fn_timer, NULL);
ets_timer_arm(&timer, 1000, true);
//ets_timer_done(&timer);
while(1)
{
delay(1000);
}
}
```

igrr, 2017-02-27:
ets_timer functionality will not work without running the ets scheduler, a.k.a. ets_run, which also needs to have some 'user code' to run.

I think your best bet for timers in bootloader is to use hardware timers (timer group 0 and 1). You may check the driver available in components/driver/timer.c and write a stripped-down version which doesn't use FreeRTOS critical sections.

chenkaiyao, 2017-02-27:
thank you very much! I try it

chenkaiyao, 2017-02-27:
whether  bootloader  define  interrupt  table. eg._xt_interrupt_table 。 if not， should i  add  the asm files ? or have the other function.

```
xt_handler xt_set_interrupt_handler(int n, xt_handler f, void * arg)
{
    xt_handler_table_entry * entry;
    xt_handler               old;

    if( n < 0 || n >= XCHAL_NUM_INTERRUPTS )
        return 0;       /* invalid interrupt number */
    if( Xthal_intlevel[n] > XCHAL_EXCM_LEVEL )
        return 0;       /* priority level too high to safely handle in C */

    /* Convert exception number to _xt_exception_table name */
    n = n * portNUM_PROCESSORS + xPortGetCoreID();

    entry = _xt_interrupt_table + n;
    old   = entry->handler;

    if (f) {
        entry->handler = f;
        entry->arg     = arg;
    }
    else {
        entry->handler = &xt_unhandled_interrupt;
        entry->arg     = (void*)n;
    }

    return ((old == &xt_unhandled_interrupt) ? 0 : old);
}
```

chenkaiyao, 2017-03-01:
Can you help?

ESP-Marius, 2024-03-29:
Since this issue hasnt seen any activity in many years and does not actually relate to any bugs in IDF, I think i'll just close.

@chenkaiyao if this is still something that is blocking you feel free to re-open again.

closed 2024-03-29 by ESP-Marius