espressif/esp-idf #386: Flashing issue - SPI flash detection

PyBerger, 2017-02-27:
Hi, 

A friend & I are turning made trying to fash an ESP-32S module. We managed only once and now can't get anything working.

We do force GPIO0 to low, and manually toggle the EN line.

ESPTool correctly detects the bootloader, gets in download mode and is able to 'discuss' with the bootloader to for example, get the chip_id, mac & sta addresses, ...fuse etc.

But when it tries to detect the SPI flash, it fails, and consequently isn't able to download any firmware.

In normal user mode, on a terminal, the ESP32 keeps rebooting with 

> flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57
...
...

We have tried the ESP GUI Flash tool as well - same thing, com with ESP32 is fine, but it is as if the SPI Fash would be dead.

We have tried the MTDI pin forcing too. Using FTDI, CP2102 cables...
We have this on 2 branch new modules (1 of which we managed to flash once only)... 

Any idea on what we can do as we are stuck ...

We are almost at a point, we are thinking of decapsulating the module and swap the flash to see...

Thanks for your help !
PyB

projectgus, 2017-02-28:
Hi PyB,

The "flash read err, 1000" error means the first stage ROM bootloader failed to read a valid image header for the second stage software bootloader (at offset 0x1000)

The usual reasons for this are:

* Flash encryption was enabled in config and the the first flash succeeded, but now the bootloader expects to see an encrypted header (re-flashing writes a plaintext header.)
* Nothing has been flashed to offset 0x1000 (either because secure boot is mistakenly enabled. which disables flashing of the bootloader by "make flash", or because of a mistake in the esptool command line.)
* The thing flashed to 0x1000 was not bootloader.bin.
* There are power stability problems causing the flash to brown out at that moment. This can be an issue with programming bare modules without a supporting PCB, especially if the 3.3V supply is inadequate (the built-in 3.3V regulator of a USB to Serial chip is not adequate) or if there isn't enough capacitance on the module.
* The MTDI pin is driven high on reset which has caused to go into 1.8V mode, and it's browning out.

Finally, under some circumstances it seems like there are SPI flash chips which are slow to come out of reset and therefore mis-read when the ESP32 ROM first reads them. However I wouldn't expect this problem with ESP32S modules, even though they're a third party module I have a few and they all work fine.

FayeY, 2017-05-02:
Hi PyBerger, is this problem still unsolved?

closed 2017-07-07 by FayeY