espressif/esp-idf #387: Configuration of GPIO20

PyBerger, 2017-02-28:
When I try to set the direction to GPIO20 it fails with Err invalid GPIO number.

I see that is because GPIO20 does not have a GPIO pin mux reg (as it can only be GPIO) - therefore how to set the direction.

Regards,
PyB

annejan, 2017-02-28:
That's because there is no PIN 20.

Some early boards unfortunately had this pin advertised but it is NC on all boards.

PyBerger, 2017-02-28:
Understood.

closed 2017-02-28 by PyBerger

annejan, 2017-02-28:
Took me a couple of hours to find out why my display driver for our first Dev board of SHA2017 badge wouldn't work, the display reset was hooked up to the NC pin 20. 
Luckily the IDF gave me same debug info you posted.

alessandro-montanari, 2020-10-19:
Hi, 
I'm having the same problem using a custom board with the ESP32 Pico V3.
This chip supports Pin 20 but it is set to 0 in the mux reg (esp-idf/components/soc/esp32/gpio_periph.c)
However I see that it is declared in esp-idf/components/soc/include/hal/gpio_types.h

If I replace the 0 in the mux reg with IO_MUX_GPIO19_REG my application works fine.

Is it possible to include this configuration in the next IDF release so that I can use the IDF repo directly?
At the moment I'm using my own fork that has that change.

Thanks