espressif/esp-idf #390: Data sent continuously from different memeory through wifi,failure rate is different.

jixunx, 2017-03-02:
SDK version:ESP-IDF v2.0
Hardware Information:ESP-WROOM-32
Net enviroment:WiFi LAN，only a station and a AP connected(ESP32 as station,PC as AP).
Bug Description(If it is a bug not my mistake):
The UDP protocol is used to transmit data continuously through the WiFi, and if the data is from 0x3FFA_E000~0x3FFF_FFFF (Internal SRAM), the failure rate will be high (failure rate is about 70%).
When sending data from 0x3F40_0000~0x3F7F_FFFF(Cache，MMU), the failure rate is very low (about 2%).
Test Steps:
1.When I define the first address of sent data to 0x3f40000.Determine whether the send failed by the return value of sendto. Number of failures per sixty counts. The failure rate is very low (about 2%).
2.When I define the first address of sent data to 0x3FFA_E000~0x3FFF_FFFF.Determine whether the send failed by the return value of sendto. Number of failures per sixty counts. The failure rate is high (about 70%).
The first address of sent data is printed in log.Code and log are in attachment。Please Check!If you have any ideals,please tell me.
[The fail rate of sent data from 0x3F40_0000~0x3F7F_FFFF is low.zip](https://github.com/espressif/esp-idf/files/813207/The.fail.rate.of.sent.data.from.0x3F40_0000.0x3F7F_FFFF.is.low.zip)
[The fail rate of sent data from 0x3FFA_E000~0x3FFF_FFFF is high.zip](https://github.com/espressif/esp-idf/files/813208/The.fail.rate.of.sent.data.from.0x3FFA_E000.0x3FFF_FFFF.is.high.zip)

projectgus, 2017-03-02:
Hi @jixnux,

You can get some clues as to what is happening here by looking at `errno` whenever the sendto() fails. The error is always 12 aka `ENOMEM`.

When you call sendto(), the LWIP layer does two things:
* Allocates a new memory buffer and copies the UDP packet data into it.
* Pushes that buffer into a queue to send to the TCP/IP task, which manages sending it over the WiFi.

I'm not sure exactly where in LWIP the allocation is failing. It doesn't mean you're out of heap memory overall, there is plenty of that remaining, but some memory is pooled inside LWIP (to avoid using all system memory if TCP/IP stack gets backed up) and this is what is running out.

The other thing to notice is that LWIP TCP/IP task and the udp_client_task are both running on any core, which means most of the time the scheduler will put one on each core so they both run simultaneously.

So udp_client_task is trying to send UDP messages to the TCP/IP task as fast as it can, and the TCP/IP task is trying to send them out the WiFi interface as fast as it can. Something has to give! And this is the cause of the ENOMEM error.

Why does this happen more from RAM than from Flash? Reading from flash cache sometimes is a "cache miss" which means data is read from the SPI flash chip, which is much slower than reading from RAM. So the TCP/IP task gets a bit more time to catch up and send out UDP packets!

How to fix this?

* The simplest way would be to add a `vTaskDelay(1)` before `sendto()` in the loop. This gives the TCP/IP task some time to send out UDP packets. But this limits your maximum transmission rate to 1 packet per tick (approx 960KB/sec with 960 byte packets and default 1ms ticks.) This reduces the failure rate by a lot, although there are still some ENOMEM results due to timing (sometimes the TCP/IP task doesn't get a chance to clear it's backlog due to a higher priority task running instead).

* Better way is to detect ENOMEM, sleep to let the TCP/IP task send some frames and make space, and then try again. Something like:
```
for(int tries = 0; tries < 100; tries++) {
    ret=sendto(sock,test_buffer[i],960,0,(struct sockaddr*)&toAddr,sizeof(toAddr));
    if (ret > 0 || errno != ENOMEM) {
	break;
    }
    vTaskDelay(1);
}
```

This reduces the failure count to zero for me, and it should still get maximum transmission rate (it only sleeps when ENOMEM is returned, and this means there are always plenty of UDP packets queued up for the TCP task to send out.)

BTW, if you're actually sending a static buffer (ie same data each time) in your project then you may want to look into the [LWIP Netconn API](http://lwip.wikia.com/wiki/Netconn_API) which provides a way to send static UDP data without copying it into a temporary buffer. This is both faster and reduces memory overhead.

jixunx, 2017-03-03:
Hi @projectgus
Thank you so much for your reply.Your analysis is very reasonable
In fact,The reason for each failure is ENOMEM.
And "better way is to detect ENOMEM, sleep to let the TCP/IP task send some frames and make space, and then try again". I have tested the method.But if transimit data from IRAM,，failuer rate will be too high.The number of failures is abnormally high.If detect ENOMEM and vTasksleep for one second,the transimite rate will be too slow.My project is similar to micro monitoring。I need fast and real time vedio transimission.At least 115.2KB/Sec.The sooner the better.And datas that camera collects must be puted in DMA buffer(IRAM area).So it is necessary to send out the data from IRAM area instead of cache-MMU area. Buf it seems that this can't work.
What I need most is transmit data from IRAM as far as it can.And what you said “You could probably tweak the LWIP memory pool sizes and queue sizes until this doesn't happen any more for your workload” may be can povide my transimission rate a little.I will try.
And if ESP32 cannot thransmit large data from IRAM fast,espressif shoud fix the problem.This will be better for many ESP32's users.
If you have any ideal,please share with me.

projectgus, 2017-03-03:
> And if ESP32 cannot thransmit large data from IRAM fast,espressif shoud fix the problem.This will be better for many ESP32's users.

There are different levels of fast. Your example code above has a task running on one CPU sending UDP packets in a tight loop, as fast as it can with nothing interrupting it (while the other CPU has a task trying to send those packets over the Wifi.) But it will always be faster for the CPU to copy data into a queue in a tight loop than to send a data frame over the Wifi.

When you say the "failure rate will be too high", do you mean with this example or with your real project sending camera data? What UDP transmit rate are you seeing?

jixunx, 2017-03-03:
> When you say the "failure rate will be too high", do you mean with this example or with your real project sending camera data? What UDP transmit rate are you seeing?
Both this example and my real project are "failure rate will be too high".UDP rate in my real project is 330KB/s.I did a lot of tests (not only single tight loop).If large data is from IRAM ,data can'not be sent fast(330KB/s).

negativekelvin, 2017-03-04:
Wifi camera 100mbps is shown: https://mobile.twitter.com/gmag12/status/837263831873257473/photo/1

I don't think we have any public throughput benches for esp32?

projectgus, 2017-03-07:
I took some quick and rough benchmarks using the sample code above.

This is on a not-ideal WiFi network (fairly congested, about 15m to the router with no line-of-sight, my laptop reports -36dBm). Sending UDP packets from a ESP-WROVER-KIT, receiving on my laptop running this command to receive 100MB of data and then measure the data rate with pipe viewer:

```
nc -l -p 1112 -u | pv -s 100m -S > /dev/null
```

* With "send from IRAM" code unmodified from this post, I get "100MiB 0:00:41 [2.38MiB/s]"
* With the modification above (retrying on ENOMEM), I get no reported failures and "100MiB 0:00:54 [1.85MiB/s]"

These results (15-19Mbps) are far from ideal. To do much better than this I'd need to fix my crummy WiFi AP setup (and probably kick the other users off), and the code would probably need to be rewritten to use the LWIP RAW interface (or to inject the UDP packets as 802.11 data frames.)

jixunx, 2017-03-07:
@negativekelvin,I can't see the websites: "https://mobile.twitter.com/gmag12/status/837263831873257473/photo/1" .Because I am Chinese. I lived in China.I can't visit some websites abroad.Maybe you could simplely introduce the contents in the website.
@projectgus 
and the code would probably need to be rewritten to use the LWIP RAW interface (or to inject the UDP packets as 802.11 data frames.)
Is "the code"  my sample code?How to inject the UDP packets as 802.11 data frames in APP level?Is there any problems in esp-idf causes slow transmission rate.
Can problem be fixed in esp-idf ?Can  fast transimission API be provided to users?If can,how long will we need to wait?

projectgus, 2017-03-13:
> Can fast transimission API be provided to users?If can,how long will we need to wait?

Are you getting insufficient speed? I saw you said you needed 115kbps, the existing 15Mbps transmite rate is 130x faster than that.

Rewriting the code to use lower layers is possible, but it's complex and probably error prone and we don't have any sample code to base this on at the moment. I'd suggest you regard it as premature optimisation, unless you really need that performance.

igrr, 2017-05-03:
>existing 15Mbps transmit rate in the code above is 130x faster than that.

@jixunx do the above suggestions from @projectgus address your issue?

closed 2017-06-19 by FayeY