espressif/esp-idf #400: espcoredump issue

jobidon, 2017-03-05:
I have been trying to use the espcoredump utility and in my setup under arch linux (with IDF_PATH properly defined), the path to the esptool utility is broken.
When I run the tool to get the core dump file from flash, I get an error to the effect that "esptool.p" (notice the absence of the trailing letter 'y') is not found. Further investigation (excuse my ignorance of python) seem to indicate that line @698 of the script might be the culprit, trimming the last letter of the filename.
`self.path = self.path[:-1]`
A quick hack is to simply copy the file esptool.py file to esptool.p, but modification of the script would be a cleaner fix. Just not sure why the trim was added there in the first place.

gerekon, 2017-03-14:
@jobidon, it seems to happen if you run espcoredump.py when esptool.py has not been run/imported before (there is no compiled python file *.pyc only *.py). The second run of espcoredump.py should work normally.
In any case I added this problem to issues list.

igrr, 2017-04-05:
@jobidon This has been fixed in a593d8a.

closed 2017-04-05 by igrr