espressif/esp-idf #401: Router connection: 50% failure (50% success)

kas2, 2017-03-06:
Following me-no-dev suggestion I am reposting an issue I raised in the arduino-esp32 repo:

Hi,
I use this standard sketch to connect to Internet

```
#include <WiFi.h>

char* ssid = "xxxxxxxxxx";
char* password = "xxxxxxxx";

void setup() {
  Serial.begin(115200);
  delay(10);
  Serial.print("Connecting to "); Serial.println(ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(WiFi.status());
  }
  Serial.print("\nWiFi connected   IP: "); Serial.println(WiFi.localIP());
}

void loop() {}

```
ESP8266 connects after each reboot
ESP32 only connects exactly half the case:
 - connect OK
 - reboot
 - connection failed
 - reboot
 - connect OK
 - reboot
 - connection failed
 - reboot  ...

Any clue ??
Thanks

The full thread is > [here](https://github.com/espressif/arduino-esp32/issues/234#issuecomment-284212300) <
Thanks for your help

projectgus, 2017-03-07:
Are you in a position to set up ESP-IDF (rather than Arduino) using the steps here: http://esp-idf.readthedocs.io/en/latest/

If you compile and flash one of the IDF examples that uses Wifi (the HTTP or HTTPS examples in examples/protocols/ are good candidates), does the problem still occur?

kas2, 2017-03-07:
Well... I am a seasoned Arduino guy
... and a perfect noob when bash shells are concerned
Let's try

kas2, 2017-03-07:
@projectgus 
Thanks to you, I spent a couple of hours uploading MSYS, ESP-IDF and trying to run the "Hello World" example  :-)

Running this basic code, I can replicate the Arduino ide sketch beheaviour:

```
#include <string.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event_loop.h"
#include "esp_log.h"
#include "nvs_flash.h"

#define EXAMPLE_WIFI_SSID "Livebox-27C3"
#define EXAMPLE_WIFI_PASS "xxxxxxxxxxx"

static const char *TAG = "++kas++";

static EventGroupHandle_t wifi_event_group;

static esp_err_t event_handler(void *ctx, system_event_t *event)
{
    switch(event->event_id) {
    case SYSTEM_EVENT_STA_START:
     	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_START ----------");
        esp_wifi_connect();
        break;
    case SYSTEM_EVENT_STA_GOT_IP:
     	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_GOT_IP ----------");
        break;
    case SYSTEM_EVENT_STA_DISCONNECTED:
      	ESP_LOGI(TAG, "SYSTEM_EVENT_STA_DISCONNECTED ----------");
        break;
    default:
        break;
    }
    return ESP_OK;
}

static void initialise_wifi(void)
{
    tcpip_adapter_init();
    wifi_event_group = xEventGroupCreate();
    ESP_ERROR_CHECK( esp_event_loop_init(event_handler, NULL) );
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK( esp_wifi_init(&cfg) );
    ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
    wifi_config_t wifi_config = {
        .sta = {
            .ssid = EXAMPLE_WIFI_SSID,
            .password = EXAMPLE_WIFI_PASS,
        },
    };
    ESP_LOGI(TAG, "Setting WiFi configuration SSID %s...", wifi_config.sta.ssid);
    ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
    ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
    ESP_ERROR_CHECK( esp_wifi_start() );
}

static void http_get_task(void *pvParameters)
{
    while(1){}
}

void app_main()
{
    nvs_flash_init();
    initialise_wifi();
    xTaskCreate(&http_get_task, "http_get_task", 2048, NULL, 5, NULL);
}
```

Output:
...........
(132) ++kas++: Setting WiFi configuration SSID Livebox-27C2...
(132) wifi: wifi timer task: 3ffc51d0, prio:22, stack:3584
(162) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
(162) wifi: Init ampdu: 1
(162) wifi: mode : sta (24:0a:c4:05:0b:9c)
(162) ++kas++: SYSTEM_EVENT_STA_START ----------
(1042) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
(1692) wifi: state: init -> auth (b0)
(1702) wifi: state: auth -> assoc (0)
(1702) wifi: state: assoc -> run (10)
(1722) wifi: connected with Livebox-27C2, channel 6
(2582) event: ip: 192.168.1.25, mask: 255.255.255.0, gw: 192.168.1.1
(2582) ++kas++: SYSTEM_EVENT_STA_GOT_IP ----------

(Reboot)

............
(132) ++kas++: Setting WiFi configuration SSID Livebox-27C2...
(132) wifi: wifi timer task: 3ffc51d0, prio:22, stack:3584
(162) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
(162) wifi: Init ampdu: 1
(162) wifi: mode : sta (24:0a:c4:05:0b:9c)
(162) ++kas++: SYSTEM_EVENT_STA_START ----------
(1042) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
(1692) wifi: state: init -> auth (b0)
(1692) wifi: state: auth -> auth (4a0)
(2692) wifi: state: auth -> init (2)
(2692) ++kas++: SYSTEM_EVENT_STA_DISCONNECTED ----------

(Reboot)

.............
(132) ++kas++: Setting WiFi configuration SSID Livebox-27C2...
(132) wifi: wifi timer task: 3ffc51d0, prio:22, stack:3584
(162) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
(162) wifi: Init ampdu: 1
(162) wifi: mode : sta (24:0a:c4:05:0b:9c)
(162) ++kas++: SYSTEM_EVENT_STA_START ----------
(1042) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
(1692) wifi: state: init -> auth (b0)
(1702) wifi: state: auth -> assoc (0)
(1702) wifi: state: assoc -> run (10)
(1722) wifi: connected with Livebox-27C2, channel 6
(2582) event: ip: 192.168.1.25, mask: 255.255.255.0, gw: 192.168.1.1
(2582) ++kas++: SYSTEM_EVENT_STA_GOT_IP ----------

(Reboot)

................
(132) ++kas++: Setting WiFi configuration SSID Livebox-27C2...
(132) wifi: wifi timer task: 3ffc51d0, prio:22, stack:3584
(162) phy: phy_version: 329, Feb 22 2017, 15:58:07, 0, 0
(162) wifi: Init ampdu: 1
(162) wifi: mode : sta (24:0a:c4:05:0b:9c)
(162) ++kas++: SYSTEM_EVENT_STA_START ----------
(1042) wifi: n:6 0, o:1 0, ap:255 255, sta:6 0, prof:1
(1692) wifi: state: init -> auth (b0)
(1692) wifi: state: auth -> auth (4a0)
(2692) wifi: state: auth -> init (2)
(2692) ++kas++: SYSTEM_EVENT_STA_DISCONNECTED ----------

Board will connect every other time after each reboot
Can some one upload the code and replicate this strange beheaviour ??
Thanks for your help

igrr, 2017-04-05:
Based on the comments in https://github.com/espressif/arduino-esp32/issues/234 this was resolved on the Arduino core side, so closing this issue.

closed 2017-04-05 by igrr