espressif/esp-idf #402: Kconfig option configTIMER_TASK_PRIORITY

jolivepetrus, 2017-03-06:
It would be great to have a Kconfig option for set configTIMER_TASK_PRIORITY in FreeRTOSConfig.h

When using xTimerPendFunctionCallFromISR from an ISR to  defer some part of the interrupt  processing, the assigned configTIMER_TASK_PRIORITY is too low to guarantee that the deferred will be executed before the interrupt exits.

See: http://www.freertos.org/xTimerPendFunctionCallFromISR.html

projectgus, 2017-03-07:
Hi @jolivepetrus,

Thanks for requesting this. Have added this config item, will be available after it passes our internal review, test & merge process.

I would advise raising the timer task priority only as a last resort, it's a very "blunt instrument" as it overrides all timers in the system not only the high-priority event you wish to handle. You may wish to considering refactoring the code to use a queue or a semaphore, along with a dedicated high priority worker task, for some particular high priority event. The ISR can push into the queue and wake the high priority task.

Nevertheless, this feature will be added soon.

Angus

jolivepetrus, 2017-03-08:
Hi @projectgus,

Your are right about timers overrides ...

Thanks.

Jaume

igrr, 2017-05-03:
Implemented in 01ad387ac.

closed 2017-05-03 by igrr