espressif/esp-idf #403: FreeRTOS timer crash

HardwireIO, 2017-03-06:
I'm using FreeRTOS timers with latest versione of IDF. This is my code:

```
my_timer = xTimerCreate( "login", 3000 / portTICK_RATE_MS, pdFALSE, (int) rand(), , login_timeout_callback);	
if (my_timer == NULL)
{
	ESP_LOGE(BT_TAG, "Unable to create timer");
}
else
{
	xTimerStart( my_timer, portMAX_DELAY );
}

//--------------------------------------------------------

// login_timeout_callback
static void login_timeout_callback( void *pv_parameters )
{
	ESP_LOGE(BT_TAG, "Timeout");
}
```

Timer is successfully created and callback function is called correctly after 3 seconds, but I immediately get a panic message after :

```
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400856cf  PS      : 0x00060433  A0      : 0x800841c9  A1      : 0x3ffd2b40  
A2      : 0x0000001d  A3      : 0x3ffd2bd0  A4      : 0x00000010  A5      : 0x3ffcf01c  
A6      : 0x3ffd242c  A7      : 0x401196e8  A8      : 0x8008338a  A9      : 0x3ffd2b30  
A10     : 0x00000003  A11     : 0x00060423  A12     : 0x00060423  A13     : 0x3ffc0f90  
A14     : 0x00000001  A15     : 0x3ffc0fd0  SAR     : 0x00000018  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x0000002d  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x400856cf:0x3ffd2b40 0x400841c9:0x3ffd2b60 0x4008379f:0x3ffd2b80 0x40085020:0x3ffd2bc0 0x4011abac:0x3ffd2bf0 0x40119851:0x3ffd2c10 0x40119a34:0x3ffd2c30 0x40133afa:0x3ffd2c50 0x401198d9:0x3ffd2c70
```

Consider that application uses one single timer, so I don't think it's a matter of stack.

projectgus, 2017-03-07:
Hi @HardwireIO ,

I replied to you on the forums without noticing that you'd opened this issue:
https://esp32.com/viewtopic.php?f=13&t=768&p=6208#p6208

ESP_LOGE is implemented with printf() , so the advice there (either don't printf() from from timer callbacks, or enable nano-formatted I/O) should fix this problem.

Please let me know if it doesn't.

HardwireIO, 2017-03-07:
Hi @projectgus 

you're absolutely right. I can confirm you that it's a matter of ESP_LOGE.
Removing logging it works like a charm...

closed 2017-03-07 by HardwireIO