espressif/esp-idf #405: esp_bt_controller_enable(ESP_BT_MODE_BLE) results in ESP_ERR_INVALID_ARG error

lucashutchinson, 2017-03-07:
Hi All,

If I call the `esp_bt_controller_enable` function with the argument `ESP_BT_MODE_BLE` it will return an error of `ESP_ERR_INVALID_ARG`

`error = esp_bt_controller_enable(ESP_BT_MODE_BLE);`

The call used in all of the esp-idf examples seems to initialise with `ESP_BT_MODE_BTDM`
`error = esp_bt_controller_enable(ESP_BT_MODE_BTDM);` which works.

However this is the BT classic + BLE mode, which in my use case is not needed.

Is the `ESP_BT_MODE_BLE` mode not yet supported by the IDF?

igrr, 2017-03-07:
Yep, it's not supported, as the doc says:
https://github.com/espressif/esp-idf/blob/dca0377e19c6e514c0c694b521c3495a6d46c3af/components/bt/include/bt.h#L64

lucashutchinson, 2017-03-07:
Ah right,
Sorry for not thoroughly checking through the docs.

Any idea on when support for this might be added?

TianHao-Espressif, 2017-05-03:
@lucashutchinson , this will be added. But it need much work to do.

benjaminaigner, 2017-05-03:
I would like to have this feature too, plz add :-)

jkhax0r, 2017-05-03:
I'm curious how this helps over the dual mode where you just don't use classic?   Is it just an optimization to turn off the classic mode, so is there lower power, less memory, less code, etc...?

benjaminaigner, 2017-05-03:
@jkhax0r Basically it's power consumption

MrSurly, 2017-05-03:
Also, `esp_bt_controller_enable(ESP_BT_MODE_IDLE)`  doesn't work (returns an error) -- is it possible to put BT back into idle?   Or the state it was in on power-on?

Endres, 2017-05-08:
@MrSurly `esp_bt_controller_disable(ESP_BT_MODE_BTDM);` seems to work, but it doesn't disable all of the hardware, so it is still decently drawing current.

MrSurly, 2017-05-08:
@Endres Thanks, I didn't even see `esp_bt_controller_disable`, I just assumed that `ESP_BT_MODE_IDLE` was the right way to shutdown BT.

> So no, currently no possibility seems to exists to put the ESP32 back into the state it was in on power-on. Please correct me and let me be wrong, because it would be good to be able to put it back to the state suited best for power saving.

That's too bad, I agree, it would be good.  Let's hope it's implemented soon.

As a side question, does a cold reset return the hardware back to lower power consumption?

Endres, 2017-05-08:
> As a side question, does a cold reset return the hardware back to lower power consumption?

No, after a reset initiated by `esp_restart`, I am around the same current (0.093 A) as after `esp_bt_controller_disable(ESP_BT_MODE_BTDM)`.

After a hard reset (EN pin) though I am at the "idle" current again.

TianHao-Espressif, 2017-05-09:
@MrSurly , the bluetooth powersave has not start yet. So the current is not good. Powersave is in schedule.

MrSurly, 2017-05-10:
@TianHao-Espressif Thanks for the information.

closed 2017-11-22 by FayeY