espressif/esp-idf #409: espcoredump broken pipe

forthlightning, 2017-03-07:
Hi I am trying to use espcoredump, but I am getting a broken pipe issue. 

```
python $COREDUMP_PATH/espcoredump.py info_corefile -t b64 -c coredump.dat build/my_thing.elf

espcoredump.py v0.1-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

================== CURRENT THREAD REGISTERS ===================
/Users/ivan/e/ESPTools/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: struct value *value_of_register_lazy(struct frame_info *, int): Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) [answered Y; input not from terminal]
/Users/ivan/e/ESPTools/crosstool-NG/.build/src/gdb-7.10/gdb/findvar.c:290: internal-error: struct value *value_of_register_lazy(struct frame_info *, int): Assertion `frame_id_p (get_frame_id (frame))' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Create a core file of GDB? (y or n) [answered Y; input not from terminal]
GDB/MI command failed (None / None)!

==================== CURRENT THREAD STACK =====================
Traceback (most recent call last):
  File ".../esp-idf/components/espcoredump/espcoredump.py", line 1106, in <module>
    main()
  File ".../esp-idf/components/espcoredump/espcoredump.py", line 1101, in main
    operation_func(args)
  File ".../esp-idf/components/espcoredump/espcoredump.py", line 1009, in info_corefile
    p.stdin.write("-interpreter-exec console \"bt\"\n")
IOError: [Errno 32] Broken pipe
```

nkolban, 2017-03-10:
What is in "coredump.dat"?

forthlightning, 2017-03-10:
```
+CUAAA4AAAB8AQAA
2N/8P8Dc/D/M3/w/
kN78P2Df/D/kpwEAICT8PyAk/D/Y3/w/GCT8PxQAAACkqvw/pKr8P9jf/D8AAAAA
BQAAANDb/D9zZG1fcmVhZGVyX3Rhc2sA////f8zf/D8BAAAAIA4GAAUAAABljp+R
BQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGIf8P4CH/D/oh/w/
AAAAAAAAAAABAAAAAAAAALCeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADH2nw=
p9z8PwOSAEA2CAYAJEcIgIDd/D+IAAAAiAAAABAAAAD/////AAAAAOX///80Ivw/
AQAAACQAAAAkAAAACAAAAP////8AAAAA/////wUAAAABAAAAAAAAAEnDAEBrwwBA
/////wgAAAD/////vA8IQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAOIJAAAAAAAAph1APwAAAACaNQiAoN38P2Nvbm4AAAAA
rBAIQPC/AEAwDwYALEMIgMAWCIDQ3fw/BJz+P5wlQD8EnP4/8L8AQDAJBgAsQwiA
nCVAPwAAAAAAAAAAAAAAAGIXCIAQ3vw/BJz+PwEAAACQBfw/AAAAAAEAAAAAAAAA
AAAAAAEAAAABAAAAEwAAAFSc/j8AAAAAAAAAAAwAAACl9A2AQN78P/An/D8jAAYA
rBAIQKxLDkAwAAYANCEOgAEAAAABAAAAAAD/AAAAAP+HGAiAYN78P9bfnbQqSgUA
IAAGAAaoAQALAAAAAgAAAGy8AICA3vw/WOD8P8De/D8oQAiAIMn8PwMAAAAjAAYA
UxgAgKDe/D9Y4Pw/wN78PwAAAAAW3/w/AwAAAAAAAABIIA6AwN78PwAAAAAAAAAA
AAAAAMDS/D8jCQUAAAAAAAAAAADw3vw/AwAAAAAAAMHoAwAA8N78PwAAAMG+pQEA
yKr8PwAAAADkqvw/AAAAAAAAAABg3/w/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAQADAADAAAAAwAAAKAmv1gAAADBAwAAAAAAAAAAAAAA
AAMEAAwAArAqAAAAAAAAwVwp/D8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAABs3/w/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
aKD+P2Cf/j9coP4/
YJ/+P/Cf/j+xqQEADCT8Pwwk/D9ooP4/BCT8PxUAAADk4fw/5OH8P2ig/j8AAAAA
BAAAAGCc/j9tZXNzYWdlX3JvdXRlcl8A////f1yg/j8AAAAAIAIGAAwAAAD1ITJQ
BAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGIf8P4CH/D/oh/w/
AAAAAAAAAAABAAAAAAAAALCeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAACPo1A=
AAAAAIs5CIAgAgYAAAAAAAAAAADAn/4/AAAAAAAAAAAAAAAAyaUBACACBgDIsgEA
6AMAAMCf/j8AAAAAAAAAAAji/D8AAAAAAQAAAAUAAAAAAAAA8J/+PwAAAAAAAAAA
AwAAAJ0mv1gAAADBAgAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAD8n/4/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAA
yMr8P1DJ/D+8yvw/
UMn8P1DK/D/MRg4ovCP8P7wj/D/Iyvw/tCP8PxkAAADxDKXcu7rgX8jK/D8AAAAA
AAAAAIDF/D9JRExFACbTzHOyOI6ymecAAAAAALzK/D8AAAAAIQAGAAMAAADymnZp
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGIf8P4CH/D/oh/w/
AAAAAAAAAAABAAAAAAAAALCeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABYsDo=
rBAIQIgYDUAwBAYAJEcIgBDK/D8IAAAAAQAAACMCBgDIyvw/AQAAAAAAAAB4Ivw/
8Mn8PwAAAAAgBAYAIAcGAKbSAAABAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAK0SCECm0gAA2FIIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAACAHBgCm0gAAAQAAAAAAAAAAAAAAMMr8PwAAAAAAAAAA
IwIGAMjK/D8BAAAAAAAAAAAAAABQyvw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAXMr8PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
fBn9PzAY/T9wGf0/
MBj9PxAZ/T8LqAEA7A39PwTE/D98Gf0/oCP8PwcAAABoEP0/aBD9P3wZ/T9gEP0/
EgAAAHQR/T90aVQAxF/P0gVv1LIIER8A////f3AZ/T8AAAAAIAAGAAcAAACr0T3r
EgAAAAEAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAMAAAAGIf8P4CH/D/oh/w/
AAAAAAAAAAABAAAAAAAAALCeQD8AAAAASB0AQAAAAAAAAAAAAAAAAHy//j8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABOcJU=
AAAAAIs5CIAgAAYAAAAAADhLEICQGP0/GGX8P+AY/T8AAAAAp6cBAAAAAAAAAAAA
[... 129 lines cut ...]
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAADiUzk=
AAAAAIs5CIAgAgYAWEj8PwAAAAAAV/0/kHj8P1hI/D8AAAAA3qcBAKISKgZYSPw/
/////wBX/T+QePw/WEj8P2Ax/T8AAAAAAQAAABAAAAAAAAAAQFf9PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABAAAAAAAAAAAAAAAAAAAACUePw/GAAAAB9SCUD0dvw/
AAAAAAAAAAAAAAAAAAAAAAAAAAABAAAATFf9PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
cLL8P4Cx/D9ksvw/
gLH8PwCy/D9zNUEcaPz9P0wj/D9wsvw/RCP8PwEAAAD4rPw/+Kz8P3Cy/D/wrPw/
GAAAACit/D9pcGMwAPMlMKPOvdU42gIAAAAAAGSy/D8AAAAAIAAGAAEAAAA79kJO
GAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGIf8P4CH/D/oh/w/
AAAAAAAAAAABAAAAAAAAALCeQD8AAAAASB0AQAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAshss=
AAAAAIs5CIAgAAYAAAAAAAAAAADgsfw/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
/////wAAAAAAAAAAAAAAAByt/D8AAAAAAQAAAAAAAAAAAAAAALL8PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAPoKCIAAPP4/NCP8P+wl/D8AAAAAAAAAAAyy/D8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
```

nkolban, 2017-03-10:
I just noticed ... it looks like you are using a custom build of the toolchain.  I have been using the distributed binaries of the tool chain found here ...http://esp-idf.readthedocs.io/en/latest/linux-setup.html#step-1-download-binary-toolchain-for-the-esp32

It may be that there are features of the coredump technology that are specific to this downloadable version of the code ... or maybe specific compilation flags or feature flags were enabled.

forthlightning, 2017-03-13:
New toolchain seems to have cleared up my initial issue so I am not sure if this worked. So cheers once again Neil! 

I will close when I test the coredump.

How did you tell that I was using an old toolchain? I am pretty sure I downloaded this precompiled from a link on the github readme. 

@espressif folks, is there a change I can watch for in the IDF when a new toolchain comes out?

forthlightning, 2017-03-14:
after updating the toolchain I am seeing the same issue.

gerekon, 2017-03-14:
@forthlightning, It seems to be the same issue as described here https://github.com/espressif/esp-idf/issues/321.
Please, see my explanation and workaround (using 'dbg_corefile' command of core dump script) in that topic.
Let me know if you have further problems.

forthlightning, 2017-03-14:
Ah it looks like that opens up fine. Thanks! Now to learn how to gdb...

FayeY, 2017-05-02:
Hi forthlightning, is this problem solved? Can we close this issue?

forthlightning, 2017-05-04:
yup, this looks to be all squared away!

closed 2017-05-04 by forthlightning